merbanan/rtl_433 #1941: Cannot Decode TP20 Thermometer Without FM Demod

mwaterbu, 2022-01-11:
For some reason, I need FM demod active to be able to receive/decode the temperature data from my TP20.

Issue concerns: [97]  Thermopro TP08/TP12/TP20 thermometer

When running without specifying any device decoding protocols, this sensor is read without issue.
`rtl_433`

However, when this device is the only decoding protocol specified ~90% of the signals are missed.
`rtl_433 -R 97`

If any other device is additionally specified which would enable FM demod (ex. anything using FSK modulation, like [104]  Wireless M-Bus), the decoding will work again.
`rtl_433 -R 97 -R 104`

In the initialization, the demod settings being used:
`baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us`

I am fairly new to this project, so let me know if I am missing something obvious here, thanks.

zuckschwerdt, 2022-01-11:
Can you grab a sample file (`-S known` when it's working, or `-S unknown` if not) and upload as zip? S.a. https://triq.org/rtl_433/ANALYZE.html

mwaterbu, 2022-01-12:
When I tested again today, I was unable to reproduce this behavior at first, leading me to believe this is some sort of noise issue. When the transmitter is very close to my antenna, I notice this issue, when I move a several feet away, it resolves itself, and the signals are decoded without issue. The page you linked to gave the following advice:

> Do not plug the receiver directly in a USB port, avoid noise and use a short usb cable.

I did add a USB cable between the computer the the receiver, but this did not change the situation (close to transmitter, high signal loss, further away, reception is normal). So it seems FM demod is making the system less sensitive to noise in some way.

Here is the file you requested:
[rtl_433_issue1941_TP20.zip](https://github.com/merbanan/rtl_433/files/7851899/rtl_433_issue1941_TP20.zip)

zuckschwerdt, 2022-01-12:
Both samples look good and decode with just `-R 97`.

I can't easily picture how the FSK demod could play into the OOK demod. The signal processing is one-way there and the Thermopro-TP12 uses the OOK only.

Toggling the FSK demod will impact the cpu load quite much (FSK is the biggest processing load), so the noise profile might be different, but I wouldn't expect that dramatic effects.

Try to capture a broken sample. Maybe use `-S all` then look at the file times to know there is a signal, say every minute. Then try to decode every file and throw away the good ones.

closed 2023-03-01 by mwaterbu