merbanan/rtl_433 #1942: Oregon Scientific BTHGN129: 55.9C and 55.8C where -0.1C is expected

badrpc, 2022-01-13:
I have Oregon Scientific BTHGN129 outdoor sensor (I don't have a corresponding indoor unit). rtl_433 usually decodes reasonable values from this sensor (I have a different sensor to  compare). However I noticed that sometimes this rtl_433 decodes absurdly high temperature values (55.9 and 55.8) and an unexpected drop in humidity. Looking at the graphs it seems to happen when the temperature is supposed to drop to -0.1. Interestingly enough -0.2 and -0.3 has been decoded correctly so far. Since I don't have an indoor unit I cannot rule out the faulty sensor but I though I would check here before looking for a replacement.

I'm attaching relevant graphs and signals recorded with -S known. Here are some examples with recorded files:

g011_433.94M_250k.cu8 - correctly decoded 0.0:
```
% rtl_433 -M time -M protocol -M level -M noise -M stats -M bits -r rtl_433/g011_433.94M_250k.cu8
rtl_433 version 21.12 (2021-12-14) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/ols/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 176 out of 207 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-207 ]
Test mode active. Reading samples from file: rtl_433/g011_433.94M_250k.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-13 20:10:17                    Protocol  : 12
model     : Oregon-BTHGN129                        House Code: 251
Channel   : 1            Battery   : 1             Celsius   : 0.00 C        Humidity  : 92 %          Pressure  : 1030.00 hPa
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 21.0 dB       Noise     : -21.1 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-13 20:10:17                    Protocol  : 12
model     : Oregon-BTHGN129                        House Code: 251
Channel   : 1            Battery   : 1             Celsius   : 0.00 C        Humidity  : 92 %          Pressure  : 1030.00 hPa
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 20.5 dB       Noise     : -20.7 dB
```

g013_433.94M_250k.cu8 - 55.9 - supposed -0.1:
```
% rtl_433 -M time -M protocol -M level -M noise -M stats -M bits -r rtl_433/g013_433.94M_250k.cu8
rtl_433 version 21.12 (2021-12-14) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/ols/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 176 out of 207 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-207 ]
Test mode active. Reading samples from file: rtl_433/g013_433.94M_250k.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-13 20:10:34                    Protocol  : 12
model     : Oregon-BTHGN129                        House Code: 251
Channel   : 1            Battery   : 1             Celsius   : 55.90 C       Humidity  : 60 %          Pressure  : 1030.00 hPa
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 19.3 dB       Noise     : -19.4 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-13 20:10:34                    Protocol  : 12
model     : Oregon-BTHGN129                        House Code: 251
Channel   : 1            Battery   : 1             Celsius   : 55.90 C       Humidity  : 60 %          Pressure  : 1030.00 hPa
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 21.3 dB       Noise     : -21.4 dB
```

g109_433.94M_250k.cu8 -0.2:
```
% rtl_433 -M time -M protocol -M level -M noise -M stats -M bits -r rtl_433/g109_433.94M_250k.cu8
rtl_433 version 21.12 (2021-12-14) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/ols/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 176 out of 207 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-207 ]
Test mode active. Reading samples from file: rtl_433/g109_433.94M_250k.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-13 20:42:01                    Protocol  : 12
[... 18 lines cut ...]
Trying conf file at "/home/ols/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 176 out of 207 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-207 ]
Test mode active. Reading samples from file: rtl_433/g123_433.94M_250k.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-13 20:42:58                    Protocol  : 12
model     : Oregon-BTHGN129                        House Code: 251
Channel   : 1            Battery   : 1             Celsius   : -0.30 C       Humidity  : 92 %          Pressure  : 1030.00 hPa
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 19.8 dB       Noise     : -19.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-01-13 20:42:58                    Protocol  : 12
model     : Oregon-BTHGN129                        House Code: 251
Channel   : 1            Battery   : 1             Celsius   : -0.30 C       Humidity  : 92 %          Pressure  : 1030.00 hPa
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 21.1 dB       Noise     : -21.2 dB
```

[image]
[g011_433.94M_250k.cu8.gz](https://github.com/merbanan/rtl_433/files/7865912/g011_433.94M_250k.cu8.gz)
[g013_433.94M_250k.cu8.gz](https://github.com/merbanan/rtl_433/files/7865913/g013_433.94M_250k.cu8.gz)
[g109_433.94M_250k.cu8.gz](https://github.com/merbanan/rtl_433/files/7865914/g109_433.94M_250k.cu8.gz)
[g123_433.94M_250k.cu8.gz](https://github.com/merbanan/rtl_433/files/7865917/g123_433.94M_250k.cu8.gz)

badrpc, 2022-01-14:
Thank you for confirming, somehow I was suspecting the bad sensor but wasn't sure. I have seen two bad values for the temperature: 55.9 and 55.8, which I guess corresponds to 95 50 and 85 50. I don't have other recorded signals and low temperatures are not very often here so I'm not sure when I will be able to make one. However I have the following data in Prometheus: same weird temperature values, different humidity: 
[image]

Given this looks like a broken sensor I'm not entirely sure rtl_433 has to work around this problem at all. I can add filtering at a higher level. On the other hand I saw one review on Amazon mentioning what appears to be the same issue. I suppose I just need a different sensor. The only problem is that most models have barometric pressure sensor build into the indoor units with no easy access to the data.

zuckschwerdt, 2022-01-14:
Just a hunch, are the batteries low?

badrpc, 2022-01-14:
Ha! I was thinking of the same. The sensor is reporting `Battery   : 1` however I currently have rechargeable batteries there. And it turns out I don't have spare non-rechargeable AAAs right now. Will have to get some, replace and wait. I wonder if I can get to -0.1 by putting the sensor into the fridge/freezer. I suspect I might not even get any signals from there.

merbanan, 2022-01-18:
The sensor is rated up to 60C. But we could filter out all values >50C. @zuckschwerdt I think that is the best compromise we can do.

Do you agree?

rct, 2022-01-25:
This device probably warrants specific work arounds in the decoder.  I wanted to mention issue #1964 for a general discussion of filtering and sanity checking.

gdt, 2023-10-14:
Where are we on this?  Do we know if this is a type failure, where all devices are buggy, or is it an individual failed device?  Does the official receiver show bad data?  What did the manufacturer's tech support say?

badrpc, 2023-10-15:
I could find a few amazon reviews describing the same or very similar problem: [1](https://www.amazon.fr/review/R1NA5X8J80PHV2/ref=cm_cr_srp_d_rdp_perm?ie=UTF8&ASIN=B00N1AUKLU), [2](https://www.amazon.com/review/RKLGDM9XDDOWP/ref=cm_cr_srp_d_rdp_perm?ie=UTF8&ASIN=B00LSOUX8Q), [3](https://www.amazon.it/review/R3ABJQ1CGOXQO9/ref=cm_cr_srp_d_rdp_perm?ie=UTF8&ASIN=B00N1AUKLU). I don't have an official receiver, only use it with rtl_433 (this was one of few devices with barometric pressure in the outdoor unit making these data available to rtl_433). In the review [1](https://www.amazon.fr/review/R1NA5X8J80PHV2/ref=cm_cr_srp_d_rdp_perm?ie=UTF8&ASIN=B00N1AUKLU) above they mention that they were using this unit with official receiver RAR213HG and were observing incorrect readings. I doubt I will be able to get any support from the manufacturer since I believe this is an old and discontinued product.

With the above in mind I would say that this seems like a common failure scenario for this this type of sensor.

gdt, 2023-10-15:
So I think we have arrived at:
  - the device is transmitting bad data
  - rtl_433 is decoding it correctly
  - #1964 is about the larger issues of dealing with this situation, and pending resolution on approach there is nothing to do
  - this issue is xref'd in #1964 
  - therefore this issue should be closed.

Does that seem right to you?

badrpc, 2023-10-15:
Sounds good.

closed 2023-10-15 by badrpc