merbanan/rtl_433 #1945: 868 MHz only receive Honeywell and Mbus

sdramateur, 2022-01-19:
After several hours running
Using a high tower mast in a city area
- 868MHz
Only see honeywell no other devices shown 
No TPM, 
No KEYFOP, No weather stations

Do you know if Mbus are Smart meters?

[image]

merbanan, 2022-01-19:
Mbus is used for meters. Electrica, water, gas.

sdramateur, 2022-01-19:
Thank you.
I notice 868 MHz  is very, very busy,  is there a setting or way to find unknown devices or unknown protocol to shown?

zuckschwerdt, 2022-01-19:
Maybe try with  `-Y autolevel -Y minmax -Y magest -M level -M noise -M time:usec -C si -F http` then open 127.0.0.1:8433 and switch Mode to raw.

closed 2022-03-28 by merbanan