merbanan/rtl_433 #1947: Not parsing Apollo/Watchman Oil Tank Meter

petercoxphoto, 2022-01-20:
Hi -
Installed through apt on Ubuntu 20.04:
```

zsh 37 % /usr/bin/rtl_433 -V 
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
```

I'm trying to receive updates from my Apollo oil meter (which is protocol 43 in your list), but nothing shows up when I run:

```
zsh 40 % /usr/bin/rtl_433 -R 43
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pcox/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 149 device decoding protocols [ 43 ]
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
^CSignal caught, exiting!
```

Likewise when I just run the program with no arguments:

```
zsh 41 % /usr/bin/rtl_433      
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pcox/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
^CSignal caught, exiting!
```

I killed the program after about 10 seconds in both cases above - I have the oil meter next to me on the desk in spam mode so it's sending out a pulse every second or so. 

Below are outputs from -A and -a 4 - 

```
zsh 38 % /usr/bin/rtl_433 -A
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pcox/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
Detected FSK package	2022-01-20 09:39:06
Analyzing pulses...
Total count:   63,  width: 80.66 ms		(20164 S)
[... 157 lines cut ...]
Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 20318
bitbuffer:: Number of rows: 0 
*** signal_start = 382699, signal_end = 423023, signal_len = 40324, pulses_found = 1
Distance coding: Pulse length 20323

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 20323
bitbuffer:: Number of rows: 0 
*** signal_start = 484931, signal_end = 525249, signal_len = 40318, pulses_found = 1
Distance coding: Pulse length 20317

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 20317
bitbuffer:: Number of rows: 0 
^CSignal caught, exiting!
```

I'd appreciate any help you can offer! Is there some other debugging output I can send?

Cheers,
Peter

petercoxphoto, 2022-01-20:
Update, I looked in gqrx and found the data pulse at 432.091MHz and so tuned specifically to that. Nothing shows up still when I run normally, but -A shows different results:

```
zsh 47 [130] % rtl_433 -f 432091K 
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pcox/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 432.091MHz.
Allocating 15 zero-copy buffers
^CSignal caught, exiting!
```
```
zsh 48 % rtl_433 -f 432091K -A 
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pcox/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 432.091MHz.
Allocating 15 zero-copy buffers
Detected OOK package	2022-01-20 09:50:50
Analyzing pulses...
Total count:   58,  width: 80.04 ms		(20009 S)
Pulse width distribution:
 [ 0] count:   40,  width:  428 us [368;476]	( 107 S)
 [ 1] count:   17,  width:  948 us [912;976]	( 237 S)
 [ 2] count:    1,  width: 3604 us [3604;3604]	( 901 S)
Gap width distribution:
 [ 0] count:   39,  width:  588 us [548;660]	( 147 S)
 [ 1] count:    1,  width: 1608 us [1608;1608]	( 402 S)
 [ 2] count:   17,  width: 1088 us [1068;1136]	( 272 S)
Pulse period distribution:
 [ 0] count:   34,  width: 1016 us [948;1092]	( 254 S)
 [ 1] count:   12,  width: 2076 us [1988;2560]	( 519 S)
 [ 2] count:   11,  width: 1532 us [1508;1576]	( 383 S)
Level estimates [high, low]:   1000,     49
RSSI: -12.1 dB SNR: 13.1 dB Noise: -25.2 dB
Frequency offsets [F1, F2]:    1540,      0	(+5.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 428, long_width: 948, reset_limit: 1612, sync_width: 3604
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=428,l=948,r=1612,g=0,t=0,y=3604'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2 
[00] {57} ff a9 9f fa f2 8f d7 00 
[01] { 0}                         : 

Detected OOK package	2022-01-20 09:50:51
Analyzing pulses...
Total count:   59,  width: 80.02 ms		(20004 S)
Pulse width distribution:
 [ 0] count:   41,  width:  416 us [368;464]	( 104 S)
 [ 1] count:   17,  width:  932 us [908;980]	( 233 S)
 [ 2] count:    1,  width: 2048 us [2048;2048]	( 512 S)
Gap width distribution:
 [ 0] count:   40,  width:  600 us [548;656]	( 150 S)
 [ 1] count:    2,  width: 1612 us [1612;1612]	( 403 S)
[... 567 lines cut ...]
Total count:   59,  width: 80.50 ms		(20124 S)
Pulse width distribution:
 [ 0] count:   41,  width:  476 us [444;516]	( 119 S)
 [ 1] count:   17,  width:  988 us [972;1020]	( 247 S)
 [ 2] count:    1,  width: 2528 us [2528;2528]	( 632 S)
Gap width distribution:
 [ 0] count:   40,  width:  540 us [512;596]	( 135 S)
 [ 1] count:    2,  width: 1584 us [1584;1584]	( 396 S)
 [ 2] count:   16,  width: 1044 us [1028;1076]	( 261 S)
Pulse period distribution:
 [ 0] count:   34,  width: 1016 us [976;1052]	( 254 S)
 [ 1] count:   11,  width: 2136 us [2024;2568]	( 534 S)
 [ 2] count:   13,  width: 1524 us [1488;1568]	( 381 S)
Level estimates [high, low]:   1000,     43
RSSI: -12.1 dB SNR: 13.7 dB Noise: -25.8 dB
Frequency offsets [F1, F2]:     132,      0	(+0.5 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 476, long_width: 988, reset_limit: 1588, sync_width: 2528
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=476,l=988,r=1588,g=0,t=0,y=2528'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2 
[00] {58} ff a9 9f fa da 8f dd 80 
[01] { 0}                         : 

```

petercoxphoto, 2022-01-20:
And adding the suggested flex decoder to the last:

```
zsh 52 % rtl_433 -f 432091K -A -X 'n=name,m=OOK_PWM,s=480,l=992,r=1596,g=0,t=0,y=4076'
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pcox/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 123 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 432.091MHz.
Allocating 15 zero-copy buffers
Detected OOK package	2022-01-20 09:57:01
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2022-01-20 09:57:01
model     : name         count     : 2             num_rows  : 2             rows      : 
len       : 60           data      : ffa99ffaf2afef7, 
len       : 0            data      : 
codes     : {60}ffa99ffaf2afef7, {0}
Analyzing pulses...
Total count:   61,  width: 80.54 ms		(20134 S)
Pulse width distribution:
 [ 0] count:   45,  width:  480 us [456;520]	( 120 S)
 [ 1] count:   15,  width:  992 us [968;1020]	( 248 S)
 [ 2] count:    1,  width: 2588 us [2588;2588]	( 647 S)
Gap width distribution:
 [ 0] count:   44,  width:  536 us [508;576]	( 134 S)
 [ 1] count:    2,  width: 1568 us [1560;1576]	( 392 S)
 [ 2] count:   14,  width: 1040 us [1016;1072]	( 260 S)
Pulse period distribution:
 [ 0] count:   38,  width: 1016 us [972;1060]	( 254 S)
 [ 1] count:   10,  width: 2084 us [1992;2560]	( 521 S)
 [ 2] count:   12,  width: 1528 us [1512;1544]	( 382 S)
Level estimates [high, low]:   1000,     62
RSSI: -12.1 dB SNR: 12.1 dB Noise: -24.2 dB
Frequency offsets [F1, F2]:     154,      0	(+0.6 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 480, long_width: 992, reset_limit: 1580, sync_width: 2588
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=480,l=992,r=1580,g=0,t=0,y=2588'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2 
[00] {60} ff a9 9f fa f2 af ef 70 
[01] { 0}                         : 

Detected OOK package	2022-01-20 09:57:02
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2022-01-20 09:57:02
model     : name         count     : 2             num_rows  : 2             rows      : 
len       : 60           data      : ffa99ffaf2afef7, 
len       : 0            data      : 
codes     : {60}ffa99ffaf2afef7, {0}
Analyzing pulses...
Total count:   61,  width: 80.52 ms		(20129 S)
Pulse width distribution:
 [ 0] count:   45,  width:  484 us [444;520]	( 121 S)
 [ 1] count:   15,  width:  988 us [968;1008]	( 247 S)
 [ 2] count:    1,  width: 2580 us [2580;2580]	( 645 S)
Gap width distribution:
 [ 0] count:   44,  width:  532 us [504;580]	( 133 S)
 [ 1] count:    2,  width: 1572 us [1568;1576]	( 393 S)
 [ 2] count:   14,  width: 1036 us [1016;1068]	( 259 S)
Pulse period distribution:
 [ 0] count:   38,  width: 1020 us [980;1052]	( 255 S)
 [ 1] count:    1,  width: 2560 us [2560;2560]	( 640 S)
 [ 2] count:   12,  width: 1528 us [1496;1576]	( 382 S)
[... 83 lines cut ...]
Analyzing pulses...
Total count:   61,  width: 80.56 ms		(20140 S)
Pulse width distribution:
 [ 0] count:   45,  width:  484 us [452;532]	( 121 S)
 [ 1] count:   15,  width:  992 us [968;1028]	( 248 S)
 [ 2] count:    1,  width: 2612 us [2612;2612]	( 653 S)
Gap width distribution:
 [ 0] count:   44,  width:  532 us [488;572]	( 133 S)
 [ 1] count:    2,  width: 1572 us [1560;1588]	( 393 S)
 [ 2] count:   14,  width: 1040 us [1016;1068]	( 260 S)
Pulse period distribution:
 [ 0] count:   38,  width: 1016 us [964;1076]	( 254 S)
 [ 1] count:   10,  width: 2084 us [1996;2564]	( 521 S)
 [ 2] count:   12,  width: 1532 us [1504;1600]	( 383 S)
Level estimates [high, low]:   1000,     43
RSSI: -12.1 dB SNR: 13.7 dB Noise: -25.8 dB
Frequency offsets [F1, F2]:     133,      0	(+0.5 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 484, long_width: 992, reset_limit: 1592, sync_width: 2612
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=484,l=992,r=1592,g=0,t=0,y=2612'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2 
[00] {60} ff a9 9f fa f2 af ef 70 
[01] { 0}                         : 
```

zuckschwerdt, 2022-01-20:
https://triq.org/rtl_433/ANALYZE.html -- don't exactly tune to the device frequency.
Looks like FSK, so maybe a "Oil Ultrasonic STANDARD FSK" (protocol 83).

closed 2022-01-20 by petercoxphoto