merbanan/rtl_433 #1949: immediate exit on windows or just no output

lordmundi, 2022-01-21:
I'm new to rtl-sdr, but i've tried a few things and i'm not sure what else to try. I have this rtl-sdr stick: https://smile.amazon.com/gp/product/B011HVUEME/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

Also, I went through the quick-start on the rtl-sdr site and installed the WinUSB driver with zadig and also confirmed I can see everything working in sdr#. I can tune into FM broadcasts and also hear the blips on 345 MHz in sdr#.

I tried the rtl_433 windows binary for 21.12 and 21.05 and both generally had the same behavior. With no config, if I run with all defaults I just see it register all the protocols then exit on its own:

```
Downloads\rtl_433-win-x64-21.12>rtl_433_64bit_static.exe
rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Users\frankie\Downloads\rtl_433-win-x64-21.12\rtl_433.conf"...
Trying conf file at "C:\Users\frankie\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 176 out of 207 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-207 ]

```

If I specify protocol 70 (one that i'm interested in and devices I have in my home), it just sits at this prompt forever and doesn't output anything. It should be noted that when I go to exit with ctrl-c, sometimes it locks and immediately pegs the CPU core to 100% until I end the task in task manager):

```
Downloads\rtl_433-win-x64-21.12>rtl_433_64bit_static.exe -v -v -v -f 345M -R 70
rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Users\frankie\Downloads\rtl_433-win-x64-21.12\rtl_433.conf"...
Trying conf file at "C:\Users\frankie\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registering protocol [70] "Honeywell Door/Window Sensor, 2Gig DW10/DW11, RE208 repeater"
Registered 1 out of 207 device decoding protocols

```

I've tried with a config file or with command line switches. I've tried with the rtl-sdr plugged into a USB hub and also plugged directly into the machine. I also installed the rtl_433 plugin for sdr# and tried that but could never get it to list any devices or any messages. I tried 21.12 and 21.05

Any pointers or things to try would be greatly appreciated! I was hoping to play and experiment with this on Windows before moving over to a linux machine.

gvanem, 2022-01-22:
> Any pointers or things to try would be greatly appreciated! 

Looks like an issue with the USB protocol or libusb (yuk!) itself. Try with:
```
set LIBUSB_DEBUG=5
rtl_433.exe -s2M -c NUL 
```
and see if you get any debug-traces. Or something gets stuck.

Also, get [Process Explorer](https://docs.microsoft.com/en-gb/sysinternals/downloads/process-explorer) and watch the I/O-rate while `rtl_433.exe` is running. With a 2Mbit/s sample-rate, it should look like this:
[image]

lordmundi, 2022-01-22:
thanks so much for the feedback. Here is what I get when running with the libusb debug:

```
Downloads\rtl_433-win-x64-21.12>rtl_433_64bit_static.exe -s2M -c NUL
rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Registered 176 out of 207 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-207 ]
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00003cf8] libusb: debug [libusb_init] created default context
[ 0.000997] [00003cf8] libusb: debug [libusb_init] libusb v1.0.22.11312
[ 0.000997] [00003cf8] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.000997] [00003cf8] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.001994] [00003cf8] libusb: debug [windows_init_clock] hires timer available (Frequency: 10000000 Hz)
[ 0.001994] [00003cf8] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.002991] [00003cf8] libusb: debug [htab_create] using 1021 entries hash table
[ 0.004986] [00003cf8] libusb: info [winusbx_init] libusbK DLL is not available, will use native WinUSB
[ 0.005983] [00003cf8] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.005983] [00003cf8] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.005983] [00003cf8] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.006981] [00003cf8] libusb: info [windows_init] UsbDk backend is not available
[ 0.006981] [00003cf8] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.006981] [00003cf8] libusb: debug [libusb_get_device_list]
[ 0.014030] [00003cf8] libusb: debug [winusb_get_device_list] allocating new device for session [5A]
[ 0.014030] [00003cf8] libusb: debug [winusb_get_device_list] allocating new device for session [5B]
[ 0.015027] [00003cf8] libusb: debug [winusb_get_device_list] allocating new device for session [5C]
[ 0.016024] [00003cf8] libusb: debug [winusb_get_device_list] allocating new device for session [5D]
[ 0.017022] [00003cf8] libusb: debug [winusb_get_device_list] allocating new device for session [5E]
[ 0.017022] [00003cf8] libusb: debug [get_api_type] driver(s): RTSUER
[ 0.017022] [00003cf8] libusb: debug [winusb_get_device_list] allocating new device for session [22]
[ 0.018019] [00003cf8] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.018019] [00003cf8] libusb: debug [get_api_type] matched driver name against HID API
[ 0.018019] [00003cf8] libusb: debug [winusb_get_device_list] allocating new device for session [44]
[ 0.019078] [00003cf8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.019078] [00003cf8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.019078] [00003cf8] libusb: debug [winusb_get_device_list] allocating new device for session [45]
[ 0.020075] [00003cf8] libusb: debug [get_api_type] driver(s): xusb22
[ 0.020075] [00003cf8] libusb: debug [winusb_get_device_list] allocating new device for session [46]
[ 0.020075] [00003cf8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.021072] [00003cf8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.021072] [00003cf8] libusb: debug [winusb_get_device_list] allocating new device for session [47]
[ 0.022070] [00003cf8] libusb: error [enumerate_hcd_root_hub] program assertion failed - HCD 'PCI\VEN_1B21&DEV_1242&SUBSYS_86751043&REV_00\4&27259A83&0&00E0' child not found
[ 0.023067] [00003cf8] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.023067] [00003cf8] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.023067] [00003cf8] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.024064] [00003cf8] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.024064] [00003cf8] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.024064] [00003cf8] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.024064] [00003cf8] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.025061] [00003cf8] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.025061] [00003cf8] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.025061] [00003cf8] libusb: debug [libusb_unref_device] destroy device 0.0
```

It pretty much exits immediately after that, so I'm going to go on an assumption that I wouldn't see the USB I/O. I also tried running as administrator just in case it was something permission related, but got the same thing. Any ideas?

gvanem, 2022-01-22:
Not sure what this means:
``` 
error [enumerate_hcd_root_hub] program assertion failed - HCD 'PCI\VEN_1B21&DEV_1242&SUBSYS_86751043&REV_00\4&27259A83&0&00E0' child not found
``` 

But looks like really important. 

The USB Vendor ID; `VID == 0x1B21` does looks like a unsupported RTL-stick to me. 
Here is one such supported list: https://github.com/old-dab/rtlsdr/blob/master/src/librtlsdr.c#L337-L380

No `0x1b21` there.

lordmundi, 2022-01-22:
man, i hope not. and everything seems to work fine in SDR#. Maybe I have something wrong with how it is installed in windows. I don't have an usb_enum program (or perhaps you can point me to one) but I did look in the Windows device manager and it is still listed under "Other devices" as "Bulk-In Interface" and that there isn't a driver installed for it. I followed the guide on rtl-sdr, and the SDR# software does work, but perhaps rtl_433 needs something more?  When I open the events tab on the device and look at the information, the "Parent Device" at the bottom seems to have a vendor ID and product ID that matches the RTL2838 VID and PID you have. See here:

[image]

gvanem, 2022-01-22:
What does the **Driver** tab for the device says? Here's mine:
[image]

Note it says *libwdi autogenerated*. Strange that it works in SDR# though. Maybe you'll need another session with Zadig?
And it looks fish that it says *Bulk-in Interface*. Mine says what it really is; *RTL2838UHIDIR*. And the VID/PID is okay.

lordmundi, 2022-01-22:
Well, if you look on the rtl-sdr quick start guide, they actually tell you to look for the "Bulk-in Interface" as part of the install. See here: https://www.rtl-sdr.com/rtl-sdr-quick-start-guide/

On the general tab it says the drivers for this device are not installed (Code 28). And on the Driver tab, it has:
[image]

lordmundi, 2022-01-22:
ok, for anyone else who comes here, here is what I did to get it to work.

I ran windows update and looked in the Optional updates section for drivers. I updated things for my motherboard and also eventually saw one in there for the Realtek RTL device. I did that one too and rebooted. Once it came back up, of course, things didn't work because it was using the Microsoft driver. But I ran zadig again to reinstall the winUSB driver. After that finished, everything started to work.

closed 2022-02-02 by lordmundi