merbanan/rtl_433 #1950: AcuRite Atlas 01007M Weather Station not decoding, need help

MiddleSiggy, 2022-01-21:
Need help to provide samples for new weather station:

I purchased the following from Amazon:
AcuRite Atlas 01007M Weather Station with Temperature and Humidity Gauge, Rainfall, Wind Speed, Direction & Lightning Detection for Home Forecast
https://www.amazon.com/gp/product/B074XKB239/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

and I have my pi set up and working with the radio. It does receive stuff ???, but it does not seem to decode anything.

How can I collect the samples and send them or help in decoding. I know how to code, I just don't understand the radio stuff.. yet!

Let me know how I can help, and how I can help to get this working, ultimately I want to monitor my local station and import it into home assistant.

Thank you

zuckschwerdt, 2022-01-21:
https://triq.org/rtl_433/ANALYZE.html should get you sample files. Upload as zip here.

MiddleSiggy, 2022-01-21:
So I brought the weather station inside and I began to get decoding.

So is it the antenna?

time      : 2022-01-21 18:41:00
model     : Acurite-Atlas id        : 500
channel   : A            sequence_num: 1           Battery   : 1             message_type: 37          Wind Speed: 2.0 mi/h      temperature: 17.2 F
humidity  : 72 %         strike_count: 0           strike_distance: 31       data_exception: 0         raw_message: c5f46581843c48009f46

closed 2022-01-21 by MiddleSiggy

zuckschwerdt, 2022-01-22:
Check that the signal is well in the band you are watching (not close to the edges) and that the center freq is not exactly at your signal. Check that no ofter (stronger) signal is shadowing the transmission.