merbanan/rtl_433 #1951: Support for Eberle Instat868r1

dottoreD, 2022-01-22:
Hi,
I have installation of Eberle Instat 868r1 for floor heating which is not detected or identified at the moment. Recordings for two versions 0101 and 0204 are uploaded at [
https://github.com/dottoreD/rtl_433_tests/tree/master/tests/Eberle/Instat%20868r1](url). The actual knowledge I have is shared by additional files.
The protocol after Manchester decoding seems to consist in 6 nibbles and to have a checksum/hash number at the end, that is looking as a Xor with a modification which I have not understood.

dottoreD, 2022-03-11:
The problem with Checksum is solved. In fact it was wrong decoding that I used, but not able to identify without deeper knowledge about codes that are transmitted.
Kindly check if below information is sufficient to introduce as additional device.

```
868MHz, 2-FSK Modulation, low frequency "0", high frequency "1"
always 3 repetitions

raw data: 82 Bit

xxxxxxxxxxxxxxxxxxxxxxxxxx xx xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
00000000000011111110000000 11 000101010101010101010101010101010101010101010101010101
|<-- Preamble         -->| | ?? ||<- data                                        ->|

Decoding: differential Manchester
real data starts after "000", the "01"s and "10"s in expanded preamble creating additional overhead

DeCoded: 30 Bit

111111010101010101010101010101
|    ||<- data             ->|
  /\
  ||
Overhead

General:30 Bit, 6Bit Overhead, 24 Bit data in 6 Nibbles LSB
"1000" => 1
"0100" => 2
"1100" => 3

 xxxx  xxxx  xxxx  xxxx  xxxx  xxxx
 0123  0123  0123  0123  0123  0123

  ||    ||    ||    ||    ||    ||
  ||    ||    ||    ||    ||    ||
  ||    ||    ||    ||    ||    ||
  ||    ||    ||    ||    ||    \/
  ||    ||    ||    ||    ||   MIC 0-F
  ||    ||    ||    ||    ||
  ||    ||    ||    ||    \/
  ||    ||    ||    ||   counter 0-F
  ||    ||    ||    ||
  ||    ||    ||    \/
  ||    ||    ||   Actioncode 0-F
  ||    ||    \/
  ||    ||    ID2, 0-F, randomly selected for each learning phase
  ||    ||
  ||    \/              
  ||    ID1, 0-F, randomly selected for each learning phase
  ||    
  \/              
 ID0, 0-F, randomly selected for each learning phase

Actioncodes:
A: Learn
9: Reset
7: on
0: off

Counter:
simple increment with "1" for each action
Excemption: for actioncode "A" counter is not incremented

MIC:
calculation: 15 - ((Sum Nibble 1-5) mod 15)
Validity Check: Sum Nibble 1-6 Mod 15 = 15

Learn:
xxx81x, transmttted every 6 seconds

Reset: sequence of actioncodes
9/0/7/0/7/0/7/0/0

xxx91x
xxx02x
xxx73x
xxx04x
etc.

```

crasu, 2022-11-27:
I build a gnuradio decoder for the device years ago: https://github.com/crasu/gnu-heating

gdt, 2023-10-14:
@dottoreD @crasu Is anyone working on a PR to add support to rtl-433?

crasu, 2023-10-25:
No - I figured out that my usecase will not really work.