merbanan/rtl_433 #1954: Water meter Itron EverBlu Cyble Enhanced v2.1

MauricioXavier13, 2022-01-23:
I'm using the add-on rtl_433 and auto discovery in _Home Assistant_.
I would like to read my water meter.
Brand **Itron EverBlu Cyble Enhanced v2.1**. 

I'm using a **Nooelec NESDR Mini 2 USB RTL-SDR **usb dongle attached to my nuc running HA.

I'm reading some data but nothing related to my water meter.

Can you please help in how to diacover and create a config file that can read this device or add support to it.

Thanks for you work. I really appreciate...

zuckschwerdt, 2022-01-23:
Did you try https://triq.org/rtl_433/ANALYZE.html to see if data packets are received?

MauricioXavier13, 2022-01-24:
Well. Thanks for the quick answer.
I'm kind newbie to this, so first of all sorry for my kind of questions.

I've created a config file and with the suggestion that you gave em from that link i add the line:

`frequency 433.82M

convert si

**signal_grabber -S unknown**

report_meta newmodel
report_meta time:tz
report_meta level 3`

How can i check the output in Home Assistant directories?

Thank you!

zuckschwerdt, 2022-01-24:
The config `signal_grabber -S unknown` (or `-S unknown` on the command line) will produce a lot of files, one per undecode signal. Those files will likely be in the home directory of the user account running rtl_433.
Try to find those files and lokk through them. Upload a zip here with files you think are from the meter.

Do not run the signal grabber for long times. It might fill up your drive with files.

MauricioXavier13, 2022-01-24:
> The config `signal_grabber -S unknown` (or `-S unknown` on the command line) will produce a lot of files, one per undecode signal. Those files will likely be in the home directory of the user [...]

When you say home directory, you're talking in the machine running HA or in some specific folder inside HA installation? I cannot find in HA directories.

zuckschwerdt, 2022-01-25:
I'm not familiar with HASS Add-On's but according to https://developers.home-assistant.io/docs/add-ons the rtl_433 add-on would be it's own Docker container.

Our next rtl_433 release will contain options to remotely inspect rtl_433. Hopefully that will make tings easier.
Another option available to you would be the http output, add the option line `output http` then connect to the running rtl_433 at `http://<RTL_433_DOCKER_IP>:8433/`

MauricioXavier13, 2022-01-25:
> I'm not familiar with HASS Add-On's but according to https://developers.home-assistant.io/docs/add-ons the rtl_433 add-on would be it's own Docker container.
>
> Our next rtl_433 release will contain [...]

I'll give it a try. Thank you for your patience!

enavarro222, 2022-01-25:
Hi @MauricioXavier13 

I got the same water meter. For now I didn't had time to experiment with it. However I search information about the protocol (it use indeed a 433Mhz RF link). And there is an entire reverse of the protocol here (in french) :
http://www.lamaisonsimon.fr/wiki/doku.php?id=maison2:compteur_d_eau:compteur_d_eau

A code for a CC1101 radio on a rpy is even available here : http://www.lamaisonsimon.fr/wiki/lib/exe/fetch.php?media=maison2:compteur_d_eau:radian_trx.zip (password is the file name "radian_trx")

The  important point is that that water meter only transmit data when asked to. So if you are just listening RX com with RTL433 you will need to wait someone else to wake-up your meter and ask for the data (the water distribution company should did it regularly...)

I'm really curious to see what you manage to retrieve with RTL_433 !

MauricioXavier13, 2022-01-28:
> I'm not familiar with HASS Add-On's but according to https://developers.home-assistant.io/docs/add-ons the rtl_433 add-on would be it's own Docker container.
>
> Our next rtl_433 release will contain [...]

Well with signal grabber enable for days i cannot see nothing related to the meter.
Maybe i have something wrong or my meter is really in a sleep mode only awaked by the provider maybe?

[image]
Nothing related to the watermeter? But nothing unknown detected also.

[image]

GSzabados, 2022-04-30:
I have left a comment on the Home Assistant Community forum already, but I will add here a link as well to a project made for Home Assistant and Everblu Cyble Enhanced especially:

https://github.com/neutrinus/everblu-meters

You will need a CC1101 to be able to woke up the meter first.

MauricioXavier13, 2022-04-30:
> I have left a comment on the Home Assistant Community forum already, but I will add here a link as well to a project made for Home Assistant and Everblu Cyble Enhanced especially:
> 
> https://github.com/neutrinus/everblu-meters
> 
> You will need a CC1101 to be able to woke up the meter first.

Thanks for your answer @GSzabados .
So, i can't wake up with the meter with my Nooelec NESDR Mini 2 USB RTL-SDR?

2 Questions

1. Can i wake up and receive data with the CC1101?
2. Can i use a VM in proxmox instead of a rpi, right?

GSzabados, 2022-05-01:
> Nooelec NESDR Mini 2 USB RTL-SDR

The RTL-SDRs generally designed as a receiver and not for transmission as I understand. There are some project how you can use it as a transceiver but I cannot really help with that.

The CC1101 is a Low-power Sub-1 GHz wireless transceiver, cost about 5-10 Euro. Probably the best is to get one with the SMA connector.

The project what I linked above with the RPi uses the RPi's pins to communicate with the CC1101.

Possibly you can get a USB version of the CC1101 and you can use it with a Proxmox VM, considering that you will update the code, get the right driver and manage to pass through the CC1101 to the VM, but the code in the project was written for the RPi's setup. 

The project is using the code from the above posted Frech forum, as I understand. 

If you look a bit further, there is another fork of the project:

https://github.com/psykokwak-com/everblu-meters-esp8266

It uses an ESP with a CC1101 device instead of an RPi.

I have contacted the developer and asked is it working already, but he hasn't had the hardware yet to test it. So it is work in progress. It is far more easier to get an ESP nowadays than getting an RPi.

In the RADIAN protocol you need to wake up the meter first with a message and then the meter replies, what can be decoded.

MauricioXavier13, 2022-05-01:
Well, i find a raspberry pi in my garage.

PCB says: "Raspberry Pi (c) 2011"
I guess is Model B Revision 1.0 (256MB). 

I think i will try with your support. First, I need to find how to power it and some Ethernet cable to share network with the device.

Going to amazon! 🙄

What OS should i install to run it?

GSzabados, 2022-05-01:
> I guess is Model B Revision 1.0 (256MB).

Yes, it looks like a Model B.

> What OS should i install to run it?

For the answer check this forum thread. 
https://forums.raspberrypi.com/viewtopic.php?t=260755

I would start with a Raspberry Pi OS Lite or Raspberry Pi OS with desktop, depending on how you want to use at the first place.

alramir2020, 2022-05-21:
Hi MauricioXavier13,
Can you contact me via email to alramir2020@gmail.com

gdt, 2023-10-02:
I'm calling this an answered question.  Please open a new discussion or ask on the list if you have further questions.

closed 2023-10-02 by gdt