merbanan/rtl_433 #1960: Responsible disclosure policy

JamieSlome, 2022-01-24:
Hey there!

I belong to an open source security research community, and a member (@aug5t7) has found an issue, but doesn’t know the best way to disclose it.

If not a hassle, might you kindly add a `SECURITY.md` file with an email, or another contact method? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this best practice to ensure security issues are responsibly disclosed, and it would serve as a simple instruction for security researchers in the future.

Thank you for your consideration, and I look forward to hearing from you!

(cc @huntr-helper)

zuckschwerdt, 2022-01-24:
Sorry, we don't want to imply that security is a primary concern. We can't offer the assurances or the handling.
This software is explicitly *not to be run with privileges*. Further it should *not be exposed to the internet*.
(and if you use the output to do things, know that it isn't a trusted source.)

closed 2022-01-24 by zuckschwerdt

zuckschwerdt, 2022-01-24:
To any user of rtl_433 reading this, please note: We aim to make rtl_433 *safe to use*, but it should not be assumed *secure*.
There is no reason to e.g. run with `sudo`, we do read and write files without any checks.

The output is literally pulled from thin air, it's not to be trusted. If you feed downstream system with data make sure edge cases are checked and handled. Network inputs and outputs are for use in a trusted local network, will contain unfiltered data, and might overload the recipient (know that e.g. the MQTT output can be controlled by anyone with a radio sender).

merbanan, 2022-01-24:
@JamieSlome just open an issue and we will attend to it accordingly.

But as @zuckschwerdt says this tool should mainly be used as a rf to network bridge. And the use of the data is mainly for visualization purposes. The parsing code handles >100 different protocols, it is very likely that someone can craft a specific radio transmission to that triggers unwanted behavior.

JamieSlome, 2022-01-24:
@merbanan @zuckschwerdt - thanks for the response, both!

It is unlikely that the reports are consequential given your initial comments here, but both can be read here:

https://huntr.dev/bounties/6c9cd35f-a206-4fdf-b6d1-fcd50926c2d9/
https://huntr.dev/bounties/78eee103-bd61-4b4f-b054-04ad996b39e7/

merbanan, 2022-01-24:
The issues are valid and will be fixed.

JamieSlome, 2022-02-03:
@merbanan - thanks, appreciate your time and effort here!

If possible, could you approve and confirm fixes against both reports, so that the researcher gets rewarded for their efforts?