merbanan/rtl_433 #1965: Ecowitt / Fineoffset Support for GW1000 / WS80 devices

JonFear, 2022-01-25:
As discussed in the group, some samples to take a look at and see if you can see anything more than squiggles!

When the GW2001 turns up in March I will raise a new ticket with the new samples...

BW

Jon

[GW1000 WS80 Ecowitt 868M Jon Fear cu8 files.zip](https://github.com/merbanan/rtl_433/files/7937057/GW1000.WS80.Ecowitt.868M.Jon.Fear.cu8.files.zip)

zuckschwerdt, 2022-01-26:
There is at least three different senders in there. Drop g032_868M_1000k.cu8 on https://triq.org/pdv/ to see three in one picture.
I guess e.g. g022_868M_1000k.cu8 and g025_868M_1000k.cu8 are the interesting ones. But which is which?

The cleanest signal can be decoded with `-X 'n=name,m=FSK_PCM,s=58,l=58,r=1500'`
You get:
```
g003_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4daab800b85001fffa343800
g005_868M_1000k.cu8: {137}55555555555516ea4001801c800048d4da8
g006_868M_1000k.cu8: {219}55555555555516ea4001801c800048d4daab802205001ffffd44000
g008_868M_1000k.cu8: {219}55555555555516ea4001801c800048d4daab801f05001fff8b4f000
g011_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4daab804e05001fffc7aa800
g013_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4daab82bc84001ffffa4d000
g015_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4daab803504001ffffb44000
g018_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4daab801304001fff8b32000
g021_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4daab800004001fffb347000
g023_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4daab800584001fff8e27800
g025_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4da2b800e83001fffda7b000
g028_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4da2b800e83001fffda7b000
g031_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4da2b800e83001fffda7b000
g034_868M_1000k.cu8: {219}55555555555516ea4001801c800048c4da2b833685801fffddac000
g037_868M_1000k.cu8: {220}2aaaaaaaaaaa8b752000c00e400024626d15c26ac2c00fffc3fb400
```
Which has the very common `2dd4` sync word. A [BitBench with data](https://triq.net/bitbench#c=%7B219%7D55555555555516ea4001801c800048c4daab800b85001fffa343800&c=%7B137%7D55555555555516ea4001801c800048d4da8&c=%7B219%7D55555555555516ea4001801c800048d4daab802205001ffffd44000&c=%7B219%7D55555555555516ea4001801c800048d4daab801f05001fff8b4f000&c=%7B219%7D55555555555516ea4001801c800048c4daab804e05001fffc7aa800&c=%7B219%7D55555555555516ea4001801c800048c4daab82bc84001ffffa4d000&c=%7B219%7D55555555555516ea4001801c800048c4daab803504001ffffb44000&c=%7B219%7D55555555555516ea4001801c800048c4daab801304001fff8b32000&c=%7B219%7D55555555555516ea4001801c800048c4daab800004001fffb347000&c=%7B219%7D55555555555516ea4001801c800048c4daab800584001fff8e27800&c=%7B219%7D55555555555516ea4001801c800048c4da2b800e83001fffda7b000&c=%7B219%7D55555555555516ea4001801c800048c4da2b800e83001fffda7b000&c=%7B219%7D55555555555516ea4001801c800048c4da2b800e83001fffda7b000&c=%7B219%7D55555555555516ea4001801c800048c4da2b833685801fffddac000&c=%7B220%7D2aaaaaaaaaaa8b752000c00e400024626d15c26ac2c00fffc3fb400&f=8h&a=Preamble&m=aa2dd4&cw=4).

So we can refine that flex decoder to:
`-X 'n=name,m=FSK_PCM,s=58,l=58,r=1500,preamble=aa2dd4'`

Now watch those output codes and note values from the station at the same time. A pattern should appear.

zuckschwerdt, 2022-01-26:
The last byte is an addition checksum, the byte before that a CRC, e.g.
`add_bytes(&b[0], 17) - b[17] == 0x00;` and 
`crc8(&b[0], 17, 0x31, 0x00) == 0;`

JonFear, 2022-01-27:
Hmm. Many thanks for what you have written. I did run the command line but as I do not really understand what I am looking at could not see any patterns showing when comparing to the output on the Ecowitt site. I understand that there is an API to interface the GW1000 with the Home Assistant system. I suspect that I need to take a closer look at that at the weekend and see if I can see the output in more like real time that way to compare. FWIW Ecowitt have said they will be looking to publishing support for HASS with the release of the GW2001 new system... More when I get it however I seem to have a blind spot with code so...!

zuckschwerdt, 2022-01-27:
No problem, just more leg work then. Table the codes with the values you see on the console. Paste here, or preferably as BitBench Link and we'll take a look!

zuckschwerdt, 2022-01-27:
E.g. like [this BitBench](https://triq.net/bitbench#c=55555555555516ea4001801c800048c4daab800b85001fffa343800%5B20%3A26%3A10%5D&c=55555555555516ea4001801c800048d4daab802205001ffffd44000%5B20%3A26%3A18%5D&c=55555555555516ea4001801c800048d4daab801f05001fff8b4f000%5B20%3A26%3A24%5D&c=55555555555516ea4001801c800048c4daab804e05001fffc7aa800%5B20%3A26%3A28%5D&c=55555555555516ea4001801c800048c4daab82bc84001ffffa4d000%5B20%3A26%3A34%5D&c=55555555555516ea4001801c800048c4daab803504001ffffb44000%5B20%3A26%3A38%5D&c=55555555555516ea4001801c800048c4daab801304001fff8b32000%5B20%3A26%3A42%5D&c=55555555555516ea4001801c800048c4daab800004001fffb347000%5B20%3A26%3A48%5D&c=55555555555516ea4001801c800048c4daab800584001fff8e27800%5B20%3A26%3A52%5D&c=55555555555516ea4001801c800048c4da2b800e83001fffda7b000%5B20%3A26%3A56%5D&c=55555555555516ea4001801c800048c4da2b800e83001fffda7b000%5B20%3A27%3A02%5D&c=55555555555516ea4001801c800048c4da2b800e83001fffda7b000%5B20%3A27%3A06%5D&c=55555555555516ea4001801c800048c4da2b833685801fffddac000%5B20%3A27%3A16%5D&c=2aaaaaaaaaaa8b752000c00e400024626d15c26ac2c00fffc3fb400%5B20%3A27%3A20%5D&f=TYPE%3A8h%20ID%3A8h%20TEMP%3F8h8h%208h8h%208h8h%208h%20HUM%3F8d%208h8h%208h8h%208h8h%20CRC%3A8h%20SUM%3A8h%208h8h%20&a=Preamble&m=aa2dd4&cf=ADD&cl=136&cw=8), enter values from the console where the time currently is.

JonFear, 2022-02-01:
Sorry for the delay in getting back to this, I realised I had some Netatmo devices still running so have disabled them and re-run some stuff and have zipped it all up and is attached. Did try the bitbench route but nothing seemed to make any sense so...(no surprise there). I have captured what the GW1000 cron plugin for Weather Display was capturing at the time, that is included along with the exe which produces it. I wonder if Brian the Dev 
[weather output.zip](https://github.com/merbanan/rtl_433/files/7980398/weather.output.zip)
can help out a bit more there...

zuckschwerdt, 2022-02-01:
Here is [a BitBench from your codes](https://triq.net/bitbench#c=%7B154%7D80030039004d9289ed5d26fe51013fff7d9f000%20%5B16%3A43%3A46%5D&c=%7B154%7D80030039004d9289ed5d3edf51013fff0924000%20%5B16%3A43%3A50%5D&c=%7B154%7D80030039004d9289ed5d28ce51013fffe0d4000%20%5B16%3A43%3A55%5D&c=%7B154%7D80030039004d9289ed5d1ad751013fff6150000%20%5B16%3A44%3A00%5D&c=%7B154%7D80030039004d9289ed5d4af351013fff75b0000%20%5B16%3A44%3A05%5D&c=%7B154%7D80030039005d92a9ed5d1f0153013fffe030000%20%5B16%3A44%3A19%5D&c=%7B154%7D80030039004d9289ed5d46de53013fffcff3000%20%5B16%3A44%3A24%5D&c=%7B154%7D80030039004d9289ed5e2bea65013fffb2da000%20%5B16%3A44%3A33%5D&c=%7B154%7D80030039004d92a9ed5d120165013fff3e83000%20%5B16%3A44%3A38%5D&c=%7B154%7D80030039004d9289ed5d1ae95c013fffdae6000%20%5B16%3A44%3A57%5D&c=%7B154%7D80030039005d9289ed5d1cc35c013fff372f000%20%5B16%3A45%3A02%5D&c=%7B154%7D80030039004d9289ed5d31ce5c013fff6169000%20%5B16%3A45%3A06%5D&c=%7B154%7D80030039005d9289ed5d31e641013fff697e000%20%5B16%3A45%3A25%5D&c=%7B154%7D80030039005d9289ed5e26d941013fff9391000%20%5B16%3A45%3A40%5D&c=%7B154%7D80030039004d9289ed5e35da40013fff4f4c000%20%5B16%3A45%3A44%5D&c=%7B154%7D80030039004d9289ed5d28da40013fff200f000%20%5B16%3A45%3A49%5D&c=%7B154%7D80030039004d9289ed5e22d349013fff3420000%20%5B16%3A46%3A08%5D&c=%7B154%7D80030039004d9289ed5e30dc49013fff7a7d000%20%5B16%3A46%3A13%5D&c=%7B154%7D80030039004d9289ed5d4cd24f013fffb0ca000%20%5B16%3A46%3A18%5D&c=%7B154%7D80030039004d9289ed5e60ed60013fff2984000%20%5B16%3A46%3A27%5D&c=%7B154%7D80030039005d9289ed5e40f84c013fffbf01000%20%5B16%3A46%3A56%5D&c=%7B154%7D80030039004d9289ed5e2fcb4c013fff03f7000%20%5B16%3A47%3A10%5D&c=%7B154%7D80030039004d9289ed5d60e060013fffd11e000%20%5B16%3A47%3A19%5D&c=%7B154%7D80030039004d9289ed5d22e660013fff8196000%20%5B16%3A47%3A38%5D&c=%7B154%7D80030039004d9289ed5e1dcc3e013fffe7bc000%20%5B16%3A47%3A53%5D&c=%7B154%7D80030039004d9289ed5e31d43e013fff291a000%20%5B16%3A47%3A57%5D&c=%7B154%7D80030039004d9289ed5d29f64c013fffbcd4000%20%5B16%3A48%3A12%5D&c=%7B154%7D80030039004d9289ed5d2be94c013fff0b18000%20%5B16%3A48%3A21%5D&c=%7B154%7D80030039004d9289ed5d26e14c013fffd5d5000%20%5B16%3A48%3A26%5D&c=%7B154%7D80030039005d9289ed5d2be64c013fff112b000%20%5B16%3A48%3A31%5D&c=%7B154%7D80030039004d9289ed5d27f74c013fff768d000%20%5B16%3A48%3A35%5D&c=%7B154%7D80030039004d9289ed5d33ef4c013fffdcf7000%20%5B16%3A48%3A45%5D&c=%7B154%7D80030039003e9289ed5d47f44c013fff284d000%20%5B16%3A48%3A50%5D&c=%7B154%7D80030039004d9289ed5d16d347013ffff8d5000%20%5B16%3A48%3A59%5D&f=TYPE%3A8h%20ID%3A8h%20TEMP%3F8h8h%208h8h%208h8h%208h%20HUM%3F8d%208h8h%208h8h%208h8h%20CRC%3A8h%20SUM%3A8h%208h8h%20&cf=ADD&cl=136&cw=8).
Pretty sure we got the humidity right. The other data needs to be figured out:
```
outdoor temp 9.6C
outdoor hum 93%
wind direction 257
wind speed 11.16kmh
gust speed 29.88kmh
solar radiation 7.3 wm2
solar radiation 930.0 lux
UV 0.0
UV I 0.0
extra temp1 20.6C
extra temp1 69.1F
extra hum1 51%
extra temp2 19.4C
extra temp2 66.9F
extra hum2 51%
daily wind max 60.48kmh
rain rate 0.000mm/hr
rain day 0.00mm
rain week 0.80mm
rain month 0.00mm
rain year 38.60mm
```

zuckschwerdt, 2022-02-01:
The easiest way would be if we could observe a single value slowly changing. Maybe get codes at different times and note the expected values down.

JonFear, 2022-02-02:
I wonder if we are going about this the wrong way and trying to reinvent the wheel. I have found a bit of weather related software that interprets the information sent out from the GW1000 device and presets it as a web page. The code is open source and it strikes me that you may be able to derive what is going on from that code base. The git is https://github.com/weewx/weewx and whilst is not perfect it is better than. If you want to see the interface PM me and I will give you an external link...

zuckschwerdt, 2022-02-02:
IIRC, the data format used by the wired backend is quite different to the radio packets. We are not talking with the GW1000 but the WS80 here ;)
We have a good understanding of the usual FineOffset encodings and these packets fit that. We just need to find the byte offsets to the values. I.e. temperature should be present as x10 +40, 9.6 C should be 496.

If you can get codes with slightly different known values we can likely spot them.

davidefa, 2022-02-03:
Here my findings on ws80 decoding.

Let's call:
- D0 the first data byte, D1 the second...
- D0.0 the lowest order bit of D0 ( bit 0 of D0 )...

This is a sample of data received from my WS80

D0 D1 D2 D3 D4 D5 D6 D7 D8 D9 D10 D11 D12 D13 D14 D15 D16
YY  I II II LL LL VV BB TT HH  WW  BB  GG              CC
80 0a 00 3b 00 00 88 8a 59 38  18  6d  1c  00  ff  ff  d8

Y = fixed 0x80
I = device ID ( in my case a003b as reported by app WsView )
L = light value
V = battery voltage
B = bit field D7.0 = temp.8; D7.1 = temp.9; D7.5 = bearing.8
T = temperature ( temp.0-temp.7 lowest 8 bits of temperature )
H = humidity 
W = wind speed ( wind.0-wind.7 lowest 8 bits of wind speed )
B = wind bearing ( bearing.0-bearing.7 lowest 8 bits of wind bearing)
G = wind gust ( gust.0-gust.7 lowest 8 bits of wind gust )
C = check

light value = ??
battery voltage = D6 * 0.02V ( 0x88 -> 2.72V )
temperature =  ( temp0-9 - 400 ) / 10 ( 0x259 -> 20.1°C )
humidity = D9
wind speed = wind0-8 * 0.36Km/h ( 0x18 -> 8.64Km/h )
wind bearing = bearing0-8 ( 0x6d -> 109° )
wind gust = gust0-8 * 0.36Km/h ( 0x1c -> 10.08Km/h )
Don't know where wind.8 and gust.8 are ( in my opinion in unused bits of bit field D7 )
Data D13-D15 are fixed in all my recordings

davidefa, 2022-02-04:
update to previous post:

light value = ( D4 * 256 + D5 ) * 0.078925W/m2 ( 0x05a1 -> 113.73W/m2 )
uv index = int (D13 / 10)

zuckschwerdt, 2022-02-04:
That looks perfect, thanks! I'll add full device support in the next days.

zuckschwerdt, 2022-02-06:
@davidefa the W/m2 conversion factor matches a 10 Lux value, i.e. the raw value is likely in units of 10 Lux.
(e.g. 0x05a1 -> 1441 -> 14410 Lux * 0.0079 = 113.8 W/m2)

davidefa, 2022-02-06:
Yes that make sense. 
Mine figures are based on values shown on WsView app. [Here a log](http://www.davidefabbri.net/files/ws80.ods) I have compiled in libreoffice ods format.

zuckschwerdt, 2022-02-06:
The last 16 bit value (0x3fff) is likely an option that's not installed. Perhaps a rain counter?

davidefa, 2022-02-06:
I think so.
I would like to know where wind/gust bit 8 is.

zuckschwerdt, 2022-02-06:
Preliminary support for the WS80 is added. Please watch "flags" for the wind MSB and maybe "unknown" if someone has that option on a WS80.

closed 2022-02-06 by zuckschwerdt

zuckschwerdt, 2022-02-06:
For other devices people used a motor or hair dryer to spin the wind sensor fast ;) Spin a short time for the gust reading, and much longer for the average reading.

davidefa, 2022-02-06:
I've used a compressor, but it's not enough... ( maybe he likes a trip on my car? )

zuckschwerdt, 2022-02-06:
hehe, true hacker spirit! Is it maybe an optical encoder? If so perhaps use a strobe light?

davidefa, 2022-02-06:
No, sadly nothing fancy, only paralleled 2 compressors ( that did the trick ):
D7.4 = wind.8; 
D7.6 = gust.8
( updated also original post )

zuckschwerdt, 2022-02-06:
Wonderful! Can You verify that I got his right?

- 0x03 = 2 MSB temp
- 0x10 = 1 MSB wind
- 0x20 = 1 MSB bearing
- 0x40 = 1 MSB gust
- maybe 0x80 or 0x08 = battery good? they seem to be always on in the examples.

zuckschwerdt, 2022-02-06:
Also, do you want full name credit instead of "@davidefa" in the source file?

davidefa, 2022-02-06:
Yes, the bits are ok
Regarding bit 0x08 it is always high, only in two cases I found low ( when using the compressors ), the second and third sample
Need to verify the other logs

```
[RSSI-30][full RX]80 0a 00 3b 00 00 89 8a 6b 38 54 8b d5 00 ff ff a9
14:32:23.765>
[RSSI-31][full RX]80 0a 00 3b 00 00 89 d2 6b 38 77 e3 77 00 ff ff 87
14:32:28.513>
[RSSI-32][full RX]80 0a 00 3b 00 00 89 e2 6c 38 ad 0b 77 00 ff ff 69
14:32:33.267>
[RSSI-34][full RX]80 0a 00 3b 00 00 89 ca 6b 38 7d 16 77 00 ff ff 2e
14:32:38.014>
[RSSI-34][full RX]80 0a 00 3b 00 00 89 ca 6c 38 f5 dc 77 00 ff ff 0d
14:32:42.763>
[RSSI-34][full RX]80 0a 00 3b 00 00 89 ea 6c 38 ab 14 77 00 ff ff ee
14:32:47.513>
[RSSI-33][full RX]80 0a 00 3b 00 00 89 ca 6d 38 37 44 77 00 ff ff 81
14:32:52.265>
[RSSI-52][full RX]80 0a 00 3b 00 00 89 ca 6c 38 00 fa 08 00 ff ff 45
14:32:57.016>
[RSSI-39][full RX]80 0a 00 3b 00 00 89 8a 6c 38 00 fa f5 00 ff ff 1f
14:33:01.765>
[RSSI-32][full RX]80 0a 00 3b 00 00 89 8a 6d 38 00 fa f5 00 ff ff 56
14:33:06.516>
[RSSI-39][full RX]80 0a 00 3b 00 00 89 8a 6d 38 00 fa ab 00 ff ff 9b
14:33:11.431>
```

davidefa is ok for me

zuckschwerdt, 2022-02-13:
Do you mean "abspress"? We suspected it might be a rain counter option. We only ever see two values 0xffff = 65535, and 0x3fff = 16383. Those usually flag "not available".

jemo07, 2022-02-14:
Hi, 

For some reason, I had this in my notes when I was fooling around... 
```

"Raw abs pressure in inches: ${AbsPress}"
AbsPress=`echo "(${AbsPress}*33.8637526)" | bc`  # convert to hPa
#echo "Raw abs pressure in hPa: ${AbsPress}"

```
[image]
You can see the capture vs the message the G1000 is trying to send to the server, and it's also correlated on rtl_433 output... Absolute Pressure is missing IMHO. 

```
time      : 2022-02-14 15:25:51
model     : Fineoffset-WS80                        ID        : 030010
Battery   : 1.170        Battery Voltage: 3280 mV  Temperature: 11.9 C       Humidity  : 39 %
Wind direction: 317      Wind speed: 8.4 m/s       Gust speed: 8.4 m/s       UVI       : 5.3           Light     : 69750.0 lux
Flags     : aa           Integrity : CRC
```
Again, thank you for the amazing job you and the team are doing getting this out. 

Regards.

jemo07, 2022-02-14:
Yes, I-m struggling with the value on how it gets to 1100. I currently have 3 of WS80, they are at the same FW version, and I'm seen 65535 on two of them and 16383 on one of them consistently. I just moved one to it's final destination at my hangar, but as two of them where sitting side by side, it's really strange to get two different value. If I find something relevant then I'll open an issue, else I'm really grateful for the work put into RTL_433. Cheers and thank you for taking the time.

claudobahn, 2022-06-09:
@zuckschwerdt thanks for the direct link, that's helpful. I've [updated my bitbench](https://triq.net/bitbench#c=8001002c005d2e8a8c3316091e013fffb1ae000&c=8001002c001f2d8a8933004500003fffce90000&c=8001002c005d728a9f30008100013fff3cd1000&c=8001002c005d728a9d2e008100013fffea7b000&c=8001002c006c728a9c2f008100013fff2dcd000&c=8001002c006c72aa9c2e005700013fff079c000&c=%7B154%7D8001002c006c728a8634005900013fffab12000%20%5B16%3A43%3A46%5D&c=%7B154%7D80030039004d9289ed5d3edf51013fff0924000%20%5B16%3A43%3A50%5D&c=%7B154%7D80030039004d9289ed5d28ce51013fffe0d4000%20%5B16%3A43%3A55%5D&c=%7B154%7D80030039004d9289ed5d1ad751013fff6150000%20%5B16%3A44%3A00%5D&c=%7B154%7D80030039004d9289ed5d4af351013fff75b0000%20%5B16%3A44%3A05%5D&c=%7B154%7D80030039005d92a9ed5d1f0153013fffe030000%20%5B16%3A44%3A19%5D&c=%7B154%7D80030039004d9289ed5d46de53013fffcff3000%20%5B16%3A44%3A24%5D&c=%7B154%7D80030039004d9289ed5e2bea65013fffb2da000%20%5B16%3A44%3A33%5D&c=%7B154%7D80030039004d92a9ed5d120165013fff3e83000%20%5B16%3A44%3A38%5D&c=%7B154%7D80030039004d9289ed5d1ae95c013fffdae6000%20%5B16%3A44%3A57%5D&c=%7B154%7D80030039005d9289ed5d1cc35c013fff372f000%20%5B16%3A45%3A02%5D&c=%7B154%7D80030039004d9289ed5d31ce5c013fff6169000%20%5B16%3A45%3A06%5D&c=%7B154%7D80030039005d9289ed5d31e641013fff697e000%20%5B16%3A45%3A25%5D&c=%7B154%7D80030039005d9289ed5e26d941013fff9391000%20%5B16%3A45%3A40%5D&c=%7B154%7D80030039004d9289ed5e35da40013fff4f4c000%20%5B16%3A45%3A44%5D&c=%7B154%7D80030039004d9289ed5d28da40013fff200f000%20%5B16%3A45%3A49%5D&c=%7B154%7D80030039004d9289ed5e22d349013fff3420000%20%5B16%3A46%3A08%5D&c=%7B154%7D80030039004d9289ed5e30dc49013fff7a7d000%20%5B16%3A46%3A13%5D&c=%7B154%7D80030039004d9289ed5d4cd24f013fffb0ca000%20%5B16%3A46%3A18%5D&c=%7B154%7D80030039004d9289ed5e60ed60013fff2984000%20%5B16%3A46%3A27%5D&c=%7B154%7D80030039005d9289ed5e40f84c013fffbf01000%20%5B16%3A46%3A56%5D&c=%7B154%7D80030039004d9289ed5e2fcb4c013fff03f7000%20%5B16%3A47%3A10%5D&c=%7B154%7D80030039004d9289ed5d60e060013fffd11e000%20%5B16%3A47%3A19%5D&c=%7B154%7D80030039004d9289ed5d22e660013fff8196000%20%5B16%3A47%3A38%5D&c=%7B154%7D80030039004d9289ed5e1dcc3e013fffe7bc000%20%5B16%3A47%3A53%5D&c=%7B154%7D80030039004d9289ed5e31d43e013fff291a000%20%5B16%3A47%3A57%5D&c=%7B154%7D80030039004d9289ed5d29f64c013fffbcd4000%20%5B16%3A48%3A12%5D&c=%7B154%7D80030039004d9289ed5d2be94c013fff0b18000%20%5B16%3A48%3A21%5D&c=%7B154%7D80030039004d9289ed5d26e14c013fffd5d5000%20%5B16%3A48%3A26%5D&c=%7B154%7D80030039005d9289ed5d2be64c013fff112b000%20%5B16%3A48%3A31%5D&c=%7B154%7D80030039004d9289ed5d27f74c013fff768d000%20%5B16%3A48%3A35%5D&c=%7B154%7D80030039004d9289ed5d33ef4c013fffdcf7000%20%5B16%3A48%3A45%5D&c=%7B154%7D80030039003e9289ed5d47f44c013fff284d000%20%5B16%3A48%3A50%5D&c=%7B154%7D80030039004d9289ed5d16d347013ffff8d5000%20%5B16%3A48%3A59%5D&f=TYPE%3A8h%20ID%3A24h%20SOLAR%3F16d%20VOLTAGE%3F8d%20BITFIELD%3F8h%20TEMP%3F8d%20HUM%3F8d%20WIND%3F8d%20BEAR%3F8d%20GUST%3F8d%20UV%3F8d%20UNK%3F16h%20CHECKSUM%3F8h%20CRC%3F8h%208hh%0ATYPE%3A8h%20ID%3A24h%20SOLAR%3F16d%20VOLTAGE%3F8d%20BITFIELD%3F8d%20TEMP%3F8d%20HUM%3F8d%20WIND%3F8d%20BEAR%3F8d%20GUST%3F8d%20UV%3F8d%20UNK%3F16h%20CHECKSUM%3F8h%20CRC%3F8h%208hh%0ATYPE%3A8h%20ID%3A24h%20SOLAR%3F16d%20VOLTAGE%3F8d%20BITFIELD%3F8b%20TEMP%3F8d%20HUM%3F8d%20WIND%3F8d%20BEAR%3F8d%20GUST%3F8d%20UV%3F8d%20UNK%3F16h%20CHECKSUM%3F8h%20CRC%3F8h%208hh%0ATYPE%3A8h%20ID%3A24h%20SOLAR%3F16d%20VOLTAGE%3F8d%20UNK_BITS%3Fb%20GUST_BIT%3Fb%20BEAR_BIT%3Fb%20WIND_BIT%3Fb%20UNK_BIT%3F2b%20TEMP_BITS%3F2b%20TEMP%3F8d%20HUM%3F8d%20WIND%3F8d%20BEAR%3F8d%20GUST%3F8d%20UV%3F8d%20UNK%3F16h%20CHECKSUM%3F8h%20CRC%3F8h%208hh%0A&cf=ADD&cl=136&cw=8) to hopefully match what I read in your documentation.

@jemo07 do you have photos/documentation of how to connect a serial to a GW1000? Voltage levels, etc? I'd be curious to get the serial debug output to match my RF packets.

btw, I noticed there's some text strings in the WS80's firmware that suggest it may have a serial console of some kind, as well. Not sure if anybody has tried connecting to that.

fankyy, 2022-07-03:
Is there a rtl_433 version for debian buster that supports this driver? If I try to update / install with apt-get install rtl_433 (or rtl-433) or apt-get upgrade rtl-433 it says package not found

gdt, 2022-07-03:
> Is there a rtl_433 version for debian buster that supports this driver? If I try to update / install with apt-get install rtl_433 (or rtl-433) or apt-get upgrade rtl-433 it says package not found. Or can I update the drivers manually on the already installed (and probably over 7 months old) rtl_433?

Questions about buster should be addressed to debian.   But the short answer is that if you want recent code, you can compile it yourself.

mpavlica, 2022-07-05:
Which FSK chip is used? I would like to make my own receiver, and I got WS80, so, if anyone have any GW or HP unit opened, I am interested in knowing which FSK chip is used. Thanks!

commit 8b559835 (2022-02-06): Add support for FineOffset WS80 weather station (closes #1965)