merbanan/rtl_433 #1973: RING Smart Transformer

mores, 2022-02-04:
I am able to capture this data using:
`rtl_433 -f 923.3M -X 'n=name,m=FSK_PCM,s=1,l=1,r=1024'`

Attached find the output for Lights On, and Lights Off 

[off.txt](https://github.com/merbanan/rtl_433/files/8005363/off.txt)

[on.txt](https://github.com/merbanan/rtl_433/files/8005364/on.txt)

What is the next step in getting this added as a known device ?

zuckschwerdt, 2022-02-04:
Those values are not plausible, 1 µs is not likely. Grab a sample and upload as zip.
https://triq.org/rtl_433/ANALYZE.html

mores, 2022-02-04:
zip files from
rtl_433 -f 923.3M -S unknown
I turned on and off about 5 seconds between.
[data.zip](https://github.com/merbanan/rtl_433/files/8005823/data.zip)

zuckschwerdt, 2022-02-05:
That looks like LoRa (up and down chirps over a wide band). We don't have support for that.
The samples might be interesting to see if we can add a demod for LoRa. But the bandwidth requirements are not well suited to simple SDR I guess.

merbanan, 2022-02-05:
This is LoRa and not supported (yet). The payload is most likely aes encrypted so it is quite the effort to be able to decode the protocol. If we get LoRa supported in the future this can be reopened.

closed 2022-02-05 by merbanan

merbanan, 2022-02-05:
@zuckschwerdt The min max fsk demodulation logic works fine with this type of signal. But the actual symbol recovery logic is tricky.

https://user-images.githubusercontent.com/124883/108997340-c44a1880-769f-11eb-9af7-56dbd792a23a.png

When a symbol transitions through a chirp frequency restart I get spikes in the symbol levels. I need some kind of cyclical vector space for my calculation.

zuckschwerdt, 2022-02-05:
It might be interesting to detect LoRa. But don't think we can match the robustness of dedicated chips. Ideally just some part of every chirp should be enough and that's a very different kind of "demod" (ignores amplitude). And then there are cheap ESP+LoRa boards widely available. (I have e.g., and would recommend, dedicated chips for EnOcean, ZigBee that are reliable at -90 dB.)

merbanan, 2022-02-05:
Yeah, we will never be able to match dedicated chips with the current code. But it is not impossible we will be able to support sensors in close proximity.

I got some help with the symbol recovery and might have a working method.