merbanan/rtl_433 #1975: Support for ITRON Support for ITRON Water meter (m_bus device)

heeplr, 2022-02-05:
This meter isn't supported by rtl_433 and I've [uploaded sample datagrams to the wmbusmeters project](https://github.com/weetmuts/wmbusmeters/issues/468) which might be interesting to extend the current m_bus driver.
I could also provide further information if needed.

merbanan, 2022-02-05:
We need lots of signal samples to be able to add support.

heeplr, 2022-02-05:
Even though the signal is already decoded by m_bus.c? 
json output of rtl_433: [itron.txt](https://github.com/merbanan/rtl_433/files/8008739/itron.txt)

gdt, 2023-09-29:
@heeplr What's the status of this?  Plans?  Are you going to work on a PR?

heeplr, 2023-09-29:
no, i'm using wmbusmeters & jq to decode the packets which works fine for me:
```
rtl_433 -d ":00000868" -f 868.95M -s 1.2M -g 40.0 -C si -M level -F json | grep <id> | stdbuf -oL jq --raw-output .data | wmbusmeters --format=json stdin:hex water auto <id> NOKEY
```

closed 2023-09-29 by gdt

Anas92230, 2024-05-22:
> no, i'm using wmbusmeters & jq to decode the packets which works fine for me:
> 
> ```
> # id for example 12345678
> rtl_433 -d ":00000868" -f 868.95M -s 1.2M -g 40.0 -C si -M level -F json | grep 12345678 | jq --raw-output .data | wmbusmeters --format=json stdin:hex water auto 12345678 NOKEY
> ```

Hello @heeplr ,
You have a G3-ITR?
Can you explain me what is the id? and how did you get it?
thanks for your help.

heeplr, 2024-05-22:
@Anas92230 it's the device "id" field in the output and iirc also printed on the device. You can just run ```rtl_433 -d ":00000868" -f 868.95M -s 1.2M -g 40.0 -C si -M level -F json``` alone to see which devices you receive and choose the id you like to decode.

famillenounoux, 2024-05-22:
> @Anas92230 it's the device "id" field in the output and iirc also printed on the device. You can just run `rtl_433 -d ":00000868" -f 868.95M -s 1.2M -g 40.0 -C si -M level -F json` alone to see which devices you receive and choose the id you like to decode.
With my watermeter G3 ITR the command rtl_433 -f 868.95M -s 1.2M -g 40.0 -C si -M level -F json got nothing

heeplr, 2024-05-22:
Are you sure you are in range and it's transmitting on 868.95MHz?
You could check what it's actually transmitting with a program like gqrx or baudline.

Anas92230, 2024-05-22:
Ok i need to add - at the end because i receive an error
i have only 868MHz wirte on watermeter and no datasheet :-(
it's very difficult to find the right frequency because you have lot of device on this band :-(.
I let it turn :-)

Anas92230, 2024-05-22:
result for only:
rtl_433 -d ":00000868" -f 868.95M -s 1.2M -g 40.0 -C si -M level -F json
rtl_433 version 21.05 (2021-05-09) inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 169 out of 199 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 ]

lo92fr, 2025-05-04:
Hello guys,

Come to this post as I was searching information for my Itron Cyble G3 ITR watermeter.
Can you confirm which version you have : it seems that there is LoRaWAN, Sigfox, and Homerider.
Mine seems to be a Sigfox version, so I think it's not applicable to it ? And don't know if there is anyway do decode the sigfox signal ?

[image]

piars777, 2025-11-23:
Hello, i have the same meter but with rtl_433 i catch noting. Does anyone have successfully catch something?