merbanan/rtl_433 #1981: FYI - Error building latest on ancient Raspbian Jessie -- can't find -lpthreads

rct, 2022-02-14:
Just an FYI - I have an ancient RPI 1B+ that is still running Jessie, building rtl_433 used to work, now fails due to `cannot find -lpthreads` when trying to build the pthreads_create test.  The last time I updated, 21.05 code from 2021-08-26 built without error..

There is a `/usr/lib/arm-linux-gnueabihf/libpthread.so` and `/usr/lib/arm-linux-gnueabihf/libpthread.a`, but ld through CMake doesn't seem to find it for some reason.

I realize this is a self inflicted problem and I should rebuild^Wreplace that system. Posting this in case it helps anyone else

```
Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/rct/proj/rtlsdr/merbanan/rtl_433/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec865147291/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec865147291.dir/build.make CMakeFiles/cmTryCompileExec865147291.dir/build
make[1]: Entering directory '/home/rct/proj/rtlsdr/merbanan/rtl_433/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/rct/proj/rtlsdr/merbanan/rtl_433/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec865147291.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTryCompileExec865147291.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.0/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec865147291
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec865147291.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTryCompileExec865147291.dir/CheckFunctionExists.c.o  -o cmTryCompileExec865147291 -rdynamic -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTryCompileExec865147291.dir/build.make:88: recipe for target 'cmTryCompileExec865147291' failed

```

zuckschwerdt, 2022-02-14:
Yes, that should be `-pthread`, not `-lpthread`. I guess it comes from `${CMAKE_THREAD_LIBS_INIT}`, needs some investigation why Cmake thinks pthreads needs linking as a lib?

zuckschwerdt, 2022-02-14:
Oh wait, that's not actually our build failing but the Cmake threads test from `find_package(Threads)` which is a Cmake built-in module...

gdt, 2023-09-30:
Still valid?

gdt, 2024-06-23:
A bug from 2022 about raspbian that was ancient then....  Please feel free to start over if git master rtl_433  fails on any OS receiving maintainance, or file a PR to make it work on any OS regardless of upstream status.

closed 2024-06-23 by gdt