merbanan/rtl_433 #1982: Help decoding HS1527-based magnetic door sensor (SONOFF DW2-RF)

wipeout666, 2022-02-16:
I'm having trouble understanding the signal coming from my magnetic door sensor SONOFF DW2-RF. Since it is HS1527-based, I am expecting a OOK_PWM modulation that RTL_433 would easily recognize, maybe with a flex decoder. However, upon a closer look with universal radio hacker, I myself cannot recognize anything coming out of that RF device.

Top signal is the capture signal from the SONOFF door sensor
Bottom signal is a signal from another device, something I recognize as ASK / OOK_PWM. I was expecting something similar to this signal.

[image]

I did not include a .cu8 file because running RTL_433 with -A isn't picking up the signal from the door sensor. If it's helpful to record one anyway, let me know and I'll do it.

Can you help decoding this with RTL_433?

Thanks !

zuckschwerdt, 2022-02-16:
Is that the waveform of the signals? If that's I/Q then you should offset the carrier from DC.
This is a single HS1527 packet and a zoom in (logic level, I, Q, amplitude):

[image]

[image]

wipeout666, 2022-02-17:
Yes this is the raw waveform, as capture when tuned to 433.92MHz.

Can you elaborate more on the DC offset? How is that applicable to RTL_433?

zuckschwerdt, 2022-02-17:
It looks like the carrier is nearly at DC (0 Hz), you don't get a wave but only parts of slopes. You should offset this 10 kHz or more so you can better see the carrier (the green+yellow above).

wipeout666, 2022-02-18:
Yup, that was part of the problem, good catch!. I've re-captured with a 20kHz offset:

[image]

[image]

Capture: https://www.dropbox.com/s/v0ifbe7o2itkyys/RTL-SDR-20220217_205658-433_9MHz-1MSps-1MHz.zip?dl=1

I'm still not able to build a flex decoder in RTL_433. I think the following should work, but it does not: -R 0 -X "n=doorsensor,m=OOK_PWM,s=250,l=750,r=900,bits=25" -f 433.9M

The signal seems to slowly "die" at the end, loosing amplitude until zero. This seems to confuse RTL_433; how can I get around that?

zuckschwerdt, 2022-02-18:
The fading at the end is expected for a HS 1527 encoder and won't be a problem.

zuckschwerdt, 2022-02-18:
The sample isn't like anything we expect to encounter live (no noise floor, prominent DC). But it exposed a logic flaw in the pulse detector. We reset the detector on spurious short pulses, which we likely shouldn't (only noticeable if there is already pulse data from a previous buffer that had no marked frame end). I'll fix that shortly.

wipeout666, 2022-02-19:
"_the sample isn't like anything we expect to encounter live_" - But this is exactly as I've captured. OK, the transmitter is close to the antenna, but I did not create a perfect environment. Just the environment it's going to work in.

"_We reset the detector on spurious short pulses_" - Where do you see that ?

Just curious; good if it can be fixed! Otherwise, what can I try, now?

zuckschwerdt, 2022-02-19:
The capture isn't post-processed? What SDR did you capture with? Why is there no noise floor?

You can set [PD_MIN_PULSE_SAMPLES](https://github.com/merbanan/rtl_433/blob/master/include/pulse_detect.h#L21) to a lower value, like 1 and this will then work:
`rtl_433 -X 'n=name,m=OOK_PWM,s=125,l=380,g=500,r=1000'`

wipeout666, 2022-02-19:
No, it is not post processed. It was captured with universal radio hacker, so unless the record tool is post processing without my knowledge...

The radio is RTL2832U. I'll try your suggestion in a few hours, when I get back home

zuckschwerdt, 2022-02-19:
If that was an RTLSDR then the sample is heavily post-processed. Capture with `-S unknown` or perhaps something like `-w myfile.cu8 -T 30` and compare in https://triq.org/pdv/

wipeout666, 2022-02-21:
The red arrows are the magnetic door sensor event's (each time two events each ~1s appart, then 3 quick ones, then two events ~1s appart). Using "-w", the signal, apparently, is clipping with auto gain:

[image]
Ref: clipping.cu8

I manually set the gain to 40dB, which yields:
[image]

https://www.dropbox.com/s/mjp4f0tkzl8w75n/DoorSensorCapture.zip?dl=1

Even with gain set at 40dB, I encouter the same struggles ... Any clues in this?

wipeout666, 2022-02-21:
Setting PD_MIN_PULSE_SAMPLES to 1 does work with the following:

 rtl_433 -f 433.91M -R 0 -X "n=rfdoor,m=OOK_PWM,s=250,l=750,r=7850,g=760,repeats>=2,match={25}c9c47e8"

Is the change to PD_MIN_PULSE_SAMPLES going to adversely impact demodulation of my other stuff?

zuckschwerdt, 2022-02-21:
> Is the change to PD_MIN_PULSE_SAMPLES going to adversely impact demodulation of my other stuff?

Yes, it will impact FSK very much.

wipeout666, 2022-02-22:
Thanks for confirming; not an issue for me at the moment. Is there a mecanism to know when the "_logic flaw in the pulse detector_" will be fixed? An issue to track? I'll need to keep PD_MIN_PULSE_SAMPLES at 1 until then.

Thanks for your help, I couldn't have done it without your support!

zuckschwerdt, 2022-02-22:
I'm working on a fix, but it's a critical piece which could cause big regressions, so hard to change.

wipeout666, 2022-02-24:
Can PD_MIN_PULSE_SAMPLES be set in the .conf file instead of recompiling pulse_detect.h?

zuckschwerdt, 2022-02-24:
No, sorry. It's not meant to be a configurable value but just a sanity check for the pulse detector.

zuckschwerdt, 2022-02-28:
It should be fixed now, you will get bitbuffers for OOK even if the pulses end with a spurious short pulse.

closed 2022-02-28 by zuckschwerdt

sheilbronn, 2022-05-12:
Hello, I have a Sonoff DW2-RF, too. However, it seems the signal is not being picked up in the my current version of rtl_433 (21.12-95-gddbc8420 branch master at 202204012124), but only in the Sonoff 433MhZ Bridge running in parallel. 
Could somebody **please confirm that the DW2 should be working in the current version**?! Thx!
Is there anything I could do to support debugging?

wipeout666, 2022-05-13:
The fix was introduced in this commit:
[image]

So my guess is that you do not have the fix in this version; you should upgrade.

Also note that you need to change the tuner's frequency to be a little offset from 433.92MHz, such as: -f 433.91M

Turns out my SONOFF device is exactly tuned at 433.92MHz, unlike my other 433MHz devices. Being tuned at exactly the carrier's frequency doesn't work, as everything is shifted to DC, which is not decodable.

Hope that helps

commit ca4df453 (2022-02-28): Fix pulse reset on spurious pulses (closes #1982)