merbanan/rtl_433 #1984: Marlec iBoost not working. Device 188

whiteduck22, 2022-02-17:
Hi there. I see under #1739 problems with this device but not a resolution but since that thread went quiet you have added native support with device 188. For me with latest build and command options rfl_433 -f 868.3M it doesn't work. Running -A returns data every few seconds. Happy to share debug info but wondered if others have found a resolution such as a different sample rate perhaps. Thank you.

That-Dude, 2022-08-13:
Hi @whiteduck22 I'm in a similar position. I've just emailed Marlec support to ask them for the correct encoding of their devices - I'll report back with their response.

zuckschwerdt, 2022-08-13:
You can always try to grab a sample (https://triq.org/rtl_433/ANALYZE.html) and upload as zip here, we will take a look why it won't decode.
But the payload is a mystery, hopefully you can get some info from the vendor.

We do know it's 20 µs (50kbit/s) FSK with `aaaaaa` preamble and `d391 d391` sync word, then there is a length byte, data (of length byte len), and a CRC-16 checksum (poly 0x8005 init 0xffff).

whiskerp, 2022-09-08:
I am getting more reliable capture and decoding with a pulse length of 10:
rtl_433 -R 0 -f 868.298M -X 'n=Marlec,m=FSK_PCM,s=10,l=10,g=1000,r=2000,preamble=aad391d391' -s 1000000

I changed the marlec_solar.c module locally to reflect this and it now CRCs OK too:

time : 2022-09-07 16:30:45
model : Marlec-Solar Raw data : e3940100000000000000040d1929200f1d1c110900000000000000000000140c01001e10211a120700000000 Integrity : CRC

time : 2022-09-07 16:30:55
model : Marlec-Solar Raw data : e39401000000000000000000030e192b25122522170b0000000000000000122821181c2d23112621170b0000 Integrity : CRC

Code change:

r_device marlec_solar = {
.name = "Marlec Solar iBoost+ sensors",
.modulation = FSK_PULSE_PCM,
.short_width = 10,
.long_width = 10,
.gap_limit = 1000,
.reset_limit = 2000,
.decode_fn = &marlec_solar_decode,
.fields = output_fields,
};

Much shorter gap and reset limits seems to chop up the bitstream.

zuckschwerdt, 2022-09-08:
That does not sound right. The PCM slicer will auto-adjust to the actual bitrate by looking at the `aaaa...` preamble.
Run with verbose device output, e.g. `-vvv -R 0 -X ...` and look for "Exact bit width (in us) is"

Can you grab a capture and upload here as zip?

whiskerp, 2022-09-08:
# rtl_433 -f 868.297M -vvv -R 0 -X 'n=Marlec,m=FSK_PCM,s=10,l=10,g=1000,r=2000,preamble=aaaad391d391' -s 1000000 2>&1 | grep Exact

Exact sample rate is: 1000000.026491 Hz
General purpose decoder 'Marlec': Exact bit width (in us) is 9.22 vs 10.00, 23 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.43 vs 10.00, 23 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 8.91 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.17 vs 10.00, 24 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.17 vs 10.00, 23 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.48 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.42 vs 10.00, 24 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.48 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.10 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.12 vs 10.00, 26 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.41 vs 10.00, 27 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.48 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.09 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 10.05 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.64 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 10.00 vs 10.00, 27 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.45 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.14 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.14 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.67 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 10.00 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.05 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.41 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.55 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.39 vs 10.00, 23 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.67 vs 10.00, 24 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.22 vs 10.00, 27 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.59 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.32 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.60 vs 10.00, 25 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.58 vs 10.00, 24 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.35 vs 10.00, 23 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 8.76 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.70 vs 10.00, 23 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.00 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.43 vs 10.00, 23 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.17 vs 10.00, 24 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.55 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.62 vs 10.00, 24 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 10.00 vs 10.00, 23 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.16 vs 10.00, 25 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.46 vs 10.00, 26 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.48 vs 10.00, 25 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.50 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.23 vs 10.00, 22 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 9.81 vs 10.00, 21 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 10.05 vs 10.00, 21 bit measured
Exact bit width (in us) is 10.07 vs 10.00, 28 bit preamble
General purpose decoder 'Marlec': Exact bit width (in us) is 10.01 vs 10.00, 68 bit measured
General purpose decoder 'Marlec': Exact bit width (in us) is 10.11 vs 10.00, 101 bit measured
Exact bit width (in us) is 10.04 vs 10.00, 28 bit preamble
[sample.tar.gz](https://github.com/merbanan/rtl_433/files/9527574/sample.tar.gz)

zuckschwerdt, 2022-09-08:
Thanks!
How strange. E.g. your sample `g028_868.297M_1000k.cu8` is 10 µs (100kbps), like you say. But the older files captured, e.g. `g008_868.3M_1000k.cu8` from #1739, are really 20 µs (50 kbps).

We know there are different devices involved (transceivers), and now it's different rates too?

gdt, 2023-10-02:
This looks like a help request about this decoder, but I'm not sure where it is, and if any work needs to be done, and if so where.  Please add a comment with status and next steps.

closed 2024-06-23 by gdt