merbanan/rtl_433 #1986: duplicate key 'light_lux'

johannesjaai, 2022-02-19:
key light_lux is now duplicate in

https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py

```
    },
    "light_lux": {
        "device_type": "sensor",
        "object_suffix": "lux",
        "config": {
            "name": "Outside Luminancee",
            "unit_of_measurement": "lux",
            "value_template": "{{ value|int }}",
            "state_class": "measurement"
        }
    },
    "light_lux": {
        "device_type": "sensor",
        "object_suffix": "lux",
        "config": {
            "name": "Outside Luminancee",
            "unit_of_measurement": "lux",
            "value_template": "{{ value|int }}",
            "state_class": "measurement"
        }
    },
```
introduced with a combination of commits from issues #1930 and #1548

zuckschwerdt, 2022-02-20:
Thanks! Do you want to PR a fix?

joanwa, 2022-02-20:
sure, just wondering if `lux` is still required for some backwards/legacy compatibility... so either 

1. kick the duplicate 
2. renaming one `light_lux` back to `lux`

zuckschwerdt, 2022-02-20:
That was just a bad combination of merges on my part. There is still an about even split of "light_lux" and "lux" fields in current decoders. We should rename it back.
For the future we should formalize more of the weather device keys and possibly migrate.

joanwa, 2022-02-20:
fixed with https://github.com/merbanan/rtl_433/pull/1989/commits/88b7a2474783016195021a7ccd9ec6f3bc4b1f1c

closed 2022-02-20 by zuckschwerdt