merbanan/rtl_433 #1987: PlutoSDR not working in Windows 10

robcazzaro, 2022-02-20:
I'm pretty sure I did something wrong, but I can't make rtl_433 work with my PlutoSDR.

I installed PothosSDR version PothosSDR-2021.05.01-vc16-x64, enabling the experimental plutosdr support. I did install all the dependencies (I think). I installed rtl_433 version rtl_433-win-msvc-x64-21.12 by copying the files in a work directory (the static binary of the 64 bit windows doesn't seem to support SoapySDR, even if there is a SoapySDR dll in the ZIP file, so I used the MS VC version)

Here's more info
```
C:\Work\rtl_433>soapysdrutil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

←[1m←[31m[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail←[0m
←[1m←[31m[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.←[0m
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.2; Boost_107500; UHD_4.0.0.HEAD-0-g90ce6062
WARNING: Unknown parameter '0' in <context>
WARNING: Unknown parameter '23' in <context>
WARNING: Unknown parameter 'v0.23' in <context>
←[1m←[31m[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed←[0m
Found device 0
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = Default Device

Found device 1
  default_input = False
  default_output = False
  device_id = 3
  driver = audio
  label = Line In (MX3)

Found device 2
  default_input = False
  default_output = False
  device_id = 4
  driver = audio
  label = Microphone Array (Intel« Smart Sound Technology for Digital Microphones)

Found device 3
  device = plutosdr
  driver = plutosdr
  uri = usb:1.9.5

C:\Work\rtl_433>SoapySDRUtil --probe="driver=plutosdr,uri=usb:1.12.5"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=plutosdr,uri=usb:1.12.5
WARNING: Unknown parameter '0' in <context>
WARNING: Unknown parameter '23' in <context>
WARNING: Unknown parameter 'v0.23' in <context>
WARNING: Unknown parameter '0' in <context>
WARNING: Unknown parameter '23' in <context>
WARNING: Unknown parameter 'v0.23' in <context>

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=PlutoSDR
  hardware=ADALM-PLUTO
  ad9361-phy,model=ad9364
  ad9361-phy,xo_correction=39999868
  backend_version=0.23 (git tag: v0.23  )
  fw_version=v0.34
  hw_model=Analog Devices PlutoSDR Rev.C (Z7010-AD9364)
  hw_model_variant=1
  hw_serial=1044734c9605000df9ff0500deb64f2518
  library_version=0.19 (git tag: 5f5af2e)
[... 163 lines cut ...]

C:\Work\rtl_433>rtl_433-rtlsdr-soapysdr -d "driver=plutosdr,hostname=192.168.2.1"  -v
rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Work\rtl_433\rtl_433.conf"...
Trying conf file at "C:\Users\Roberto\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
[...]
Registering protocol [207] "SmartFire Proflame 2 remote control"
Registered 176 out of 207 device decoding protocols
Failed to open sdr device matching 'driver=plutosdr,hostname=192.168.2.1'

C:\Work\rtl_433>rtl_433-rtlsdr-soapysdr -d "driver=plutosdr,uri=usb:1.12.5"  -v
rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Work\rtl_433\rtl_433.conf"...
Trying conf file at "C:\Users\Roberto\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
[...]
Registering protocol [207] "SmartFire Proflame 2 remote control"
Registered 176 out of 207 device decoding protocols
Failed to open sdr device matching 'driver=plutosdr,uri=usb:1.12.5'.
```

Programs like CubicSDR (part of the Pothos distribution) work just fine using SoapySDR, so I know that PlutoSDR gets recognized, at least partially...

zuckschwerdt, 2022-02-20:
Could be that the SoapySDR lib being loaded is configured for a different module path, so no drivers are available.
Maybe locate the DLL from the Pothos install and replace the one alongside rtl_433. Or just delete the one DLL from the rtl_433 download, I guess the Pothos install should be found then.
We perhaps need some debug output similar to `SoapySDRUtil --info`.

robcazzaro, 2022-02-20:
Thanks for the super-fast reply. Here's the output from SoapySDRUtil --info

```
C:\Work\rtl_433>SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-PothosSDR-2021.05.01-vc16-x64
API Version: v0.8.0
ABI Version: v0.8
Install root: C:\Program Files\PothosSDR
Search path:  C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspyhfSupport.dll (0.1.1-37445c9)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspySupport.dll   (0.1.2-10d697b)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/audioSupport.dll    (0.1.1-91080cb)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/bladeRFSupport.dll  (0.4.1-70505a5)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/HackRFSupport.dll   (0.3.3-5152ab9)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/IrisSupport.dll     (2020.02.0.1-f100723)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/LMS7Support.dll     (20.10.0-a45e482d)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/miriSupport.dll     (0.2.6-585c012)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/netSDRSupport.dll   (0.1.0-dc4e579)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/osmosdrSupport.dll  (0.2.6-585c012)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/PlutoSDRSupport.dll (0.2.1-ca40ea9)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/RedPitaya.dll       (0.1.1-3d576f8)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/remoteSupport.dll   (0.6.0-c09b2f1)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/rtlsdrSupport.dll   (0.3.2-53ee8f4)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/sdrPlaySupport.dll  (0.3.0-e6fdb71)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll      (0.4.1-9a738c3)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/volkConverters.dll  (37d639d)
Available factories... airspy, airspyhf, audio, bladerf, hackrf, iris, lime, miri, netsdr, osmosdr, plutosdr, redpitaya, remote, rtlsdr, sdrplay, uhd
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, F64, S16, S32, S8, U16, U8]
 -   F64 -> [F32]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [F32, S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

There is no SoapySDR installation in the rtl_433 download, so the only working copy is the one from the Pothsoware install. I could not find installation instructions for rtl_433 and PlutoSDR, so from my searches I figured out that I needed a Windows build of rtl_433 compiled with soapysdr support plus install Pothosware. Which is what I did

I'm not set up to build my own version of rtl_433 or SoapySDR... 

Based on your comment, I tried deleting the SoapySDR.dll in the rtl_433 folder, and that seemed to work (the SoapySDR.dll in the rtl_433 is identical to the one in the Pothosware directory). I can now see the devices I expected to see, albeit it's super slow in starting: between "registered x out of y decoding protocols" and  "[INFO] Using format CS16." it takes a good 20 seconds. But also a SoapySDRUtil --probe="driver=plutosdr" command seems to take 15 seconds

Just for my information, what are the right steps to install rtl_433 with PlutoSDR support in Windows 10? I really don't need Pothosware, and I installed it only for the rtl_433 PlutoSDR support

zuckschwerdt, 2022-02-20:
Those are the correct steps. Install Pothos for all the driver modules, then just run rtl_433 from any directory without installing. Good to know that we can't use a SoapySDR.dll bundled with the rtl_433.exe

robcazzaro, 2022-02-20:
Perfect, thanks. Closing this issue

zuckschwerdt, 2022-02-20:
The delay might be from timeout in enumeration from (one of) the unused drivers. You might want to try removing the unused ones. Could be the sdrPlaySupport.dll. Perhaps remove them one by one and thus check with trial-and-error who's causing the delay.

robcazzaro, 2022-02-20:
I actually already removed all the drivers, leaving only plutoSDR, but it's still delayed. Not a big issue for now, and  will follow up on that with the SoapySDR folks, thanks for all the help!

```
C:\Work\SDR\rtl_433>soapysdrutil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-PothosSDR-2021.05.01-vc16-x64
API Version: v0.8.0
ABI Version: v0.8
Install root: C:\Program Files\PothosSDR
Search path:  C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/PlutoSDRSupport.dll (0.2.1-ca40ea9)
Available factories... plutosdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

closed 2022-02-20 by robcazzaro