merbanan/rtl_433 #1992: Support for Arad and Master meter water utility meter

avicarmeli, 2022-02-22:
There is some success as written in [that](https://www.reddit.com/r/RTLSDR/comments/k9dmz9/master_meter_water_utility_meter/) tread.
I had no luck: Using rtl_433 -f 916.0M -s 2400000 -g 280 -a 4 -X "n=ARAD,modulation=FSK_MC_ZEROBIT,s=8.4,l=8.4,r=3000,invert"
I get nothing reasonable.
The data I get is changing in bit length and also in payload.
I have found that [data](https://arad.co.il/app/uploads/Interpreter_Eng.pdf) about my meter.
I was sampling raw data with:
rtl_433 -f 916.2M -g25 -s1000k -S all - with no antenna about 10cm away from the meter.
rtl_433 -f 916.2M -g25 -s1024k -S all - with antenna about 50cm away from the meter.
The data can be found [here](https://drive.google.com/drive/folders/1o2rRn8eZM9zINk7zcZ138HbxPmDh7y2G?usp=sharing). (Let me know if there is a problem accessing it).
In the folder you will also find a picture of the meter about the time of sampling the data, as well as file named out_916.2M_1000k.txt containing the console data when running rtl_433 -f 916.2M -s 1000k -g 25 -X "n=ARAD,modulation=FSK_MC_ZEROBIT,s=8.4,l=8.4,r=3000,invert"

merbanan, 2022-02-22:
I looked at 2 samples and the signal is really nice and clear.

rtl_433 -Y minmax -r ../signals/ARAD/*.cu8 -X "n=ARAD,modulation=FSK_MC_ZEROBIT,s=8.4,l=8.4,r=3000,invert,preamble=0000"

Now all data look the same and it is time to start logging so we can see if the data is scrambled and if a pattern emerge.

`3e690aec7ac84bfa1c9073003a060000056731e3c6c0f8`

In the middle is a space with lots of 0's maybe this is the area for the counter.

If I take the id, 09444602 go to hex -> 901CFA and then reverse it FA1C90 I can then find the id in the bit stream:
`3e690aec7ac84b [fa 1c 90] 73003a060000056731e3c6c0f8`

Now you have to collect a more messages and then you can with the help of bitbench figure out the message structure and from there you can create a decoder.

avicarmeli, 2022-02-22:
Great. Will do.

avicarmeli, 2022-03-06:
OK I got [that](https://triq.net/bitbench#c=000000003e690aec7ac84bfa1c9073005206000005d011128ae8f8&f=56x%20%3C24dc%208x%20%3C24d%20hhhhhhhhhhhhhhhh%20%20&a=Preamble&m=00000000&cw=4) from bitbucket.
What's next? How do I write a decoder? Is there a template I can follow?
Another issue: I can't seem to decode the last 8 bytes of the payload. I suspect it is some kind of checksum or CRC as it changes only when there is a change in the counter reading. I was trying to match it with some online CRC and Checksum calculators but with no luck. I believe the last byte is a suffix as it is unchanged for all payloads.
[Here](https://docs.google.com/spreadsheets/d/1fr3CjJ80GyLRn6SUTFUwmuDK4FnvCTCb/edit?usp=sharing&ouid=111642821344170719731&rtpof=true&sd=true) you can find about 24 hours recording of my neighborhood's water meters.

merbanan, 2022-03-06:
I had a look and it looks like only 4.5 bytes of crc/checksum. And I'm not sure that is what it is either.

```
000000003e690aec7ac86b68e3de73403af6040008 [cad99888f|0|] f8
000000003e690aec7ac86b68e3de73403af7040008 [6a9c3fa77|0|] f8
000000003e690aec7ac86b68e3de73403bf1040008 [eaa92c71d|8|] f8
000000003e690aec7ac86b68e3de73403cf1040008 [aeb0a251a|8|] f8
000000003e690aec7ac86b68e3de73403cf2040008 [4e7e84041|8|] f8
```
The last position || in crc area [] is only 8 or 0. Most likely we only have 4 bytes. A full brute force search should take 1-2 hours. Try with reveng.

There is a template decoder that you can use to base the decoder of. Or just base it from some other decoder.

avicarmeli, 2022-03-10:
Running:

> reveng -w 32 -s 3E690AEC7AC84BDAB18C7300390E000005BFF1DB9F 3E690AEC7AC84B13FE8B7300100500000507CBFECD 3E690AEC7AC84BDAB18C73003B0E0000055C91EB1C 3E690AEC7AC84B3EC68C7300720E000005C6E1CCD5

I get immediately :

>  no models found

So if I am not doing anything wrong, Reveng doesn't seem to find anything with brute force.
Any other idea?

merbanan, 2022-03-10:
Drop the last nibble and try again. 5C 6E 1C CD 5 <- this is 9 digits. For a crc-32 only 8 should be output. So try again and if it not working we need to call @zuckschwerdt for some advice. You could prepare a long list of unique messages and add them to bitbench. Maybe we need to figure out the other parameters first.

avicarmeli, 2022-03-10:
I already tried many options include dropping that nibble. Actually the above is after dropped the preamble and the last nibble as in **5**C 6E 1C CD 5 , the first 5 is fixed for all massage's.

merbanan, 2022-03-10:
Ok, can you add a set of unique codes to a bitbench and post a link? Does the counter match your meter reading? If so you can at least start implementing a decoder. The crc is not really needed for a decoder but it is nice to have. The signal integrity is quite high with manchester coding used.

zuckschwerdt, 2022-03-10:
The data doesn't look right. Note that the MC decode will just skip problems. Best convert all cu8 to ook (`rtl_433 -Y minmax -w file.ook file.cu8`), then drop on https://triq.org/pdv/ and inspect. For good files go to raw PCM (`rtl_433 -Y minmax -X "n=ARAD,modulation=FSK_PCM,s=8,l=8,r=30" file.cu8`).
Collect the raw PCM to a BitBench and check again, then MC decode there.

Check if it really is IEEE and not Thomas, and confirm an alignment. Only then will a CRC test work.

avicarmeli, 2022-03-10:
On it.

zuckschwerdt, 2022-03-10:
An [example BitBench](https://triq.net/bitbench#c=%7B463%7D956aaaaaaaaaaaaaaaaaaaaaa955a59a6a999596a559969aa6995559aa569a6aa5696aaaa959aaa5aaaaaaaaaaa665a5696a55a956a596aa9554&f=8h8h%20&a=Preamble&m=aaa9&d=MC) with the code from g001_916.2M_1024k.cu8.

avicarmeli, 2022-03-10:
@merbanan The serial number and the counter readings are stable for either my meter as well as another 5 meters in my neighborhood.

avicarmeli, 2022-03-10:
@zuckschwerdt how would I tell if the OOK file is OK. What should I look for?

zuckschwerdt, 2022-03-10:
There must only be pulses and gaps of length 1x and 2x (ignore if there is some leading garbage).
You can try the MC slicer in pdv and it will flag bad pulse/gap widths.

avicarmeli, 2022-03-11:
After running all the cu8 files I had through the process @zuckschwerdt recommended, I conclude:
1. The Manchester code is definitely IEEE.
2.  the majority of the  massage's bits are decoded correctly  with 

>  -X "n=ARAD,modulation=FSK_MC_ZEROBIT,s=8.4,l=8.4,r=3000,invert,preamble=0000"

flex decoder with only the last nibble being different.
While the flex decoder place **8** or **C** at the last nibble for all transmissions, Bitbucket place either **1**  or **3** which seems reasonable for me. When examining the last bit in the PDV I can see that the timing is a bit off (about 2 us longer than the rest) but I am not sure what does that mean.
[image]

I will collect some more recordings and analyze them in the same manner to confirm my conclutions.

merbanan, 2022-03-16:
I think it is just a stray last bit.

avicarmeli, 2022-03-19:
@merbanan How do you think the settings of the flex decoder, as well as sampling rate and frequency should be changed when trying to decode the US versions of the meter where Direct Sequence (DS) and Wide Band (WB) are used (902 - 928MHz)?

merbanan, 2022-03-20:
I need a signal sample to say anything concrete but it might just be a change of frequency,

assafms, 2023-02-01:
Hi , I am the OP in the original thread about using SDR for the ARAD watermeter. just adding that i found the hardware used by arad and created a dedicated hardware circuit (arduino + xl4432) for about 6$ to decode the signal without SDR.  the details are [here](https://www.reddit.com/r/RTLSDR/comments/10nri4r/dialog3g_water_meter_reading_now_with_dedicated/)
this might help you if still needed.

drorya1, 2023-08-28:
@avicarmeli  did you manage to decode it ?

avicarmeli, 2023-08-28:
@drorya1 
I ended up just compiling very limited version of the driver since I can only read the water consumption and could not decode the rest of the massage.
You can see my code [here](https://github.com/avicarmeli/rtl_433/tree/master).
One more thing is that I might made the preamble longer in order to read only my meter , so in that [file](https://github.com/avicarmeli/rtl_433/blob/master/src/devices/arad_ms_meter.c) you might want to change line 48 to shorter preamble or to one which comply to your meter's serial number.
It works fine for me to read the water counter.

If need any specific help don't hesitate to write me.

avicarmeli, 2023-08-28:
@assafms 
Were you able to decode anything more than the counter reading?
Also I would like to help to write a driver for Tasmota to read the xl4432 data.

assafms, 2023-08-28:
I didn't try to decode the rest of the message. but I did write a custom component for ESPHome that connects to the xl4432, configures it, and sends the information to Home assistance.  I can send you the code if needed. not the best-written code :) but it works.

drorya1, 2023-08-28:
@avicarmeli @assafms 
Thanks a lot 
Avi - I will review what you shared 
What do you use ? SDR ? or Assaf solution using SI4432 and ESP32 board ?
I have insane water consumption and need to better understand what is the root cause .
I really dont understand why they do not expose it ? ( I can only assume that they are concerned with load on their APIs etc)

assafms, 2023-08-28:
you need to remember that in some places the meter broadcast is not received continually by a receiver. they have cars driving around once every few days/weeks to collect the latest readings.  In other places, they have "concentrators" that receive the RF broadcast and relay using a cellular connection.  Also, they charge Cities for a website to provide the data for the public and some opt out due to the extra cost.

drorya1, 2023-08-28:
> you need to remember that in some places the meter broadcast is not received continually by a receiver. they have cars driving around once every few days/weeks to collect the latest readings. In [...]

You have a valid point - yet I see the reading on a daily basis on their website 
I assume they read it constantly in my city , yet they do not expose the reading to their users .( So I need either to either deploy a hall effect water reading solution inline to the main water pipe , or preferably decode the info and read it )
Did you end up using SDR or are you using the SI4442 and ESP32 at this time ?
Can you share your code and solution ?

assafms, 2023-08-28:
I designed a small PCB for an ESP32 and the xl4432 , and created a custom component for ESPHome. Works great. about 99.9999 of the messages are decoded correctly and no need to maintain an SDR or Rasp. I can share both the board design and the code. will upload to GitHub later. 

board: 
[image]

and the connection to the home assistant allows for monitoring :
(the Yellow is the current % of messages received out of the expected number for the day) 

[image]

drorya1, 2023-08-28:
@assafms  Whooooo really impressive work you done!!!
Will be glad to get it 
I need it badly !!!
Looking forward for the info so I can implement it !
Much appreciated

assafms, 2023-08-28:
https://github.com/assafms/Dialog3g-Decoder/

gdt, 2023-09-29:
Where are we on this issue (lots of useful info!)?  Is someone going to prepare a PR so that rtl_433 will decode these devices?  Something else?

drorya1, 2023-09-29:
Why dont you use the simpler solution shared by assaf ?
I ordered the pcb + components and planning to assemble it in the upcoming
weeks .

DrMacinyasha, 2024-06-28:
Some good news for anyone else who comes across this: https://github.com/merbanan/rtl_433/pull/2984

Looks like Arad(/Master Meter) meters should work if using a build from after that PR was merged. Going to be testing this out over the weekend to see if I'm able to pick up my home's water meter, and will update if there's any issues.

DrMacinyasha, 2024-06-29:
Yeah, issues. rtl_433 is spitting out a _lot_ of garbage, and I'm guessing that's due in no small part to the sheer amount of stuff broadcasting nearby (dense suburban neighborhood, every house has a gas, electric, and water meter broadcasting, plus all the solar meters and who knows what else) and I've got an LNA connected on my test rig due to having issues picking up my own electric meter. PDV can't make heads or tails of the capture I'm taking off the test rig, either.

Going to grab an SDR and laptop tomorrow morning and try to get some captures from the water meter while the antenna's up close and the gain's dialed back. If anyone's got any other recommendations, I'd be happy to hear them.

avicarmeli, 2024-06-29:
@DrMacinyasha could you elaborate on the garbe. maybe example would clarify.

DrMacinyasha, 2024-06-29:
@avicarmeli Here's a test sample for 60 seconds. The config file contents are included in the console output text file, along with the md5sum for each of the files: https://drive.google.com/drive/folders/1nIt0QpFlA0afiCE0wDJCJIM_h6RmIK1v

The meter ID should be 04447485.

avicarmeli, 2024-06-29:
OK I can easily suppress the garbage with longer preamble (as I should have done ) but I am not sure you are using the right frequency. The band is 905.43-923.55 and the American version uses Direct Sequencing and Wide Band. When I asked about that @merbanan answered [that](https://github.com/merbanan/rtl_433/issues/1992#issuecomment-1073271970). 
So maybe he can direct you how to record the right stuff so we can proceed with that. 
In the recording you send I can't see anything familiar.

assafms, 2024-06-30:
Hi Guys, I spent a lot of time trying to guess the CRC used and failed. but I think I was looking at this the wrong way. the block at the end might **not be** CRC/Hash - it might be ECC (Error correction). there is a device Arad is selling DC500 that is a receiver of the messages and relays to GSM (Dialog3G™ Cellular 500 – Remote Small Concentrator). 
 it has "• Error Correcting - Up to 3 Errors" in the PDF. this cannot be done with CRC.

gdt, 2024-06-30:
Very interesting idea.  I do not remember seeing that before in this kind of device, but that doesn't mean it isn't true.

I always lean to implementing a decode with what's understood and outputting the bits from the unknown fields.

The only ideas I have are
 - read ECC textbooks to figure out what 3 errors needs for bits and guess what they did 
 - set up to transmit to the receiver (with a wire so as not to violate the regulations) and basically do fuzzing

avicarmeli, 2024-07-02:
@zuckschwerdt Is it possible to pass optional arguments to the decoder? I believe that it would be valuable to pass the serial number of the meter to the decoder so one would get only their own reading.

zuckschwerdt, 2024-07-02:
Arguments to decoders, esp. to filter output are not a thing rtl_433 wants to do. You can always filter the output (and aggregate, deduplicate, …) with a short script, see https://github.com/merbanan/rtl_433/tree/master/examples

avicarmeli, 2024-07-02:
I believe that  it might be unlawful to read others meter in some countries. 
Maybe I should print out  a warning about that?

merbanan, 2024-07-02:
We can not inform about every possible law in every jurisdiction. But we could add a generic info text in the top directory.

avicarmeli, 2024-07-02:
Got it.

avicarmeli, 2024-07-02:
@merbanan Could you direct @DrMacinyasha how to record usefull data assuming that US version uses Direct sequencing or Wide spectrum?

merbanan, 2024-07-02:
Well if this is a US device it will have a FCC id. Find that and then look it up on the web. It should contain the frequency used. Then go from there the usual way.

avicarmeli, 2024-07-02:
[Here](https://fcc.report/FCC-ID/NTAXMETER21/1088257) is the FCC data. I am not sure what to do with that.

avicarmeli, 2024-07-02:
There is also test [report](https://fcc.report/FCC-ID/NTAXMETER21/1088262) with spectrum graphs. for both WB and DS.

merbanan, 2024-07-02:
905-923MHz is used and it looks like the signal is 500kHz wide. Now it is possible to search for the signal.

avicarmeli, 2024-07-02:
@DrMacinyasha can you use SDR# or similar  while the antenna's up close and the gain's dialed back to look for periodic (~30 s) signal and verify the center frequency in the above band?

DrMacinyasha, 2024-07-02:
@avicarmeli Yup, that's the plan as soon as it cools down a little here. Worst case scenario, I'll run some 60s captures in 1.5 MHz chunks using an NESDR SMArt. Though I'm also trying to get my HackRF to play nicely to see if I can do a wider bandwidth capture and save some time.

DrMacinyasha, 2024-07-04:
@avicarmeli Had a little bit of time yesterday evening to grab a few samples, though my time was a bit limited and some of the signals were a bit odd. But the antenna was about 6" away from the meter down in the hand hole, and the gain was set to minimum. https://drive.google.com/drive/folders/1rBWP2A-U0dOvEX2PIJDG39_zUQawYXlq?usp=sharing

avicarmeli, 2024-07-05:
@DrMacinyasha Looking at the pics you uploaded I think it would be useful to record some cu8 files  with 920.1M as frequency.
The spectrums look very much like what I get here apart from the frequency and the fact that it seems like there are more transmission separate in time but in close but not quite the same frequency.

avicarmeli, 2024-07-05:
@zuckschwerdt I added some outputs to the decoder in order to debug it and I get some unstable output format: 
```
time      : 2024-07-05 18:08:46
model     : AradMsMeter-Dialog3G                   Serial No : 14895702 
Water Reading: 4.731 M^3 Scale Factor: 0x40        Flags     : 0x000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-07-05 18:08:47
model     : AradMsMeter-Dialog3G                   Serial No : 14607208S
Water Reading: 715.739 M^3                         Scale Factor: 0x40        Flags     : 0x000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-07-05 18:08:52
model     : AradMsMeter-Dialog3G                   Serial No : 00633398 
Water Reading: 5.486 M^3 Scale Factor: 0x40        Flags     : 0x000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-07-05 18:09:03
model     : AradMsMeter-Dialog3G                   Serial No : 09444602S
Water Reading: 764.700 M^3                         Scale Factor: 0x00        Flags     : 0x000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-07-05 18:09:16
model     : AradMsMeter-Dialog3G                   Serial No : 14895702 
Water Reading: 4.731 M^3 Scale Factor: 0x40        Flags     : 0x000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-07-05 18:09:17
model     : AradMsMeter-Dialog3G                   Serial No : 14607208S
Water Reading: 715.739 M^3                         Scale Factor: 0x40        Flags     : 0x000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-07-05 18:09:22
model     : AradMsMeter-Dialog3G                   Serial No : 00633398 
Water Reading: 5.486 M^3 Scale Factor: 0x40        Flags     : 0x000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-07-05 18:09:26
model     : AradMsMeter-Dialog3G                   Serial No : 09225790S
Water Reading: 799.800 M^3                         Scale Factor: 0x00        Flags     : 0x000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-07-05 18:09:33
model     : AradMsMeter-Dialog3G                   Serial No : 09444602S
Water Reading: 764.700 M^3                         Scale Factor: 0x00        Flags     : 0x000

```
As can be seen the Scale factor tabulation is not stable for some reason.
Here is the function:
```
data = data_make(
        "model",       "",               DATA_STRING,    "AradMsMeter-Dialog3G",
        "id",          "Serial No",      DATA_STRING,    sernoout,
        "waterread",   "Water Reading",  DATA_FORMAT,    "%.3f M^3", DATA_DOUBLE, wread,
        "Sfctor",      "Scale Factor",   DATA_STRING,    sfout,
        "Flags",       "Flags",          DATA_STRING,    flagsout,
        //"mic",         "Integrity",      DATA_STRING,    "CHECKSUM",
        NULL);
```
Any idea?

DrMacinyasha, 2024-07-07:
@avicarmeli Grabbed from the indoor test rig, trying to run the capture through PDV while tweaking settings, and also looking through the output that the test build of rtl_433 gave, but not seeing the meter's ID (decimal, hex, or reverse hex) anywhere.  Here's the capture and logs, with the console output text file showing the config (only tweaked to tune to 920.1 MHz as recommended, and run for 90 seconds instead of 30): https://drive.google.com/drive/folders/1ECk_YTT9WKpoqkFJ3K3v8jMPNWsQCY7q?usp=sharing

avicarmeli, 2024-07-07:
@DrMacinyasha The erratic outputs are due to the lack of check whether preamble check returned nothing. Here is debug version of  [arad_msMeter.c](https://drive.google.com/file/d/1L3f_cnn33PoDUZcbcwvtFEv8BVMgeg4o/view?usp=drive_link) which correct that.
Will look at your data soon.

avicarmeli, 2024-07-07:
@zuckschwerdt could you look at the data @DrMacinyasha collected like [that](https://triq.org/pdv/#AAB014070100270067004200340015001027118494848355+AAB0120701002700670042003400150010271184A255+AAB01107020027006700420034001500102711A255+AAB0130701002700670042003400150010271184A48355+AAB016070100270067004200340015001027118484848484A255+AAB0120701002700670042003400150010271184A355+AAB01307010027006700420034001500102711A4848355+AAB01207010027006700420034001500102711A48355+AAB01507010027006700420034001500102711848484A48255+AAB01207010027006700420034001500102711A48255+AAB01407010027006700420034001500102711848484A355+AAB01107010027006700420034001500102711A255+AAB0120701002700670042003400150010271184A355+AAB01407010027006700420034001500102711A484848255+AAB01707010027006700420034001500102711848484A484848255+AAB01107020027006700420034001500102711A255+AAB017070100270067004200340015001027118485A48584B0A055+AAB01107010027006700420034001500102711A355+AAB018070100270067004200340015001027118484848484A4848255+AAB01107010027006700420034001500102711A255+AAB01507010027006700420034001500102711848484848655). I advised to use 920.1M but I am not sure. There is the Direct Sequans issue which I am not sure what to make of.
If you could give us  a lead where to go next.

zuckschwerdt, 2024-07-07:
Rename to `stream-sample_920.1M_1000k.cu8` and run rtl_433, there is SCMplus, IDM, and NetIDM data in there.
Also most transmission in there seem to be OOK and the Arad is supposed to be FSK, right?

Do you have time positions for interesting data in that file? Maybe use https://triq.org/spectrogram-next/ to scan and zoom around.

The pdv you linked seems bad, 66 & 40 µs pulses with 20, 40, 52, and 66 µs gaps?

avicarmeli, 2024-07-07:
Yes, it is suppose to be wide band BFSK with 60Kbps. So I am looking for 16 microsecond per bit.
I am not Familiar with SCMplus, IDM, and NetIDM.
Maybe you can take a look at the screen captures @DrMacinyasha [uploaded](https://github.com/merbanan/rtl_433/issues/1992#issuecomment-2208387686), and recommend better frequency to record?

zuckschwerdt, 2024-07-07:
Sorry, I don't have any experience with that band or utility meters.

avicarmeli, 2024-07-08:
@merbanan could you take a look at @DrMacinyasha [spectrograms] (https://github.com/merbanan/rtl_433/issues/1992#issuecomment-2208387686) and direct us where to go next?

[14 more replies]

closed 2025-07-02 by gdt