merbanan/rtl_433 #1993: BRESSER soil temperature and moisture meter battery state support

myhomeiot, 2022-02-22:
Hello, 

First I want to thanks for your great work!

The new [BRESSER soil temperature and moisture meter](https://www.bresser.de/en/Weather-Time/BRESSER-Soil-Moisture-and-Soil-Temperature-Sensor.html) for 7-in-1 weather station (Product number: 7009972, EAN: 4007922074153) show wrong battery state and non relevant wind data (this sensor meter only soil temperature and moisture).
I check the data with weather station console, temperature and moisture is OK.

Good battery:
```
bresser_6in1_decode: : {144} 94 e8 95 30 81 34 41 bb bb bb bb b0 21 86 01 bb b0 95 
time      : 2022-02-22 21:00:11
model     : Bresser-6in1 id        : 95308134
channel   : 1            Battery   : 0             Temperature: 21.8 C       Sensor type: 4            Moisture  : 0 %
Wind Gust : 44.4 m/s     Wind Speed: 44.4 m/s      Direction : 1221          Flags     : 0             Integrity : CRC
```

Bad battery, weather station console shows low battery icon:
```
bresser_6in1_decode: : {144} b3 71 95 30 81 34 41 bb bb bb bb b0 21 80 01 bb b0 9b 
time      : 2022-02-22 21:01:11
model     : Bresser-6in1 id        : 95308134
channel   : 1            Battery   : 0             Temperature: 21.8 C       Sensor type: 4            Moisture  : 0 %
Wind Gust : 44.4 m/s     Wind Speed: 44.4 m/s      Direction : 1221          Flags     : 0             Integrity : CRC
```

Difference in msg[13], good battery has 0x86, bad 0x80, looks like battery state it's last 4 bits which should be xor'ed with 0x0f?

zuckschwerdt, 2022-02-23:
Yes, for the soil probe type (b[6] == 0x4?) we should ignore the other values. We expected 0xff in those fields (= invalid) but here we get 0xbb strangely.
Not sure if (b[13] &0xf) can be the battery, other samples have `2` there, you have `6` and `0`.

myhomeiot, 2022-02-23:
I did 10+ tests with good and bad batteries, and resync weather station console each time, so I'm confident that (b[13] &0xf) it's a battery for [BRESSER soil temperature and moisture meter](https://www.bresser.de/en/Weather-Time/BRESSER-Soil-Moisture-and-Soil-Temperature-Sensor.html) I also never seen 0xff in the fields where I see 0xbb.

anthyz, 2022-02-25:
I have the Ambient Weather branded version of this sensor (TX-3102). I did some tests and found that b[13] bit 1 was set when battery was good and clear when battery was low.

Low battery (less than 3V total):

bresser_6in1_decode: : {144} 60 5a 18 70 07 93 41 ff ff ff 00 00 22 50 01 ff f2 3b 
bresser_6in1_decode: : {144} 39 88 18 70 07 93 41 ff ff ff 00 00 22 30 01 ff f2 5b 
bresser_6in1_decode: : {144} 3d 2c 18 70 07 93 41 ff ff ff 00 00 21 90 01 ff f2 fc 
bresser_6in1_decode: : {144} 3d 2c 18 70 07 93 41 ff ff ff 00 00 21 90 01 ff f2 fc 

More than 3V total:

bresser_6in1_decode: : {144} e3 ae 18 70 07 93 41 ff ff ff 00 00 22 12 01 ff f2 79 
bresser_6in1_decode: : {144} e3 ae 18 70 07 93 41 ff ff ff 00 00 22 12 01 ff f2 79 
bresser_6in1_decode: : {144} e3 ae 18 70 07 93 41 ff ff ff 00 00 22 12 01 ff f2 79 
bresser_6in1_decode: : {144} cf 47 18 70 07 93 41 ff ff ff 00 00 22 22 01 ff f2 69

Hope that helps.

zuckschwerdt, 2022-02-25:
Thanks. So `b[13] & 0x02` is a battery_good bit.

myhomeiot, 2022-05-06:
> Thanks. So `b[13] & 0x02` is a battery_good bit.

Hello, do you need more data or tests? Can you make the fix for battery state?
Thanks in advance!

zuckschwerdt, 2022-05-06:
Yes, I'll try to get this in soon, sorry.

myhomeiot, 2022-05-20:
Bump, thanks in advance!

myhomeiot, 2022-06-01:
Bump, with love! ;-)

zuckschwerdt, 2022-06-07:
Finally got around to add this. Thanks for figuring out the mapping and sticking around to see it through!

closed 2022-06-07 by zuckschwerdt

zuckschwerdt, 2022-06-08:
Oh, I missed UV. Thanks!

commit 12ec9a4a (2022-06-07): minor: Fix Bresser-6in1 soil sensor battery bit (closes #1993)