merbanan/rtl_433 #1994: Not reading Protocol 52 (Bresser Thermo/Hydro NLX)

SolarCzar, 2022-02-23:
Hey guys,

Acurite 986, 606 sensors reading/working great.  Saw some Bresser 3 in 1 sensors and color monitor at a good price and picked it up ($25 for monitor + 3 sensors).  Loaded Protocol 52 (in fact subsequently loaded all the Bresser protocols) in startup file, but not showing any sensor data.  I searched the issues, but not seeing any comments as to any Bresser issues.  

Forgive my noobness, as I'm just following the recipes you guys lay out. I'm way more interested in capturing/reacting to the data in Home Asst for integrations.  You guys have been great with guidance, so will continue appreciate a little more....

SolarCzar, 2022-02-23:
This is the actual unit I purchased...
https://www.amazon.com/gp/product/B07SCXRK2J/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

zuckschwerdt, 2022-02-24:
Those look a bit like the 3-in-1 but at least a "face lift". Can you follow https://triq.org/rtl_433/ANALYZE.html to get a good sample and then upload here as zip?

SolarCzar, 2022-02-24:
So your direction helped.  I ran `-S known` and they came up as Nexus-TH sensors, or protocol 19.  So I'm good, thanks brother Z.

zuckschwerdt, 2022-02-25:
Glad it worked out. Yes the Nexus is a very old and common protocol.

closed 2022-02-25 by zuckschwerdt