merbanan/rtl_433 #1995: Using MQTT formats prevents rtl_433 from starting

nukefrenzy, 2022-02-24:
I'm using this thru the HA add-on here: https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433.  When I just use the following everything runs fine:

`output mqtt://192.168.1.13:1883,user=mqtt,pass=MYPASSWORD`

However, as soon as I start trying to change the topic structure, rtl_433 just hangs and won't continue.  Here's an example of what I'm trying to do:

`output mqtt://192.168.1.13:1883,user=mqtt,pass=MYPASSWORD,devices=rtl_433[/id]`

Is there something incorrect with my format string?  Or is there an issue parsing it in rtl_433?  The below is all I see in the logs:

[image]

I realize this is not your add-on, but it is simply running your code within HA so I thought I'd ask.  Thank you.

zuckschwerdt, 2022-02-24:
Just a config file with that single line works without problem here.
The really strange thing is that the version output is actually two lines:
```
rtl_433 version 21.12-61-g39570f39 branch fix-pdreset at 202202181546 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
```
[There is no code ](https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L129-L130)

nukefrenzy, 2022-02-24:
When I only use the following:

`output mqtt://192.168.1.13:1883,user=mqtt,pass=MYPASSWORD`

Then my logs look like this, which match your comment above.  So it is definitely hanging up due to the added format items.

[image]

zuckschwerdt, 2022-02-24:
Would be interesting what's actually in the config file. I guess the `[/id]` gets replaced somehow? rtl_433 shouldn't choke on that though.
Are you using a manual config file (https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/run.sh#L22)?
The automatic config doesn't seem to support topic strings (https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/run.sh#L43)

nukefrenzy, 2022-02-24:
This is it:

verbose 	1
frequency 	345M
output 		mqtt://192.168.1.13:1883,user=mqtt,pass=MYPASSWORD,state=rtl_433[/id]
convert 	si
protocol 	70

nukefrenzy, 2022-02-24:
I think the automatic option is only if the addon finds that you have the old-style config file.  So it automatically creates a new config file in the rtl_433 folder with basic mqtt settings.  Then you can edit this file as needed manually.  The next time the addon is started it will then skip all the automatic sections.

zuckschwerdt, 2022-02-24:
Oh, I see. There is actually an error there: `Invalid key "state" option.` You want `states=`. (ok, that was pre-edit of the conf above...)
But as now given that config works.

nukefrenzy, 2022-02-24:
Yes, I've been messing around with it and pasted in one of my many, many tests.  My config above is now correct, tested, and proven not to work right now.

zuckschwerdt, 2022-02-24:
As you describe it there should be something wrong with the config, but I don't see it. I tested with current version and plain 21.12.
There is a similar bug at pbkhrv/rtl_433-hass-addons/issues/78
We would really need to know if rtl_433 is blocking or aborting on reading the conf.
Is it maybe something with line-endings or invisible characters? Can you `hexdump -C your.conf` (or `xxd your.conf`)?

zuckschwerdt, 2022-02-24:
And to be sure: there are two forbidden characters for the password: `#` and '{' those start comments and clock quoting.

nukefrenzy, 2022-02-24:
I'm wondering if the way this addon creates the actual rtl_433.conf file from my rtl_433.conf.template file is messing with what I put in it.  I'm just not advanced enough to understand these lines of code: https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/run.sh#L55-L74

nukefrenzy, 2022-02-24:
my password has % in it, but that should also prevent the basic mqtt setup from working too, I would think.

zuckschwerdt, 2022-02-24:
The `%` is not a special char rtl_433 is concerned about. The heredoc mechanism should only replace `$foo` and `${foo}`, but there may be other cases in different shells?

nukefrenzy, 2022-02-24:
I changed my password in the below to resemble my actual password

```
file name: rtl_433.conf.template
mime type: 

0000-0010:  76 65 72 62-6f 73 65 20-09 31 0a 66-72 65 71 75  verbose. .1.frequ
0000-0020:  65 6e 63 79-20 09 33 34-35 4d 0a 6f-75 74 70 75  ency..34 5M.outpu
0000-0030:  74 20 09 6d-71 74 74 3a-2f 2f 31 39-32 2e 31 36  t..mqtt: //192.16
0000-0040:  38 2e 31 2e-31 33 3a 31-38 38 33 2c-75 73 65 72  8.1.13:1 883,user
0000-0050:  3d 6d 71 74-74 2c 70 61-73 73 3d 31-32 33 25 50  =mqtt,pa ss=123%P
0000-0060:  61 73 73 77-6f 72 64 2c-64 65 76 69-63 65 73 3d  assword, devices=
0000-0070:  72 74 6c 5f-34 33 33 5b-2f 69 64 5d-0a 63 6f 6e  rtl_433[ /id].con
0000-0080:  76 65 72 74-20 09 73 69-0a 70 72 6f-74 6f 63 6f  vert..si .protoco
0000-0086:  6c 20 09 37-30 0a                                l..70.
```

zuckschwerdt, 2022-02-24:
So some mixed tabs ans spaces, but that fine. I copied that verbatim and run it through the heredoc, it works just fine. No idea what breaks :/

nukefrenzy, 2022-02-24:
I can't thank you enough for looking at this.  Not an issue with this project, so it looks like I'll open on with the HA addon.  I needed a gut check to make sure I wasn't doing something stupid and not seeing it.  Before the addon switched how they handled config files I was able to get this working, but when they converted it deleted my old config file, and I had to start over.  Then I started questioning how I got it working before and why I couldn't repeat it...it's been driving me crazy.  Thanks!

zuckschwerdt, 2022-02-24:
The config file isn't a clean design but evolved, so there is a chance for a bug still ;)

closed 2022-02-24 by nukefrenzy

zuckschwerdt, 2022-02-24:
If someone could check in the running docker container when this hangs: it would be interesting if rtl_433 did choke on the config and is maybe in an infinite loop at 100% processor load, or just aborted.

nukefrenzy, 2022-02-24:
Doesn't look like it's hogging CPU:

[image]