merbanan/rtl_433 #1996: Make -Y squelch tunable

aymeric106, 2022-02-25:
Hello, and thanks for this great tool.

the -Y squelch option is a great option, (my raspberry pi goes from 33% to 3% cpu usage with all the default protocols enabled).

Sadly, when enabled, it won't catch some weaker but decent signals (noise -23dB, SNR around 8dB)

Perhaps with a non fixed squelch, we could have both a lower CPU and not too drastic filtering ?

Thank you 
Aymeric

rtl_433 -f 433650000 -s 1024000 -M protocol -M time:iso -F http  Y autolevel -g 42 -Y squelch -M noise -M level

zuckschwerdt, 2022-02-25:
I can look into that, but the squelch is already very sensitive. It needs just 3 dB over current average noise floor (so 3 dB SNR). That should be less than the demod can decode (by default 9 dB SNR are required).
https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L426

Can you check that the noise floor is correctly tracked and what the signal (Level/SNR) would be for a faint transmission (without squelch of course)?

aymeric106, 2022-02-25:
Well if its only 3dB then ignore my comment. I was pretty sure it was more selective. 
Maybe the noise floor (or estimated noise, i'm not sure how it works) changed so much that it wasnt able to catch the low signal anymore.

closed 2023-09-30 by gdt