merbanan/rtl_433 #1997: WH65 falsely recognized as WH24 creates some issues

fankyy, 2022-02-26:
Hi, my WH65 sensor is being wrongly recognized as a WH24, which creates some issues and false readings with the weewx-sdr driver, because the WH24 has other correction factors for wind and rain. These can be corrected in weewx conf, but one problem remains:

It reports fake rain and abnormous high rain readings. When i run rtl_433 i can see it reports 13.5mm rain in every 16 second data packet, when there should be 0 rain. 

So i would either need the correction values for rain count and hope it will work if i combine it with removing 13.5mm in weewx corrections, or a possiblity to manually tell rtl_433 the sensor type.

Thanks for the good work with rtl_433

fankyy, 2022-02-26:
Using -Y classic seems to have fixed everything so far.

rtl_433 -f 868.3M -M time:utc -F json -Y classic

For anyone who has had the same issue: Don't forget to change sensormap in weewx according to the right sensor identifier (FOWH65BAltPacket in my case).

e.g.:
`windSpeed = wind_speed.254.FOWH65BAltPacket`

closed 2022-02-26 by fankyy

zuckschwerdt, 2022-02-26:
The Fineoffset WH24 and WH65B are nearly identical, same protocol, but different wind speed factor and rain cup count.
There is just a slightly longer postamble on the WH65B, which we might be lucky to detect.
We might need to make that two distinct protocols so you can disable one.