merbanan/rtl_433 #1998: Barometric pressure on WH2310 / FOWHx080 outdoor sensor

fankyy, 2022-02-26:
I have a WH65 and a WH2310 ("Renkforce WH2315" [https://www.conrad.com/p/renkforce-wh2300-wh2315-wireless-digital-weather-station-forecasts-for-12-to-24-hours-1404262](url))

I don't know if that's the case for any WHx080 model, but this specific one seems to have the barometer on the outdoor sensor. Because everytime when the signal reception fails, the pressure data is lost too. 

But as it seems, rtl_433 doesn't encode any barometric pressure data for this model(?) It would be nice to use the outdoor barometer of the WH2310 to complement a WH65 SDR setup.

zuckschwerdt, 2022-02-26:
Is the WH2310 recognized as WH65B or what protocol?
Try to log the received bitbuffers, e.g. `rtl_433 -vv -R 78`

fankyy, 2022-02-26:
No it is recognized as WHx080 so that is working fine, but can't see the barometric data.

But I have a WH65B and would take the main data from it and the pressure from the WHx080

log:
# rtl_433 -f 868.3m -M time:utc -vv -R 32 -Y classic
rtl_433 version 21.12-31-g32c1f4a6 branch master at 202201101907 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "(null)/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registering protocol [32] "Fine Offset Electronics WH1080/WH3080 Weather Station"
Registered 1 out of 210 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
Fine Offset WH1080 data {80} ff a0 81 d5 30 06 09 1c 62 1c 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2022-02-26 13:54:01
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 8
Battery   : 0            Temperature: 6.9 C        Humidity  : 48 %          Wind Direction: 270       Wind avg speed: 7.34      Wind gust : 11.02
Total rainfall: 951.0    Integrity : CRC
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station
bitbuffer:: Number of rows: 1 
[00] {87} ff 41 03 aa 60 0c 12 38 c4 38 88 
Fine Offset WH1080 data {80} ff a0 81 d4 30 08 09 1c 62 1e 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2022-02-26 13:54:49
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 8
Battery   : 0            Temperature: 6.8 C        Humidity  : 48 %          Wind Direction: 315       Wind avg speed: 9.79      Wind gust : 11.02
Total rainfall: 951.0    Integrity : CRC
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station
bitbuffer:: Number of rows: 1 
[00] {87} ff 41 03 a8 60 10 12 38 c4 3c f2 
Fine Offset WH1080 data {80} ff a0 81 d4 30 08 09 1c 62 1e 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2022-02-26 13:54:49
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 8
Battery   : 0            Temperature: 6.8 C        Humidity  : 48 %          Wind Direction: 315       Wind avg speed: 9.79      Wind gust : 11.02
Total rainfall: 951.0    Integrity : CRC
pulse_demod_pwm(): Fine Offset Electronics WH1080/WH3080 Weather Station
bitbuffer:: Number of rows: 1 
[00] {87} ff 41 03 a8 60 10 12 38 c4 3c f2 `

zuckschwerdt, 2022-02-26:
The WHx080 weather messages (`ff` **`a`**) have all fields known and can't carry a barometric pressure, see https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset_wh1080.c#L26-L39
This would only work if it were a new message type, we only know about a,b,7,3,5 so far.
You will need to log raw messages (that might not even be in the list above), see https://triq.org/rtl_433/ANALYZE.html

gdt, 2023-09-30:
@fankyy Status and next steps?  Seems like we might as well close this.

fankyy, 2023-10-03:
I need some more time, currently other priorities. Thanks for the help so far. If you want to close it for cleaning up purposes it's ok.

gdt, 2023-10-03:
I'm going to close this, because it's basically "I think this device must transmit barometer" but there is no evidence of a non-decoded signal on the table.  Please feel free to open a new issue if you have captures you can't decode, new message types, etc., or to ask for help in discussions or the list.

closed 2023-10-03 by gdt