merbanan/rtl_433 #2000: Help decoding signal from unknown temp sensor

TomW1605, 2022-03-01:
hi

i have several saved signal files from a [temp sensor](https://www.altronics.com.au/p/x7064-wireless-remote-sensor-to-suit-x7063/) i bought. it has next to no markings on it and i can't tell if it is a rebrand of a standard chinese model.

i honestly have no idea where to start with processing them. it looks like they are detected reliably and consistently so the files should be clean.

i have viewed the samples in waveform analysers but i don't know where to go from there. i have attached the samples if someone can help or if you can point me in the right direction.

[samples.zip](https://github.com/merbanan/rtl_433/files/8160669/samples.zip)

zuckschwerdt, 2022-03-01:
The spectrogram looks like heavy aliasing, but the demod actually works fine and results in nice pulses.
Write the files to `.ook` (works for FSK too): `rtl_433 -w g016_433.92M_250k.ook g016_433.92M_250k.cu8`
Then throw those pulse files onto https://triq.org/pdv/ and figure out the coding. The intro at https://triq.org/rtl_433/PULSE_FORMATS.html might be some help.

zuckschwerdt, 2022-03-01:
Looks like each transmission has 3 repeats (e.g. g017 has two transmissions). The coding is not something I recognize.
With a PCM slicing of 90 µs you can get the raw pulse data, but that looks too regular and too much data for a temperature sensor. Symbols must be wider that 90 µs.
Then maybe copy a good code of each transmission out of pdv (break the three apart roughly and compare), put the [codes in a BitBench](https://triq.net/bitbench#c=FFFFFAAAAAAAAAACACA54AAA1AA55A6BAA53BAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAD0AA&c=FFFFFAAAAAAAAAACACA54AAA1AA55A6CAA03AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADAAA&c=FFFFFAAAAAAAAAACACA54AAA1AA51A6CAA03AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAD6AA&f=8h) and see if you spot a pattern in the changes of codes.

TomW1605, 2022-03-01:
it does do humidity as well so there is more than temp data but probably not much more.

i have gotten the 2 from g17 into BitBench and played around a bit but im not realy sure what i am aiming for. i also stuck them in a text comparison and there are only a few (7) pairs that are diffrent. i do thing there was an extra 1 between the 2 signals though. not sure what thats about but i had to remove it for them to line up with the preamble stuff.

zuckschwerdt, 2022-03-02:
If you use something like `ffffaaaaaaaaaacaca` as a preamble match and break the lines at the `00` runs you [get this BitBench](https://triq.net/bitbench#c=FFFFAAAAAAAAAACACA54AAA16E91A68AAC2AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA96AA&c=3FFFFF55555555559594A95542DD234D15585555555555555555555555555555555555555555555555552D54&c=7FFFFEAAAAAAAAAB2B2952AA85BA469A2AB0AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA5AA8&c=FFFFAAAAAAAAAACACA54AAA16E95A69AAF2BAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAEAA&c=3FFFFF55555555559594A95542DD2B4D355E5755555555555555555555555555555555555555555555555D54&c=7FFFFEAAAAAAAAAB2B2952AA85BA569A6ABCAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAA8&c=FFFFAAAAAAAAAACACA54AAA1AA55A6BAA53BAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAD0AA&c=3FFFFF55555555559594A9554354AB4D754A775555555555555555555555555555555555555555555555A154&c=7FFFFEAAAAAAAAAB2B2952AA86A9569AEA94EEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAB42A8&c=FFFFAAAAAAAAAACACA54AAA1AA55A6CAA03AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADAAA&c=3FFFFF55555555559594A9554354AB4D9540755555555555555555555555555555555555555555555555B554&c=7FFFFEAAAAAAAAAB2B2952AA86A9569B2A80EAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAB6AA8&c=FFFFAAAAAAAAAACACA54AAA1AA51A6CAA03AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAD6AA&c=3FFFFF55555555559594A9554354A34D9540755555555555555555555555555555555555555555555555AD54&c=7FFFFEAAAAAAAAAB2B2952AA86A9469B2A80EAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAB5AA8&f=72h%20ID%3F8h%208h%208h%20DATA%3F8h%208h%208h%208h%208h%20HUM%3F8d%20TRAILER%3F%20184h%20CHK%3F8h8h%208h&a=Sync&m=ffffaaaaaaaaaacaca).

That's already gives a good idea about the format.

Since we don't see long runs of 0's we can limit the allowed gap to break automatically and also find and align on the (assumed) preamble:
`rtl_433 -X 'n=name,m=FSK_PCM,s=90,l=90,g=800,r=9000,preamble=ffffaaaaaaaaaacaca'`

Can you use that to grab a range of codes with a known progression of values (e.g. a slow warm up or cool down)?

zuckschwerdt, 2022-03-02:
Oh, I see. the preamble is actually `ffffaaaaaaaaaacaca54` (or some rightmost part of it). That then leaves the remaining 31 bytes to add to byte 32, [see this BitBench](https://triq.net/bitbench#c=FFFFAAAAAAAAAACACA54AAA16E91A68AAC2AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA96AA&c=3FFFFF55555555559594A95542DD234D15585555555555555555555555555555555555555555555555552D54&c=7FFFFEAAAAAAAAAB2B2952AA85BA469A2AB0AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA5AA8&c=FFFFAAAAAAAAAACACA54AAA16E95A69AAF2BAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAEAA&c=3FFFFF55555555559594A95542DD2B4D355E5755555555555555555555555555555555555555555555555D54&c=7FFFFEAAAAAAAAAB2B2952AA85BA569A6ABCAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAA8&c=FFFFAAAAAAAAAACACA54AAA1AA55A6BAA53BAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAD0AA&c=3FFFFF55555555559594A9554354AB4D754A775555555555555555555555555555555555555555555555A154&c=7FFFFEAAAAAAAAAB2B2952AA86A9569AEA94EEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAB42A8&c=FFFFAAAAAAAAAACACA54AAA1AA55A6CAA03AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADAAA&c=3FFFFF55555555559594A9554354AB4D9540755555555555555555555555555555555555555555555555B554&c=7FFFFEAAAAAAAAAB2B2952AA86A9569B2A80EAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAB6AA8&c=FFFFAAAAAAAAAACACA54AAA1AA51A6CAA03AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAD6AA&c=3FFFFF55555555559594A9554354A34D9540755555555555555555555555555555555555555555555555AD54&c=7FFFFEAAAAAAAAAB2B2952AA86A9469B2A80EAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAB5AA8&f=80h%20ID%3F8h8h%20DATA%3F8h%208h%208h%208h%208h%20HUM%3F8d%20TRAILER%3F%20184h%20CHKSUM%3A8h%208h%208x&a=Sync&m=ffffaaaaaaaaaacaca54&cf=ADD&co=80&cl=248&cw=8).

TomW1605, 2022-03-02:
i think you are at least right about the `aaa1` being the id. when i switch the "channel" switch that changes to `aaa2` and `aaa3` for "channel" 2 and 3 respectively

TomW1605, 2022-03-02:
ok got 1 more reading for a diffrent humidity and it matches in the signal. so im pretty confident with that part too.

TomW1605, 2022-03-02:
im going to turn on the AC and try and cool the room a bit to get some more temp readings

TomW1605, 2022-03-02:
some of them have an error in the first repeat. i assume that is why they send it all 3 times

zuckschwerdt, 2022-03-02:
So with some cleanup and removing dups we [get this BitBench](https://triq.net/bitbench#c=ffffaaaaaaaaaacaca54aaa16e91a69aa629aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa9faa%5B25.8%2041%5D&c=ffffaaaaaaaaaacaca54aaa16e91a69aa728aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa9faa%5B25.8%2041%5D&c=ffffaaaaaaaaaacaca54aaa16e91a69aa728aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa9faa%5B25.9%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a69aab28aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa3aa%5B26.1%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a69aac28aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa4aa%5B26.2%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a69aae28aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa6aa%5B26.3%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a69aaf28aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa7aa%5B26.3%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6aaa228aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa%5B26.5%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6aaa829aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab1aa%5B26.7%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6aaa829aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab1aa%5B26.8%2041%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6aaab28aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab3aa%5B27.0%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6aaaf28aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab7aa%5B27.2%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6baa228aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaabaaa%5B27.4%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6baa528aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaabdaa%5B27.6%2040%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6baa62aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaac0aa%5B27.6%2042%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6baa829aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaac1aa%5B27.7%2041%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6baa82aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaac2aa%5B27.7%2042%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6baa929aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaac2aa%5B27.8%2041%5D&c=ffffaaaaaaaaaacaca54aaa16e91a6baaa29aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaac3aa%5B27.8%2041%5D&f=CHAN%3F8h8h%20ID%3F8h8h8h%20TEMP%3F4h8x4h%20HUM%3F8d%20TRAILER%3F%20184h%20CHKSUM%3A8h%208h%208x&a=Preamble&m=ffffaaaaaaaaaacaca54&cf=ADD&cl=248&cw=8). The layout is strange but the temp seems to be a hex value with 0.1 C resolution. But there must be more bits for very high and low temperatures.

TomW1605, 2022-03-02:
that dosnt look right. many of the signals with the same display temp have a temp signal that is 1 off from each other

also i assume you are only looking at 1 of the repeats per signal here right?

zuckschwerdt, 2022-03-02:
What you see are rounding errors in the display. Do a regression. It's C = 0.056 x T + 17.4, which works nicely but looks odd. This points to F as unit, where F = 0.1 x T + 63. Just add 630 to the raw value and you get deci-F.

The repeats are all identical.

TomW1605, 2022-03-02:
so take the `b9`, convert it to decimal, add 630, divide by 10 then convert to Celsius?

am i understanding that right?

also why `b9`? that is the first and last digit of the data section. seams like an odd way to do it. how did you pick up on that?

zuckschwerdt, 2022-03-02:
Yes, it really weird. But that's the only mapping to work. Also with the the bits found so far the range would be
about 63 F to 89 F which is about 17 C to 32 C and not reasonable. We need measurements outside that range to find more bits (probably the ones in the "middle"). Can you freeze and hot air the sensor perhaps?

zuckschwerdt, 2022-03-02:
Ok so the digit right before the found ones is part of temp. It is `6` for the older values and now `4`.
This then works out to a much nicer round offset of -900, e.g. 
```
6b9 = 1721, -900 = 82.1 F, -> 27.8  C
448 = 1096, -900 = 19.6 F, -> -6.9  C
```
Now the range is wide enough. But the digit even before that is changing too, it can't be part of the temp though.

TomW1605, 2022-03-02:
could be a charge indicator. i havnt taken the battries out since the first samples from today (i cant remember if the samples in the zip are from befor or after i pulled it appart to look for a number on the pcb or some other ID to look up the codes)

TomW1605, 2022-03-02:
it is also `6e95` in the second last one of the big block i did earlier. i thought it might have just been a transmission error but not so sure now

TomW1605, 2022-03-02:
-17.2°C
ffffaaaaaaaaaacaca54aaa16e95a38aae2daaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa9caa

got another reading. this one also has the `6e95` in it

zuckschwerdt, 2022-03-02:
That fits nicely: `16e95a38aae2d` is channel `1`, id `6e95`, temp `38e` (=910, 1 F, -17.2 C), hum `2d` (=45).

TomW1605, 2022-03-02:
right i forgot to divide by 10.
i think part of the id there might be the battery. it has changed with no apparent reason so i doubt it is all the id

TomW1605, 2022-03-02:
does that still have an upper range of 32c? definitely gets hotter than that here (Australia)

zuckschwerdt, 2022-03-02:
If we take the full 12 bit it goes from -90 F to 319 F, -67 C to 159 C.

TomW1605, 2022-03-02:
ok i doubt the sensor would even survive that range of temp.

so now we (mostly you) have worked this out how do we get it into a proper decode? if this is the part where we need some code im happy to do that. i have worked in C before and looking at other devices it doesn't look that complex.

TomW1605, 2022-03-02:
also 1 more reading from this morning after being in the freezer all night:
aaa16e95a35aac29
screen: -20c 41%
decode: -20c 41%

i think we have it sorted. thanks for all your help

zuckschwerdt, 2022-03-03:
Sure, PR a decoder for this. If you feel this is not fun though (the protocol is much weirder than usual) let me know, I can add the decoder quickly.

TomW1605, 2022-03-03:
ok i have got a build running but having had another look at the template there is a bunch of stuff i dont know about the signal that i need to. if its not too much trouble i think it might be easier for you to just do it. otherwise i will just be bombarding you with questions the whole time.

closed 2022-03-03 by zuckschwerdt

TomW1605, 2022-03-03:
thanks it works great.

one thing though. i think we are wrong about the battery bit. it looks like that is 1 normally and 0 if i press the manual sync button. probably something to tell the base station to register the transmitter id.

TomW1605, 2022-03-03:
also would it be possible to convert the temp to celcius? it makes it a lot easier to deal with and as far as we know the sensor is only available in australia where celcius is the standard.

zuckschwerdt, 2022-03-03:
A common request ;) we have `-C si`

Can you replace `& 0x04` with `& 0x0f` in https://github.com/merbanan/rtl_433/blob/master/src/devices/altronics_x7064.c#L88 and remove the `!` on line 98, then try a few things with the sync button and flat batteries or a variable voltage supply if you have one? I'm not sure where the ID ends and if/what flags there are, but there must be battery bit I guess.

TomW1605, 2022-03-03:
> A common request ;) we have -C si
great, should have known there would be something for this (although it still says "Temperature_F" in the output)

ill do some testing on the weekend. didn't think of using a variable power supply to simulate dead batteries, i was just going to leave a torch on overnight. your idea is better.
ill let you know how it goes.

TomW1605, 2022-03-04:
ok something strange happened. it looks like the way it is being received has changed. how it is seeing the 3 messages as separate messages and it is decoding to a different string but still gets the right data.

```
time      : 2022-03-04 16:57:19+0800
model     : name         count     : 3             num_rows  : 3             rows      :
len       : 351          data      : ffff55555555559594a955421e2b4d555e4f5555555555555555555555555555555555555555555555b75400,
len       : 357          data      : fffffd55555555565652a5550878ad3555793d5555555555555555555555555555555555555555555556dd5000,
len       : 357          data      : fffffd55555555565652a5550878ad3555793d5555555555555555555555555555555555555555555556dd5000
codes     : {351}ffff55555555559594a955421e2b4d555e4f5555555555555555555555555555555555555555555555b75400, {357}fffffd55555555565652a5550878ad3555793d5555555555555555555555555555555555555555555556dd5000, {357}fffffd55555555565652a5550878ad3555793d5555555555555555555555555555555555555555555556dd5000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-04 16:57:19+0800
model     : Altronics-X7064                        id        : 0f1
Channel   : 1            Battery_OK: 0             Temperature_F: 27.3       Humidity  : 39            Integrity : CHECKSUM
```

zuckschwerdt, 2022-03-04:
That's just a few bits more or less at the start of the packets. This is expected, there is no inherent (byte-)alignment in transmissions. Put the codes in BitBench and shift them a few bits, or use `aaaacaca54` as "Align" and activate "Preamble".
You might also want to add `,preamble=aaaacaca54` to the flex conf

TomW1605, 2022-03-04:
ok thanks that worked. will do the testing with the power supply

TomW1605, 2022-03-04:
ok got 2 sets of data. signals from 2.1V to 3V in 0.1V increments. at 2V it was very inconsistent with its transmitting so i started at 2.1. the first table is all from pressing the sync button and the second table is all from auto transmit

Voltage | Code
-- | --
2.1 | aaa1ca3da6caa328aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaad3aa00
2.2 | aaa1ca3da6caa328aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaad3aa00
2.3 | aaa1ca3da6caa327aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaad2aa00
2.4 | aaa1ca3da6cae327aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaad2aa00
2.5 | aaa1ca3da6caa327aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaad2aa00
2.6 | aaa1ca35a6caa327aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaacaaa00
2.7 | aaa1ca35a6caa327aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaacaaa00
2.8 | aaa1ca35a6caa327aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaacaaa00
2.9 | aaa1ca35a6caa327aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaacaaa00
3.0 | aaa1ca35a6caa327aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaacaaa00

Voltage | Code
-- | --
2.1 | aaa1ca39a6caa427aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaacfaa00
2.2 | aaa1ca39a6caa127aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaccaa00
2.3 | aaa1ca39a6baaf27aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaacaaa00
2.4 | aaa1ca39a6baac27aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaac7aa00
2.5 | aaa1ca39a6baad26aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaac7aa00
2.6 | aaa1ca39a6baab26aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaac5aa00
2.7 | aaa1ca31a6baa926aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaabbaa00
2.8 | aaa1ca31a6baa726aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab9aa00
2.9 | aaa1ca31a6baa426aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab6aa00
3.0 | aaa1ca31a6baa426aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab6aa00

zuckschwerdt, 2022-03-04:
Very nice! So 0x8 is battery_low starting at about 2.6V and 0x4 is sync_button. I'll add that now.

zuckschwerdt, 2022-03-04:
Added with d123816 now.

TomW1605, 2022-03-04:
works great. thanks.

out of curiosity what's with the `?` on the `Pairing?` attribute?

zuckschwerdt, 2022-03-04:
That's just the descriptive label, the actual key is "pairing". It indicates an option or question, but really is just a quick copy&paste from another decoder ;)

TomW1605, 2022-03-05:
ah fair. thanks for all your help. time to go get 3 more for the rest of the house (they are $14 so im happy to buy a bunch)

commit f7afe931 (2022-03-03): Add support for Altronics X7064 sensor (closes #2000)
commit d8293a75 (2022-03-03): Add support for Altronics X7064 sensor (closes #2000)
commit 63514de0 (2022-03-03): Add support for Altronics X7064 sensor (closes #2000)