merbanan/rtl_433 #2001: Support for Emos E6016 Rain Gauge sensor

stomko11, 2022-03-01:
Hello. Recently support for Emos E6016 weather station was added. My set contained also rain gauge sensor (no specific name, just additional sensor). I have uploaded signal captures to `rtl_433_tests`  (at the moment it is still pending approval to upload) and am looking for anyone willing to help on this.
@AutomGuy since you actively worked on Emos E6016 addition, wondering if you also have rain gauge too?

zuckschwerdt, 2022-03-01:
The transmission looks different than the EMOS E6016, it's not PWM but Manchester.
This should get the codes: `rtl_433 -X 'n=EMOS-RAIN,m=OOK_MC_ZEROBIT,s=500,l=500,r=1200'`
If that does not work get the raw PCM with: `rtl_433 -X 'n=EMOS-RAIN,m=OOK_PCM,s=500,l=500,r=1200'`

Try to get a progression of codes with each code +1 bucket tip. Then we can look for the fields in a BitBench.

zuckschwerdt, 2022-03-01:
Those codes look empty. E.g. from your g009 I get:
```
{97}2aaaaaaaccb532cd553532b50
{97}33354b555554b54ccaad2b528
```

stomko11, 2022-03-01:
OK, here is another capture using second command. Same method as previously.
[emos_rain2.txt](https://github.com/merbanan/rtl_433/files/8161843/emos_rain2.txt)

zuckschwerdt, 2022-03-01:
Ok, so it's not Manchester then or not recognized as MC anyway.
If we assume a data whitening of 0xaa we [get this BitBench](https://triq.net/bitbench#c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafaebbeffc%5B14%3A14%3A19%5D&c=%7B147%7Deeeeeebbeaeeafbafefaafaaaaaaafaebbef8%5B14%3A14%3A19%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafafbbfafc%5B14%3A15%3A44%5D&c=%7B147%7Deeeeeebbeaeeafbafefaafaaaaaaafafbbfa8%5B14%3A15%3A44%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafbabbfbfc%5B14%3A17%3A09%5D&c=%7B147%7Deeeeeebbeaeeafbafefaafaaaaaaafbabbfb8%5B14%3A17%3A09%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafbbbbfefc%5B14%3A18%3A34%5D&c=%7B151%7Deeeeeebbeaeeafbafefaafaaaaaaafbb77fdf8%5B14%3A18%3A34%5D&c=%7B147%7Deeeeeebbeaeeafbafefaafaaaaaaafbbbbfe8%5B14%3A18%3A34%5D&f=8h8h%208h8h%208h8h%208h8h%208h8h%208h8h%208h8h%208h%20RAIN%3F8h%20CHK%3F8h8h%20TRAILER%3F8h&x=aa&cw=4) where a counter and a checksum is visible.

It's stange that there is so much data for basically one byte rain counter and one byte checksum.
You need to perform more tests, reset the battery and such to confirm the pattern.

Collect your data in a BitBench like above.

stomko11, 2022-03-01:
I took another capture, [BitBench](https://triq.net/bitbench#c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafbfbeaafc%5B15%3A46%3A24%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafbfbeaafc%5B15%3A46%3A24%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafbfbeaafc%5B15%3A46%3A24%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafbfbeaafc%5B15%3A46%3A24%5D&c=%7B146%7Deeeeeebbd5dd5f75fdf55f5555555f7f7d550%5B15%3A46%3A24%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafeabeabfc%5B15%3A47%3A49%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafeabeabfc%5B15%3A47%3A49%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafeabeabfc%5B15%3A47%3A49%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaafeabeabfc%5B15%3A47%3A49%5D&c=%7B146%7Deeeedd77d5dd5f75fdf55f5555555fd57d570%5B15%3A47%3A49%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaaaaaaeebfc%5B15%3A48%3A12%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaaaaaaeebfc%5B15%3A48%3A12%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaaaaaaeebfc%5B15%3A48%3A12%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaaaaaaeebfc%5B15%3A48%3A12%5D&c=%7B147%7Deeeeeebbeaeeafbafefaafaaaaaaaaaaaeeb8%5B15%3A48%3A12%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaaaabaeeefc%5B15%3A49%3A37%5D&c=%7B151%7Deeeeeebbeaee5f75fdf55f55555555575dddf8%5B15%3A49%3A37%5D&c=%7B150%7Deeeeeebbeaee5f75fdf55f5555555556bbbbf0%5B15%3A49%3A37%5D&c=%7B151%7Deeeeeebbeaee5f75fdf55f55555555575dddf8%5B15%3A49%3A37%5D&c=%7B147%7Deeeeeebbeaeeafbafefaafaaaaaaaaabaeee8%5B15%3A49%3A37%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaaaabaeeefc%5B15%3A51%3A02%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaaaabaeeefc%5B15%3A51%3A02%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaaaabaeeefc%5B15%3A51%3A02%5D&c=%7B152%7Deeeeeebbeaeeafbafefaafaaaaaaaaabaeeefc%5B15%3A51%3A02%5D&c=%7B147%7Deeeeeebbeaeeafbafefaafaaaaaaaaabaeee8%5B15%3A51%3A02%5D&c=%7B109%7Db6d55a6cb57ba5d22952928dafe0%5B15%3A51%3A20%5D&c=%7B117%7Ddb6d5baa9767ba9d52954aa956cfe0%5B15%3A51%3A20%5D&c=%7B106%7Ddab66ead9767ba5c128852167f0%5B15%3A51%3A20%5D&c=%7B84%7Db552d28d3b0c0040052fc%5B15%3A51%3A20%5D&c=%7B76%7Daaa58a34ec300000294%5B15%3A51%3A20%5D&c=%7B82%7Daaa5451a761800008abf0%5B15%3A52%3A45%5D&c=%7B83%7Daaa5451a76180400455f8%5B15%3A52%3A45%5D&c=%7B81%7Daaa58a34ec300001157e0%5B15%3A52%3A45%5D&c=%7B81%7Daaa58a34ec300001157e0%5B15%3A52%3A45%5D&c=%7B76%7Daaa58a34ec300001154%5B15%3A52%3A45%5D&f=8h8h%208h8h%208h8h%208h8h%208h8h%208h8h%208h8h%208h%20RAIN%3F8h%20CHK%3F8h8h%20TRAILER%3F8h&x=aa&cw=4)
Raw output here:
[emos_rain3.txt](https://github.com/merbanan/rtl_433/files/8162508/emos_rain3.txt)

Between `15:47:49` and `15:48:12` I changed the tip and reinserted battery. That's why it is not after 1:25, but it sends report when battery is inserted.
Between `15:49:37` and `15:51:02` I did no change to tip.
Between `15:51:02` and `15:51:20` I just pulled the battery (again that's why early report at 15:51:20).
Besides those ocassions described above, I did always change tip between the reports.

stomko11, 2022-03-03:
It seems that pattern is changing after pulling battery (reports since 15:51:20). I am not able to move forward. Any help would be appreciated.

zuckschwerdt, 2022-03-03:
I suspect the transmission is no picked up reliably or the demod fails. The output isn't really plausible.
Perhaps try to optimize the reception and record a set of cu8 samples. This isn't ideal, but it seems we can't trust the bit slicing here. https://triq.org/rtl_433/ANALYZE.html

stomko11, 2022-03-03:
I will get another USB stick and try to gather more samples. I appreciate the support you are providing. This is great project and wonderful community.

stomko11, 2022-03-23:
I did get the additional tuner and tried to follow recommended article, I just don't know what to do with it. I did take few cu8 samples, analyzed them, but I don't actually know what am I looking at or what am I searching for. Or what should I see here. Example cu8 file is attached, together with generated sample based on article. Can someone help me out?
[EMOS.zip](https://github.com/merbanan/rtl_433/files/8336134/EMOS.zip)

zuckschwerdt, 2022-03-23:
At least the first sample (g001) works. The others (6,7) are too strong and swamp the receiver.
This will decode clean signals: `rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72,invert'`
And gets you `aaa58a34ec300000298` repeated a few times.

You would now need to run that and optimize distance or position to always reliably get those codes.
We can then look at the codes for different known readings, e.g. a slow ramp of temperature, to figure out how to pick the codes apart.

stomko11, 2022-03-23:
Well this is only rain gauge, does not measure anything else, only adds +0,7mm of rainfall with each swing,
I will follow above recommendations and post back.

stomko11, 2022-03-23:
Attached are samples that I took
```
21:25:47
21:27:12
6 swings
21:28:37
1 swing
21:30:02
21:31:27
5 swings + removed and reinserted battery
21:31:54
21:33:19
2 swings
21:34:44
removed and reinserted battery
21:35:00
21:36:25
```

Is the captured output any helpful?

[rain_gauge_1.txt](https://github.com/merbanan/rtl_433/files/8336870/rain_gauge_1.txt)
.

zuckschwerdt, 2022-03-24:
The received codes look very clean and regular, good reception!

You would now collect and format the codes and your notes like this:
```
{73}aaa58a34ec300000298[2022-03-23 21:25:47]
{73}aaa58a34ec300000298[2022-03-23 21:27:12]
{73}aaa58a34ec3000062f8[2022-03-23 21:28:37 +6 swings]
{73}aaa58a34ec300007308[2022-03-23 21:30:02 +1 swings]
{73}aaa58a34ec300007308[2022-03-23 21:31:27]
{73}aaa58a34ec300000298[2022-03-23 21:31:54 +5 swings + removed and reinserted battery]
{73}aaa58a34ec300000298[2022-03-23 21:33:19]
{73}aaa58a34ec3000022b8[2022-03-23 21:34:44 +2 swings]
{73}aaa58a34ec300000298[2022-03-23 21:35:00 removed and reinserted battery]
{73}aaa58a34ec300000298[2022-03-23 21:36:25]
```
and put that [in a BitBench](https://triq.net/bitbench#c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A25%3A47%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A27%3A12%5D&c=%7B73%7Daaa58a34ec3000062f8%5B2022-03-23%2021%3A28%3A37%20%2B6%20swings%5D&c=%7B73%7Daaa58a34ec300007308%5B2022-03-23%2021%3A30%3A02%20%2B1%20swings%5D&c=%7B73%7Daaa58a34ec300007308%5B2022-03-23%2021%3A31%3A27%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A31%3A54%20%2B5%20swings%20%2B%20removed%20and%20reinserted%20battery%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A33%3A19%5D&c=%7B73%7Daaa58a34ec3000022b8%5B2022-03-23%2021%3A34%3A44%20%2B2%20swings%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A35%3A00%20removed%20and%20reinserted%20battery%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A36%3A25%5D&f=8h8h%208h8h%208h8h%20RAIN%3A8h8h%20CHK%3A8h%208x%20&cf=ADD&cl=64&cw=8).

So we know the last byte is a checksum (addition over 8 bytes) and the two bytes before that are a counter.
I guess the start is part preamble, sync word, ID, and status. Not sure, but too long for just ID.
Can you perhaps insert known dead batteries or use a variable voltage power supply to see if some bits before the counter change (battery_low flag).

AutomGuy, 2022-03-24:
Good morning guys,
@stomko11 please take a look into that [emos_e6016.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/emos_e6016.c) and you will see this
Format String
MODEL?:8h8h8h ID?:8d BAT?2b DT:6d-4d-5dT5d:6d:6d CH:2d TEMP:12d HUM?8d WSPEED:8d WINDIR:4d ?4h CHK:8h REPEAT:8h
Decode Example
MODEL?:aaa583 ID?:255 BAT?10 DT:21-05-21T07:49:35 CH:0 TEMP:0201 HUM?037 WSPEED:000 WINDIR:10 ?2 CHK:c7 REPEAT:00
Your Device
Format String
MODEL?:8h8h8h ID?:8h BAT?:2b ?:6h8h RAIN:8h8h CHK:8h 8x 
Decode Example
MODEL?:aaa58a ID?:34 BAT?:11 ?:2c30 RAIN:0000 CHK:29

[Bitbench for That](https://triq.net/bitbench#c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A25%3A47%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A27%3A12%5D&c=%7B73%7Daaa58a34ec3000062f8%5B2022-03-23%2021%3A28%3A37%20%2B6%20swings%5D&c=%7B73%7Daaa58a34ec300007308%5B2022-03-23%2021%3A30%3A02%20%2B1%20swings%5D&c=%7B73%7Daaa58a34ec300007308%5B2022-03-23%2021%3A31%3A27%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A31%3A54%20%2B5%20swings%20%2B%20removed%20and%20reinserted%20battery%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A33%3A19%5D&c=%7B73%7Daaa58a34ec3000022b8%5B2022-03-23%2021%3A34%3A44%20%2B2%20swings%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A35%3A00%20removed%20and%20reinserted%20battery%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A36%3A25%5D&f=MODEL%3F%3A8h8h8h%20ID%3F%3A8h%20BAT%3F%3A2b%20%3F%3A6h8h%20RAIN%3A8h8h%20CHK%3A8h%208x%20&cf=ADD&cl=64&cw=8) 
The bytes of my wind sensor were inverted

AutomGuy, 2022-03-24:
I gues now it is easy to copy the file in the post before and modified it :+1: :)

stomko11, 2022-03-24:
Here is the capture and this [BitBench](https://triq.net/bitbench#c=%7B73%7Daaa58a34ec300000298%5Boriginal%20new%203%2C183V%5D&c=%7B73%7Daaa58a34ec300000298%5Boriginal%20new%203%2C183V%5D&c=%7B73%7Daaa58a34ec300000298%5Bfirst%20set%20of%20rechargable%20batteries%2C%202%2C669V%5D&c=%7B73%7Daaa58a34ec300000298%5Bfirst%20set%20of%20rechargable%20batteries%2C%202%2C669V%5D&c=%7B73%7Daaa58ab4ec300000a98%5Bsecond%20set%20of%20rechargable%20batteries%2C%202%2C511V%5D&c=%7B73%7Daaa58ab4ec300000a98%5Bsecond%20set%20of%20rechargable%20batteries%2C%202%2C511V%5D&c=%7B73%7Daaa58a34ec300000298%5Bthird%20set%20of%20alkaline%20batteries%2C%202%2C972V%5D&c=%7B73%7Daaa58a34ec300000298%5Bthird%20set%20of%20alkaline%20batteries%2C%202%2C972V%5D&f=8h8h%20BATTERY_LOW%3A8h8h%208h8h%20RAIN%3A8h8h%20CHK%3A8h%208x%20&cf=ADD&cl=64&cw=8)

[rain_gauge_battery.txt](https://github.com/merbanan/rtl_433/files/8339865/rain_gauge_battery.txt)

AutomGuy, 2022-03-24:
@stomko11 , May be that [BitBench](https://triq.net/bitbench#c=%7B73%7Daaa58a34ec300000298%5Boriginal%20new%203%2C183V%5D&c=%7B73%7Daaa58a34ec300000298%5Boriginal%20new%203%2C183V%5D&c=%7B73%7Daaa58a34ec300000298%5Bfirst%20set%20of%20rechargable%20batteries%2C%202%2C669V%5D&c=%7B73%7Daaa58a34ec300000298%5Bfirst%20set%20of%20rechargable%20batteries%2C%202%2C669V%5D&c=%7B73%7Daaa58ab4ec300000a98%5Bsecond%20set%20of%20rechargable%20batteries%2C%202%2C511V%5D&c=%7B73%7Daaa58ab4ec300000a98%5Bsecond%20set%20of%20rechargable%20batteries%2C%202%2C511V%5D&c=%7B73%7Daaa58a34ec300000298%5Bthird%20set%20of%20alkaline%20batteries%2C%202%2C972V%5D&c=%7B73%7Daaa58a34ec300000298%5Bthird%20set%20of%20alkaline%20batteries%2C%202%2C972V%5D&f=MODEL%3F%3A8h8h8h%20BAT%3F%3A1b%20ID%3F%3A8h%20%3F%3A7h8h%20RAIN%3A8h8h%20CHK%3A8h%208x%20&cf=ADD&cl=64&cw=8)

zuckschwerdt, 2022-03-24:
This is confusing. So the decoder for "EMOS E6016 weatherstation with DCF77" is for a sensor to the "E6016 station", but we don't have a model number for the sensor. And this rain gauge is also a sensor for that station, but again with no own model number, right?

stomko11, 2022-03-24:
Right. This one on the sticker just says "EMOS, type: E6016", SN: 401502
As displayed [here](https://www.amazon.de/-/en/E6016-Professional-Wireless-Touchscreen-Forecast/dp/B08443WR8R?th=1) there are basically 3 things - indoor unit displaying data, then station (outdoor temp, wind speed, wind direction, humidity, ...) and additional rain gauge. But this acts as standalone sensor as far as rtl_433 is concerned.

AutomGuy, 2022-03-24:
So i explain what i ment.

That is the pream before inverting the bytes at EMOS6016 Windsensor
aaa583 -> we can Identify the sensor as Windsensor 
Here we check the pream of the inverted bytes
    // model check 55 5a 7c
    if (b[0] != 0x55 || b[1] != 0x5a || b[2] != 0x7c) {
        decoder_log(decoder, 2, __func__, "Model check fail");
        return DECODE_ABORT_EARLY;
    }
the result of that are the name
  data_t *data = data_make(
            "model",            "",                 DATA_STRING, "EMOS-E6016",

the only thing what we have to do is
the pream of the rain gauge sensor is
aaa58a -> the diff is the a at last position.
    // model check aa a5 8a
    if (b[0] != 0xaa || b[1] != 0xa5 || b[2] != 0x8a) {
        decoder_log(decoder, 2, __func__, "Model check fail");
        return DECODE_ABORT_EARLY;
    }

the result of that are the name
  data_t *data = data_make(
            "model",            "",                 DATA_STRING, "EMOS-E6016_RAIN",

i hope now it is clear what i ment.
Best Regards

AutomGuy, 2022-03-24:
Has the rain gauge sensor a Channel Switch?

stomko11, 2022-03-24:
no, it does not have channel switch.

AutomGuy, 2022-03-24:
@stomko11, So i have had a look into the manual of that rain gauge sensors.
The technical data say the range og that sensor is "precipitation sensor measurement range: 0 to 2,999 mm"
That means that is the max.
{73}aaa58a34ec300bb7298[2022-03-23 21:25:47] -> the checksum is wrong (i change the rain bytes by Hand) :)

That is the format string
MODEL?:8h8h8h ID?:8h BAT?:2b ?:6h8h4h RAIN:12d CHK:8h 8x 

and here the link to the [BitBench](https://triq.net/bitbench#c=%7B73%7Daaa58a34ec300bb7298%5B2022-03-23%2021%3A25%3A47%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A27%3A12%5D&c=%7B73%7Daaa58a34ec3000062f8%5B2022-03-23%2021%3A28%3A37%20%2B6%20swings%5D&c=%7B73%7Daaa58a34ec300007308%5B2022-03-23%2021%3A30%3A02%20%2B1%20swings%5D&c=%7B73%7Daaa58a34ec300007308%5B2022-03-23%2021%3A31%3A27%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A31%3A54%20%2B5%20swings%20%2B%20removed%20and%20reinserted%20battery%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A33%3A19%5D&c=%7B73%7Daaa58a34ec3000022b8%5B2022-03-23%2021%3A34%3A44%20%2B2%20swings%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A35%3A00%20removed%20and%20reinserted%20battery%5D&c=%7B73%7Daaa58a34ec300000298%5B2022-03-23%2021%3A36%3A25%5D&f=MODEL%3F%3A8h8h8h%20ID%3F%3A8h%20BAT%3F%3A2b%20%3F%3A6h8h4h%20RAIN%3A12d%20CHK%3A8h%208x%20&cf=ADD&cl=64&cw=8)
May be with all these informations you can create a decoder. :)

AutomGuy, 2022-03-24:
I have overlooked that each swing are 0.7mm.
i have corrected this. 
[emos_e6016_rain.zip](https://github.com/merbanan/rtl_433/files/8342395/emos_e6016_rain.zip)

stomko11, 2022-03-24:
Appreciate your support on this, but I am pretty lost. So how do I actually test this? I did compile the code on my raspberry where I am testing this, by following the tutorial. But I am not quite sure I understand where to put this or what to do. I am new at all this, sorry.

stomko11, 2022-03-24:
OK first of all, thanks. I did check the file, seems that there was a typo on line 71
```
    float rain_mm  = rain_row * 0.7;
```
I did change that to `rain_raw` and corrected file is attached. However I am not able to receive anything. I did check the PR when EMOS station support was added, and basically the important part it decoder (attached) and references to it. I first tried to modify those files and add additional decoder, but I was unsuccessful. So I cloned the master repository and just renamed `emos_e6016_rain.c` to `emos_e6016.c` and compiled it. But when I run it, I don't decode anything. When I capture raw signal, there are signals sent by gauge, but when I just run `rtl_433` without any additional arguments (or specify protocol 214 in this case), I don't get any good decoding output.
[emos_e6016_rain.zip](https://github.com/merbanan/rtl_433/files/8344888/emos_e6016_rain.zip)

AutomGuy, 2022-03-25:
Morning, that was a mistake from me, it should be "rain_raw".
you have to download or clone the entire project.
then find in the directory structure "src/device/"  copy into that directory the decoder file (emos_e6016_rain.c)
then goto directory /src find "CMakeList.txt" add in this file  under "add_library("  
that  "devices/emos_e6016_rain.c" below the other  "emos_e6016".

then goto "/include" find that file "rtl_433_devices.h" add in this file under  "#define DEVICES "at last "DECL(emos_e6016_rain) \"
now you have to compile the project. And when no errors occurs
And you start "rtl_433 -R help" should the output a list of all decoders.
And the Last output could look like this "[21x] EMOS E6016 Rain Gauge"

AutomGuy, 2022-03-25:
the decoder write some logs
with this line of code
decoder_log(decoder, 2, __func__, "Repeated row fail");
@zuckschwerdt  -> christian may be you can tell us how we can take a look into the logs?
Many Thx

zuckschwerdt, 2022-03-25:
The number there is the log level. Use 0 to log always, 1 to log on `-v` and 2 to log on `-vv`

AutomGuy, 2022-03-25:
Ok understand, but is a log file available or print the decoder to stdout?

stomko11, 2022-03-25:
This is what I get when I followed the instructions above
```
[ 91%] Building C object src/CMakeFiles/r_433.dir/devices/yale_hsa.c.o
[ 92%] Linking C static library libr_433.a
[ 92%] Built target r_433
Scanning dependencies of target rtl_433
[ 92%] Building C object src/CMakeFiles/rtl_433.dir/rtl_433.c.o
[ 92%] Linking C executable rtl_433
/usr/bin/ld: libr_433.a(r_api.c.o): in function `r_init_cfg':
r_api.c:(.text+0x14c4): undefined reference to `emos_e6016_rain'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/rtl_433.dir/build.make:108: src/rtl_433] Error 1
make[1]: *** [CMakeFiles/Makefile2:1031: src/CMakeFiles/rtl_433.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
```

stomko11, 2022-03-25:
OK I figured that part out, issue issue was on line 97 of decoder. Corrected one is attached.
[emos_e6016_rain-1.zip](https://github.com/merbanan/rtl_433/files/8354059/emos_e6016_rain-1.zip)

However when I run it with `-vv` I get following when I insert batteries:
```
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {381} 55 55 2c 51 a7 61 80 00 01 4d 55 52 c5 1a 76 18 00 00 14 d5 55 2c 51 a7 61 80 00 01 4d 55 52 c5 1a 76 18 00 00 14 d5 55 2c 51 a7 61 80 00 01 48
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {381} 55 55 2c 51 a7 61 80 00 01 4d 55 52 c5 1a 76 18 00 00 14 d5 55 2c 51 a7 61 80 00 01 4d 55 52 c5 1a 76 18 00 00 14 d5 55 2c 51 a7 61 80 00 01 48
ambient_weather_decode: Checksum error, expected: 00 calculated: 7b: {48} 40 00 00 00 00 00 : 01000000 00000000 00000000 00000000 00000000 00000000
ambient_weather_decode: Checksum error, expected: 00 calculated: 7b: {48} 40 00 00 00 00 00 : 01000000 00000000 00000000 00000000 00000000 00000000
ambient_weather_decode: Checksum error, expected: 00 calculated: 7b: {48} 40 00 00 00 00 00 : 01000000 00000000 00000000 00000000 00000000 00000000
ambient_weather_decode: Checksum error, expected: 00 calculated: 7b: {48} 40 00 00 00 00 00 : 01000000 00000000 00000000 00000000 00000000 00000000
x10_rf_callback: DECODE_ABORT_LENGTH, Received message length=370
wt450_callback: wrong size of bit per row 10
x10_sec_callback: DECODE_FAIL_SANITY, b0=c0 b1=00 b2=00 b3=00
philips_aj7010_decode: wrong number of rows (6)
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
norgo_decode: wrong size of bit per row 1
ert_idm_decode: rows=1, row0 len=744
ert_idm_decode: sync_index=744
ert_netidm_decode: rows=1, row0 len=744
ert_netidm_decode: sync_index=744
markisol_decode: bits_per_row[0] = 1
markisol_decode: bits_per_row[0] = 371
```

If I register only protocol 216, I get no output:
```
pi@raspberrypi:~/rtl_433/build $ sudo rtl_433 -vv
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Reading conf from "/etc/rtl_433/rtl_433.conf".
Registered 1 out of 216 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Fitipower FC0012 tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
```

AutomGuy, 2022-03-26:
it seams to be a lot of devices in your enviroment.
try rtl_433 -R 0 - R YOUR_DECODER_NO -vv
with -R 0 you switch off all decoder , with the next -R you switch on only your decoder
may be that helps to analyze.

AutomGuy, 2022-03-26:
@zuckschwerdt , christian is that call possibel rtl_433 -R0 -R SPECIFIC_DECODER g001_433.92M_250k.cu8 ?
to test the decoder in case of you have not the device physicly
Thx

stomko11, 2022-03-26:
> it seams to be a lot of devices in your enviroment. try rtl_433 -R 0 - R YOUR_DECODER_NO -vv with -R 0 you switch off all decoder , with the next -R you switch on only your decoder may be that helps to analyze.

It is not devices in neighborhood. This pops up when I insert batteries to rain gauge. I tried several times, shortly after each attempt. Every time I do it, bunch of these logs appear. It almost look to me like the decoder or message that are sent look similar to these protocols, so it tries to decode them but fails. When I enable only protocol 216 (rain gauge), absolutely nothing happens on the console output. So again, it looks to me that decoder is not decoding these messages thus not throwing anything on the console output.

zuckschwerdt, 2022-03-26:
> Ok understand, but is a log file available or print the decoder to stdout?

Log messages currently go to stderr always. More is planned but that will take a while.

zuckschwerdt, 2022-03-26:
> possible `rtl_433 -R0 -R SPECIFIC_DECODER g001_433.92M_250k.cu8` ?

Yes, that is recommended to test with sample files.

If you have a code you can also use:

`rtl_433 -R SPECIFIC_DECODER -y '{73}aaa58a34ec300000298'`

AutomGuy, 2022-03-28:
@zuckschwerdt Christian that is allready the inverted code '{73}aaa58a34ec300000298' but not the raw code isn't it?
@stomko11 how many repeats do get when you use the flex decoder?
i have compiled that  at my machine made some mods at the decoder and get that.
i have called with 3 repeats -> this are the rawcodes
rtl_433 -R 215 -y '{73}555a75cb13cfffffd60 {73}555a75cb13cfffffd60 {73}555a75cb13cfffffd60'
and this is what rtl_433 returned
time      : 2022-03-28 14:34:33                                                                                                                                                          
model     : EMOS-E6016_RAIN                        House Code: 52
Battery_OK: 1            Rain_mm   : 0.000         Integrity : CHECKSUM
IMHO looks good

zuckschwerdt, 2022-03-28:
> not the raw code isn't it?

Inverted, yes. A random (bad) example. Your 3 repeats command looks good!

AutomGuy, 2022-03-28:
here a sample with 6 swing
rtl_433 -R 215 -y '{73}555a75cb13cffff9d00 {73}555a75cb13cffff9d00 {73}555a75cb13cffff9d00'
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 15:10:07                                                                                                                                                          
model     : EMOS-E6016_RAIN                        House Code: 52
Battery_OK: 1            Rain_mm   : 4.2           Integrity : CHECKSUM

now 7 swing
rtl_433 -R 215 -y '{73}555a75cb13cffff8cf0 {73}555a75cb13cffff8cf0 {73}555a75cb13cffff8cf0'
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 15:13:58                                                                                                                                                          
model     : EMOS-E6016_RAIN                        House Code: 52
Battery_OK: 1            Rain_mm   : 4.9           Integrity : CHECKSUM

but with the .cu8 files from above i can not reproduce this results

stomko11, 2022-03-28:
So the one I compiled, here is the output:
```
pi@raspberrypi:~ $ rtl_433
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 185 out of 216 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-216 ]
Detached kernel driver
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 15:36:07
model     : EMOS-E6016   House Code: 197
Channel   : 1            Battery_OK: 1             Temperature_C: 20.9       Humidity  : 30            WindSpeed m_s: 3.0        Wind direction: 270.0     Radio Clock: 2022-03-28T15:36:05                    Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 15:37:04
model     : EMOS-E6016   House Code: 197
Channel   : 1            Battery_OK: 1             Temperature_C: 21.2       Humidity  : 31            WindSpeed m_s: 5.0        Wind direction: 270.0     Radio Clock: 2022-03-28T15:37:03                    Integrity : CHECKSUM

```

I am not limiting protocol, so decoding all, 216 is supposed to be the rain gauge. 
As visible, I am seeing my Emos station reporting, but no reports from rain gauge. I did turn on indoor unit, just to verify, and I did a swing between each report, indoor unit shows 2.1mm of rainfall, but nothing visible on the output above.

stomko11, 2022-03-28:
@AutomGuy 
```
nice temp where you are or ist it inside :) ?
```
That's outside! :)

```
pleas try this rtl_433 -R 216 -y '{73}555a75cb13cffff8cf0 {73}555a75cb13cffff8cf0 {73}555a75cb13cffff8cf0'
```
output:
```
pi@raspberrypi:~ $ rtl_433 -R 216 -y '{73}555a75cb13cffff8cf0 {73}555a75cb13cffff8cf0 {73}555a75cb13cffff8cf0'
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 216 device decoding protocols [ 216 ]
```

```
rtl_433 -R 0 -X 'n=EMOS_RAIN,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72,invert'
```
output:
```
pi@raspberrypi:~ $ rtl_433 -R 0 -X 'n=EMOS_RAIN,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72,invert'
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 216 device decoding protocols [ ]
Detached kernel driver
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 16:01:46
model     : EMOS_RAIN    count     : 1             num_rows  : 1             rows      :
len       : 726          data      : aaa583c5963e401bc0d61d0162dc00d552c1e2cb1f200de06b0e80b16e00eaa960f1658f9006f035874058b700b554b078b2c7c803781ac3a02c5b807aaa583c5963e401bc0d61d0162dc04d552c1e2cb1f200de06b0e80b16e02c
codes     : {726}aaa583c5963e401bc0d61d0162dc00d552c1e2cb1f200de06b0e80b16e00eaa960f1658f9006f035874058b700b554b078b2c7c803781ac3a02c5b807aaa583c5963e401bc0d61d0162dc04d552c1e2cb1f200de06b0e80b16e02c
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 16:02:43
model     : EMOS_RAIN    count     : 1             num_rows  : 1             rows      :
len       : 726          data      : aaa583c5963e402a40d61d01a2ab00d552c1e2cb1f2015206b0e80d15580eaa960f1658f900a9035874068aac0b554b078b2c7c805481ac3a03455607aaa583c5963e402a40d61d01a2ab04d552c1e2cb1f2015206b0e80d15582c
codes     : {726}aaa583c5963e402a40d61d01a2ab00d552c1e2cb1f2015206b0e80d15580eaa960f1658f900a9035874068aac0b554b078b2c7c805481ac3a03455607aaa583c5963e402a40d61d01a2ab04d552c1e2cb1f2015206b0e80d15582c
```
This was 2 swings, 1 at a time

AutomGuy, 2022-03-28:
@stomko11 and this rtl_433 -R 0 -X 'n=EMOS_RAIN,m=OOK_PWM,s=280,l=796,g=0,t=0,y=1836,bits>=72' please
and post

AutomGuy, 2022-03-28:
have you used the decoder file what i have posted for approx. a hour?

stomko11, 2022-03-28:
I compiled now with that decoder. Here is one by one:
```
pi@raspberrypi:~/rtl_433/build $ rtl_433 -R 0 -X 'n=EMOS_RAIN,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72'
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 216 device decoding protocols [ ]
Detached kernel driver
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 16:57:49
model     : EMOS_RAIN    count     : 1             num_rows  : 1             rows      :
len       : 726          data      : 555a7c3a69c1bc63bf2ae3fa5dddff2aad3e1d34e0de31df9571fd2eeeff15569f0e9a706f18efcab8fe97777f4aab4f874d38378c77e55c7f4bbbbf8555a7c3a69c1bc63bf2ae3fa5dddfb2aad3e1d34e0de31df9571fd2eeefd0
codes     : {726}555a7c3a69c1bc63bf2ae3fa5dddff2aad3e1d34e0de31df9571fd2eeeff15569f0e9a706f18efcab8fe97777f4aab4f874d38378c77e55c7f4bbbbf8555a7c3a69c1bc63bf2ae3fa5dddfb2aad3e1d34e0de31df9571fd2eeefd0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 16:58:46
model     : EMOS_RAIN    count     : 1             num_rows  : 1             rows      :
len       : 726          data      : 555a7c3a69c1bc543f29e3fb5d4eff2aad3e1d34e0de2a1f94f1fdaea77f15569f0e9a706f150fca78fed753bf4aab4f874d38378a87e53c7f6ba9df8555a7c3a69c1bc543f29e3fb5d4efb2aad3e1d34e0de2a1f94f1fdaea77d0
codes     : {726}555a7c3a69c1bc543f29e3fb5d4eff2aad3e1d34e0de2a1f94f1fdaea77f15569f0e9a706f150fca78fed753bf4aab4f874d38378a87e53c7f6ba9df8555a7c3a69c1bc543f29e3fb5d4efb2aad3e1d34e0de2a1f94f1fdaea77d0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 16:59:43
model     : EMOS_RAIN    count     : 1             num_rows  : 1             rows      :
len       : 726          data      : 555a7c3a69c1bc45bf27e2fb7ddcff2aad3e1d34e0de22df93f17dbeee7f15569f0e9a706f116fc9f8bedf773f4aab4f874d383788b7e4fc5f6fbb9f8555a7c3a69c1bc45bf27e2fb7ddcfb2aad3e1d34e0de22df93f17dbeee7d0
codes     : {726}555a7c3a69c1bc45bf27e2fb7ddcff2aad3e1d34e0de22df93f17dbeee7f15569f0e9a706f116fc9f8bedf773f4aab4f874d383788b7e4fc5f6fbb9f8555a7c3a69c1bc45bf27e2fb7ddcfb2aad3e1d34e0de22df93f17dbeee7d0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 17:00:40
model     : EMOS_RAIN    count     : 1             num_rows  : 1             rows      :
len       : 726          data      : 555a7c3a69c1bbf63f26e3fc6dfdff2aad3e1d34e0ddfb1f9371fe36feff15569f0e9a706efd8fc9b8ff1b7f7f4aab4f874d38377ec7e4dc7f8dbfbf8555a7c3a69c1bbf63f26e3fc6dfdfb2aad3e1d34e0ddfb1f9371fe36fefd0
codes     : {726}555a7c3a69c1bbf63f26e3fc6dfdff2aad3e1d34e0ddfb1f9371fe36feff15569f0e9a706efd8fc9b8ff1b7f7f4aab4f874d38377ec7e4dc7f8dbfbf8555a7c3a69c1bbf63f26e3fc6dfdfb2aad3e1d34e0ddfb1f9371fe36fefd0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-28 17:01:37
model     : EMOS_RAIN    count     : 1             num_rows  : 1             rows      :
len       : 726          data      : 555a7c3a69c1bbe6bf24e4fc3d3cff2aad3e1d34e0ddf35f92727e1e9e7f15569f0e9a706ef9afc9393f0f4f3f4aab4f874d38377cd7e49c9f87a79f8555a7c3a69c1bbe6bf24e4fc3d3cfb2aad3e1d34e0ddf35f92727e1e9e7d0
codes     : {726}555a7c3a69c1bbe6bf24e4fc3d3cff2aad3e1d34e0ddf35f92727e1e9e7f15569f0e9a706ef9afc9393f0f4f3f4aab4f874d38377cd7e49c9f87a79f8555a7c3a69c1bbe6bf24e4fc3d3cfb2aad3e1d34e0ddf35f92727e1e9e7d0
```
```
pi@raspberrypi:~/rtl_433/build $ rtl_433 -R 0 -X 'n=EMOS_RAIN,m=OOK_PWM,s=280,l=796,g=0,t=0,y=1836,bits>=72'
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Bad flex spec, missing reset limit!
Use -X <spec> to add a general purpose decoder. For usage use -X help
```

Regarding rain_gauge_1.txt, I believe that was captured at the beginning with `rtl_433 -X 'n=EMOS-RAIN,m=OOK_PCM,s=500,l=500,r=1200'`

AutomGuy, 2022-03-29:
good morning at all, my ideas runs out. no -X decoder parameter works with the .cu8 files wich was posted in EMOS.zip 5 days ago.

we need the timings from -X decoder, wich produced that

time      : 2022-03-24 09:48:49
model     : name         count     : 5             num_rows  : 11            rows      :
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000290
codes     : {1}8, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000290

stomko11, 2022-03-29:
I basically followed Christian's advice: `rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72,invert'`

AutomGuy, 2022-03-29:
but when you look back with this parameters you get 726 Bytes and the first 3 Bytes tells me it is the wind sensor. when you take out the bat of the wind sensor und you try it again only with the rain sensor. 
rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72,invert'
what is the result?

AutomGuy, 2022-03-29:
my emos wind sensor has the preamble aa a5 83 and your wind sensor also otherwise the windsensor decoder doesent work.
the rain sensor has the preamble aa a5 8a i gues

AutomGuy, 2022-03-29:
the wind sensor send approx. all 61s and the rain sensor ?

stomko11, 2022-03-29:
I did not pull batteries from weather station, because every time I insert batteries to rain gauge, it sends report. So basically I can pull batteries and insert it back, it sends report. Weather station reports in periodic intervals (61s), rain gauge is 85s
So here is the output:

`rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72,invert'`
```
pi@raspberrypi:~ $ rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72,invert'
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 216 device decoding protocols [ ]
Detached kernel driver
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 09:53:40
model     : name         count     : 5             num_rows  : 11            rows      :
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000290
codes     : {1}8, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000290
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 09:53:48
model     : name         count     : 5             num_rows  : 11            rows      :
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000290
codes     : {1}8, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000290
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 09:53:54
model     : name         count     : 5             num_rows  : 11            rows      :
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000290
codes     : {1}8, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000290
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 09:53:59
model     : name         count     : 5             num_rows  : 11            rows      :
len       : 1            data      : 8,
[... 24 lines cut ...]
                = Supported device protocols =
    [01]  Silvercrest Remote Control
.
.
.
    [214]  EMOS E6016 weatherstation with DCF77
    [215]  Altronics X7064 temperature and humidity sensor
    [216]  EMOS E6016 RAIN GAUGE
```

And the last one:
```
pi@raspberrypi:~ $ rtl_433 -R 216 -y '{73}555a75cb13cffff8cf0 {73}555a75cb13cffff8cf0 {73}555a75cb13cffff8cf0'
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 216 device decoding protocols [ 216 ]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 09:56:54
model     : EMOS-E6016_RAIN                        House Code: 52
Battery_OK: 1            Rain_mm   : 4.9           Integrity : CHECKSUM
```

AutomGuy, 2022-03-29:
@zuckschwerdt , Christian what have to be here .modulation = OOK_PULSE_PWM in option -X m=OOK_PWM?
is that the same?

AutomGuy, 2022-03-29:
@stomko11 , Please try this at me this work.
rtl_433 -R 216 -y '{73}555a75cb13cffff8cf0 {1}0 {73}555a75cb13cffff8cf0 {1}0 {73}555a75cb13cffff8cf0 {1}0'
output is:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2022-03-29 10:11:37
model     : EMOS-E6016_RAIN                        House Code: 52
Battery_OK: 1            Rain_mm   : 4.9           Integrity : CHECKSUM

stomko11, 2022-03-29:
It looks the same for me:
```
pi@raspberrypi:~ $ rtl_433 -R 216 -y '{73}555a75cb13cffff8cf0 {1}0 {73}555a75cb13cffff8cf0 {1}0 {73}555a75cb13cffff8cf0 {1}0'
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 216 device decoding protocols [ 216 ]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 10:24:36
model     : EMOS-E6016_RAIN                        House Code: 52
Battery_OK: 1            Rain_mm   : 4.9           Integrity : CHECKSUM
```

AutomGuy, 2022-03-29:
@stomko11, Do this rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72,invert' without '**,inverted**' please.

stomko11, 2022-03-29:
Tried both below, wasn't sure which one you are after. Did few battery reinserts, nothing returned on the output.
```
pi@raspberrypi:~ $ rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72,invert'
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 216 device decoding protocols [ ]
Detached kernel driver
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
^CSignal caught, exiting!
Reattached kernel driver
```
```
pi@raspberrypi:~ $ rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72'
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 216 device decoding protocols [ ]
Detached kernel driver
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
^CSignal caught, exiting!
Reattached kernel driver

```

AutomGuy, 2022-03-29:
and you have to pull out and set in the batterie (rain gauge) please i will see the output like
 _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 09:53:40
model     : name         count     : 5             num_rows  : 11            rows      :
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000290
codes     : {1}8, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {

AutomGuy, 2022-03-29:
but without inverted

stomko11, 2022-03-29:
So above is the output with exact commands I used, I am not sure which one is precisely the one you want me to execute. However in both cases, I did reinsert batteries several times, but nothing on the console output.

AutomGuy, 2022-03-29:
so that is realy strange:
you post this:
pi@raspberrypi:~ $ rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72,invert'
rtl_433 version 21.12-80-gf4b73f11 branch master at 202203221518 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 216 device decoding protocols [ ]
Detached kernel driver
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 09:53:40
model     : name         count     : 5             num_rows  : 11            rows      :
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000290
codes     : {1}8, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000290
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 09:53:48
model     : name         count     : 5             num_rows  : 11            rows      :
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000290
codes     : {1}8, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000290
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 09:53:54
model     : name         count     : 5             num_rows  : 11            rows      :
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000290
codes     : {1}8, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000290
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-03-29 09:53:59
model     : name         count     : 5             num_rows  : 11            rows      :
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000298,
len       : 1            data      : 8,
len       : 73           data      : aaa58a34ec300000290
codes     : {1}8, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000298, {1}8, {73}aaa58a34ec300000290

and actually i would like have the same post 
with this command rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=300,l=800,g=1000,r=2500,bits>=72'

i expected the same data but not inverted. output above data: aaa58a. i expected 555a75

[13 more replies]

commit 0ab9adf9 (2022-03-30): minor: Clenup Emos E6016 rain gauge (closes #2001)
closed 2022-03-30 by zuckschwerdt