merbanan/rtl_433 #2002: How to debug MQTT connection issues?

petterreinholdtsen, 2022-03-01:
I'm using rtl-433version  20.11-1 from Debian, and try to tell it to publish its findings via MQTT by running it like this:

```
rtl_433 -v -v -F "mqtt://localhost:1883,user=user,pass=password"   
```

Sadly it do not work.  The script output these MQTT related lines:

```
Publishing MQTT data to localhost port 1883
Publishing device info to MQTT topic "rtl_433/vhs/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/vhs/events".
Publishing states info to MQTT topic "rtl_433/vhs/states".
...
MQTT connect error: Connection refused
```

I am unable to figure out why it fail to connect.  Is there a trick to get it to tell me more about the failure?

zuckschwerdt, 2022-03-01:
Maybe you want mqtt://127.0.0.1:1883 because you have IPv6 but the broker doesn't use it?

petterreinholdtsen, 2022-03-01:
[Christian W. Zuckschwerdt]
> Maybe you want mqtt://127.0.0.1:1883 because you have IPv6 but the
> broker doesn't use it?

Tested, but it did not help.  Besides, IPv6 is used according to lsof:

% sudo lsof -i|grep mosq
mosquitto  1403  mosquitto    5u  IPv6  18259      0t0  TCP localhost:1883 (LISTEN)
mosquitto  1403  mosquitto    6u  IPv4  18260      0t0  TCP 192.168.225.115:1883 (LISTEN)
%

-- 
Happy hacking
Petter Reinholdtsen

zuckschwerdt, 2022-03-01:
Ah ok, so perhaps `mqtt://192.168.225.115` then because you have localhost only on IPv6.
It's always tricky to guess what `localhost` might resolve to. Best to use an IP-address.

closed 2022-03-02 by zuckschwerdt

petterreinholdtsen, 2022-03-02:
Aha.  Just me misunderstanding a message printed by rtl_433 during startup, then. :)

zuckschwerdt, 2022-03-02:
But there was also a bug with parsing IPv6 addresses ;) With 8dffa04 now e.g. `mqtt://[::1]:1883` works. But hard to say if your broker is actually available as `::1` since `localhost` could be the interface address.