merbanan/rtl_433 #2003: Suggest renaming https://github.com/merbanan/rtl_433/wiki/digitech_xc0324.README for better discoverability

robcazzaro, 2022-03-02:
Hello,

I spent the last few days figuring out how to build a decoder (for https://github.com/merbanan/rtl_433/issues/1990), and after successfully doing so, I was planning to do a writeup for others to benefit. Then I bumped into https://github.com/merbanan/rtl_433/wiki/digitech_xc0324.README, which contains everything I planned to write (and more :)

I'd like to suggest changing the title, and possibly splitting it into different parts like initial investigation, flex decoder, enlisting and building, writing the decoder, etc

I know it's a wiki, and I could make th eedits. But I'm not sure what is the "etiquette" for this project, and don't want to mess things up. I would be happy to volunteer and implement the suggested changes, if you are happy with me doing it

zuckschwerdt, 2022-03-02:
Yes a bit of heads up before changes is appreciated. But please go ahead, we've been waiting for a user with that fresh perspective and writing skills :)

robcazzaro, 2022-03-02:
I definitely have a clueless fresh perspective, not sure about the writing skills :)

Closing this issue, will edit directly (already made a minor edit to https://github.com/merbanan/rtl_433/wiki/Adding-a-new-remote-device (was still using the obsolete -t instead of -S unknown)

closed 2022-03-02 by robcazzaro