merbanan/rtl_433 #2007: help decoding Arlec Wireless Door Chime (DC685)

TomW1605, 2022-03-09:
hi i have a [wireless doorbell](https://www.bunnings.com.au/arlec-high-gloss-finish-plug-in-wireless-door-chime-twin-pack_p4062534) (because the house has no built in doorbell) and i would like to use it to trigger a HA script to notify my phone.

i have had some trouble getting the files for this one. i have gotten 4 signal files that look like they are all the same (very similar number of samples in the terminal output and the files are all the same size) but when i convert them to ook to put into https://triq.org/pdv/ they all come out different with different signals and file sizes. unfortunately getting the signal is a bit inconsistent as the antenna on the transmitter is quite small. i am also not 100% sure that these are even all from the doorbell.

can someone have a look and see if they can spot a pattern in any of these?
[doorbell.zip](https://github.com/merbanan/rtl_433/files/8213681/doorbell.zip)

TomW1605, 2022-03-09:
ok i did some more capturs and now have 5 that i am sure are all from the doorbell. 4 of them are identical in the pdv but the 5th is different. i think it is just a garbled signal so am ignoring it for now. there is a lot more data there than i would expect from a doorbell so not sure what it all is but here are the samples
[doorbell 2.zip](https://github.com/merbanan/rtl_433/files/8213693/doorbell.2.zip)

zuckschwerdt, 2022-03-09:
If the broken one is g163_433.92M_250k.cu8 then yes, there is a car going by chatting about it's tires. If you drop the cu8 in pdv you can see that most have some other transmissions in them, only g165_433.92M_250k.cu8 is really clean.

Not a problem though, a doorbell repeats the signal very many times, getting just a few right works.

The signal is PWM with 333, 666 µs and a gap limit of 900 works. So:
`-X 'n=doorbell,m=OOK_PWM,s=333,l=666,g=900,r=5500'` to get raw codes.

And since the doorbell id here is {13}8998 use this to get a robust detection of the event:
`-X 'n=doorbell,m=OOK_PWM,s=333,l=666,g=900,r=5500,bits=13,match={13}8998,countonly'`

TomW1605, 2022-03-09:
ok i looked into it a bit more and reset the code by removing the transmitter battery. it looks like the first 4 bits are the code and the rest is just `89 98` repeated 97 times. using the `match={13}8998` didnt work with the new code. here are some samples with the new code. when i change it to `match={D2}8998` it worked with the new code
[doorbell 3.zip](https://github.com/merbanan/rtl_433/files/8213899/doorbell.3.zip)

TomW1605, 2022-03-09:
oh i also changed it to `l=659,g=782` because that is what the pdv gave me for the samples with the new code

zuckschwerdt, 2022-03-09:
Resetting the battery changes the code. Use the first -X i gave to figure out the current code, then change the second -X to use that code in the match.

Observe the `{13}e908` in your new set of samples. Ignore the random broken codes. That's what you want to `match`.

Your change of `l` and `g` are not needed. Fine for `l` but `g` will be somewhat tight. The `{D2}` is nonsense.

TomW1605, 2022-03-09:
why do you say the `D2` is nonsense? isnt that the new code? when i try with the `13` i just get unknown signals but when i switch it to `D2` i get an actual event

TomW1605, 2022-03-09:
also how would i get this into HA?

zuckschwerdt, 2022-03-09:
> why do you say the `D2` is nonsense?

A `{ }` is length indication in bits, decimal. The `{D2}` or `{WHATEVER}` is not valid, thus `0`. And a match of length `0` matches trivially always.

TomW1605, 2022-03-09:
ok so what should i change to make it match the new code?

zuckschwerdt, 2022-03-09:
> ok so what should i change to make it match the new code?

follow the two steps in my first comment.

As for HA, most direct approach is to add the `-F syslog` output, then adapt and run this script https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_custom.py to execute any desired command.

Basically
```
            if "model" in data and data["model"] == "doorbell":
                somehow_notify_the_phone()
```

zuckschwerdt, 2022-03-10:
`rtl_433 -X 'n=doorbell,m=OOK_PWM,s=333,l=666,g=900,r=5500' doorbell.2/*.cu8`:
```
codes     : {12}131, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {13}8988, {5}89
```

`rtl_433 -X 'n=doorbell,m=OOK_PWM,s=333,l=666,g=900,r=5500' doorbell.3/*.cu8`:
```
codes     : {12}d21, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {13}e908, {5}e9
```

The code is the thing that repeats reliably. The other things are garbage (reception at the start and the end of a signal is bad/broken).

But just to be very clear, this was the code your doorbell send before the battery change: --> **`{13}8988`** <--, and now, after the battery reset the code your doorbell will send in a fast loop as long as the button is held down is: --> **`{13}e908`** <--

TomW1605, 2022-03-11:
oh ok its the bit that repeats. honestly i didnt even notice that that part had changed. thanks for your help. just hope the battry dosnt die soon so i have to re capture the code

TomW1605, 2024-06-12:
so, im back to this and its kind of a pain to redo every time the battery is replaced (especially when others in the house replace the battery as the first thing they do when it dosnt work once and dont think to repair the receivers)

is there a way i can do the flex decoder so that it will be treated like a device and send the code as part of the mqtt topic? that way i only need to adjust home assistant when the battries are replaced. i also have a second transmitter that has the same timing charicteristics so having them be set in HA rather than needing 2 flex decoders would be nice

zuckschwerdt, 2024-06-12:
Maybe remove `,match=...` and add `,repeats=10` or such.

TomW1605, 2024-06-12:
that sends the whole row data to mqtt, it dosnt process the the code as a device id like normal devices do. it will work but makes the processing on the HA side a lot harder

zuckschwerdt, 2024-06-12:
`,get=@1:{12}:id`

TomW1605, 2024-06-12:
thanks, that worked (had to change trhe 12 to 13 though)

gdt, 2024-06-12:
@TomW1605 Are you going to submit a PR so that your new and improved flex decoder, with comments about how to use it, is in the source tree?

TomW1605, 2024-06-12:
i can i guess but i only have 2 examples of the device and they are technically different models (Arlec DC685 and Arlec DC681) so im not sure my timings are correct for every device of these models or other models in the same range

also, these are not sensor devices, so they dont really send back any actual data. the mere fact that it is sending is all the data you get. so im not sure they really fit as devices in the actual code base.

gdt, 2024-06-12:
You asked for help which is a declaration that they fit!

But seriously, caveats about two devices belongs in a comment in the decoder, and we have no rule about "sensor", it is that we decode things that send (relatively simple) signals.  The real questions are:
  - would a person with this device be better off starting with what you ended up with, or from nothing?
  - is the presence of  a decoder config in conf/ more clutter to the world compared to how useful it is (taking the existing list  as a clue)?

It seems obvious to me the answers are yes and no.

TomW1605, 2024-06-12:
fair enough, i dont have time to do it now. ill look at doing it tomorrow morning or evening. for now ill stick the decoder here in case others want it now or i forget

`decoder name=Arlec-DC68x,modulation=OOK_PWM,short=333,long=666,gap=900,reset=5500,bits=13,get=@1:{13}:id,repeats>=3,unique`

zuckschwerdt, 2024-06-12:
Should be `get=@0:...` to start at the first (0th) bit.

TomW1605, 2024-06-12:
using 1 works, ill test with 0 but it will have to wair till tomorrow. everyone is in bed now