merbanan/rtl_433 #2008: Blueline power meter reader impulses accurate?

bkenobi, 2022-03-09:
I am using a Black & Decker EM100B (OEM version of the Blueline BL28000).  When I look at the kW data derived from the gap data all looks correct.  When I look at the kWh data derived from the impulses data, the results seem ok.  If I perform a derivative of kWh and compare the resulting kW.derivative versus kW in grafana the data seems to overlay reasonably well.

[image]

However, it appears that in my case anyway the impulses only increment in factors of 4 (4, 8, 12, 16, etc) rather than 1.  I thought, based on the documentation, that impulses was intended to report the accumulated wheel revolutions.  The conversion from impulses to kWh doesn't include any factor to take this into account, so I'm wondering if others are seeing the same thing.  Perhaps Blueline found a cost savings by only reporting every 4th rotation?  I'm just curious if others see the same trend.

```
Time	impulses	impulses increment
3/7/2022 10:50		
3/7/2022 10:55	58571	
3/7/2022 11:00	58571	0
3/7/2022 11:05	58571	0
3/7/2022 11:10	58579	8
3/7/2022 11:15	58591	12
3/7/2022 11:20	58595	4
3/7/2022 11:25	58595	0
3/7/2022 11:30	58595	0
3/7/2022 11:35	58595	0
3/7/2022 11:40	58603	8
3/7/2022 11:45	58611	8
3/7/2022 11:50	58611	0
3/7/2022 11:55	58611	0
3/7/2022 12:00	58611	0
3/7/2022 12:05	58623	12
3/7/2022 12:10	58631	8
3/7/2022 12:15	58631	0
3/7/2022 12:20	58631	0
3/7/2022 12:25	58635	4
3/7/2022 12:30	58643	8
```

gdt, 2023-09-30:
Apparently nobody noticed or commented.  Feel free to open a discussion; this doesn't really fit the issue system.

closed 2023-09-30 by gdt

jbrzozoski, 2023-10-24:
Yes, Blueline only reports every 4th revolution. They hijacked the 2 lowest bits of the counter for other purposes in the payload to make everything fit.