merbanan/rtl_433 #2010: -fvisibility=hidden issue building tests

ZeroChaos-, 2022-03-16:
I get this when I build by default, but if I remove -fvisibility=hidden from CMakeLists.txt it fixes it.
```

[32m * [39;49;00mPackage:    net-wireless/rtl_433-9999
[32m * [39;49;00mRepository: maintainer-zero
[32m * [39;49;00mMaintainer: radio@gentoo.org
[32m * [39;49;00mUSE:        abi_x86_64 amd64 elibc_glibc kernel_linux rtlsdr soapysdr test userland_GNU
[32m * [39;49;00mFEATURES:   network-sandbox preserve-libs sandbox test userpriv usersandbox
&gt;&gt;&gt; Unpacking source...
 [32m*[0m Repository id: merbanan_rtl_433.git
 [32m*[0m To override fetched repository properties, use:
 [32m*[0m   EGIT_OVERRIDE_REPO_MERBANAN_RTL_433
 [32m*[0m   EGIT_OVERRIDE_BRANCH_MERBANAN_RTL_433
 [32m*[0m   EGIT_OVERRIDE_COMMIT_MERBANAN_RTL_433
 [32m*[0m   EGIT_OVERRIDE_COMMIT_DATE_MERBANAN_RTL_433
 [32m*[0m 
 [32m*[0m Fetching https://github.com/merbanan/rtl_433 ...
git fetch https://github.com/merbanan/rtl_433 +HEAD:refs/git-r3/HEAD
git symbolic-ref refs/git-r3/net-wireless/rtl_433/0/__main__ refs/git-r3/HEAD
 [32m*[0m Checking out https://github.com/merbanan/rtl_433 to /var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999 ...
git checkout --quiet refs/git-r3/HEAD
GIT update --&gt;
   repository:               https://github.com/merbanan/rtl_433
   at the commit:            45e8fc54b32ce9daf70e967b4fb396e86f9fde24
&gt;&gt;&gt; Source unpacked in /var/tmp/portage/net-wireless/rtl_433-9999/work
&gt;&gt;&gt; Preparing source in /var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999 ...
 [32m*[0m Source directory (CMAKE_USE_DIR): "/var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999"
 [32m*[0m Build directory  (BUILD_DIR):     "/var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999_build"
 [32m*[0m Hardcoded definition(s) removed in CMakeLists.txt:
 [32m*[0m     set(CMAKE_BUILD_TYPE "Release")
 [32m*[0m  set(CMAKE_BUILD_TYPE ${CMAKE_BUILD_TYPE} CACHE STRING "")
&gt;&gt;&gt; Source prepared.
&gt;&gt;&gt; Configuring source in /var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999 ...
 [32m*[0m Source directory (CMAKE_USE_DIR): "/var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999"
 [32m*[0m Build directory  (BUILD_DIR):     "/var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999_build"
cmake -C /var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DENABLE_RTLSDR=yes -DENABLE_SOAPYSDR=yes -DBUILD_TESTING=yes -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999_build/gentoo_toolchain.cmake /var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999
loading initial cache file /var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999_build/gentoo_common_config.cmake
[0mCMake Deprecation Warning at CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake &lt; 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument &lt;min&gt; value or use a ...&lt;max&gt; suffix to tell
  CMake that the project does not need compatibility with older versions.

[0m
-- The C compiler identification is GNU 11.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git version:  commit 45e8fc54b32ce9daf70e967b4fb396e86f9fde24 from 2022-03-09T13:46:57+0100
-- Using Git version tag: 21.12-75-g45e8fc54 on  at 202203091346
-- IPv6 support enabled.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Threads support will be compiled.
-- Found OpenSSL: /usr/lib64/libcrypto.so (found version "1.1.1n")  
-- OpenSSL TLS support will be compiled. Found version 1.1.1n
-- Found PkgConfig: /usr/bin/x86_64-pc-linux-gnu-pkg-config (found version "1.8.0") 
-- Found LibRTLSDR: /usr/lib64/librtlsdr.so (found version "0.6.0_p20200802") 
-- Found LibUSB: /usr/lib64/libusb-1.0.so (found version "1.0.24") 
-- RTL-SDR device input will be compiled. Found version 0.6.0_p20200802
[... 72 lines cut ...]
[242/247] : &amp;&amp; /usr/bin/x86_64-pc-linux-gnu-gcc -Os -march=native -mtune=native -pipe -frecord-gcc-switches -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic tests/CMakeFiles/test_bitbuffer.dir/__/src/bitbuffer.c.o -o tests/test_bitbuffer   &amp;&amp; :
[243/247] : &amp;&amp; /usr/bin/x86_64-pc-linux-gnu-gcc -Os -march=native -mtune=native -pipe -frecord-gcc-switches -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic tests/CMakeFiles/test_optparse.dir/__/src/optparse.c.o -o tests/test_optparse   &amp;&amp; :
[244/247] /usr/bin/x86_64-pc-linux-gnu-gcc -DGIT_BRANCH="" -DGIT_TIMESTAMP=202203091346 -DGIT_VERSION=21.12-75-g45e8fc54 -DLIBUSB1 -DMG_ENABLE_IPV6=1 -DMG_ENABLE_SSL -DOPENSSL -DRTLSDR -DSOAPYSDR -DTHREADS -D_GNU_SOURCE -I/var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999/include -I/usr/include/libusb-1.0  -Os -march=native -mtune=native -pipe -frecord-gcc-switches   -Wall -Wextra -Wvla -Wsign-compare -std=c99 -pedantic -Wshadow -Wmissing-prototypes -Wimplicit-fallthrough -fvisibility=hidden -std=c99 -w -MD -MT src/CMakeFiles/r_433.dir/mongoose.c.o -MF src/CMakeFiles/r_433.dir/mongoose.c.o.d -o src/CMakeFiles/r_433.dir/mongoose.c.o -c /var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999/src/mongoose.c
[245/247] /usr/bin/x86_64-pc-linux-gnu-gcc -DGIT_BRANCH="" -DGIT_TIMESTAMP=202203091346 -DGIT_VERSION=21.12-75-g45e8fc54 -DLIBUSB1 -DMG_ENABLE_IPV6=1 -DMG_ENABLE_SSL -DOPENSSL -DRTLSDR -DSOAPYSDR -DTHREADS -D_GNU_SOURCE -I/var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999/include -I/usr/include/libusb-1.0  -Os -march=native -mtune=native -pipe -frecord-gcc-switches   -Wall -Wextra -Wvla -Wsign-compare -std=c99 -pedantic -Wshadow -Wmissing-prototypes -Wimplicit-fallthrough -fvisibility=hidden -std=c99 -MD -MT src/CMakeFiles/rtl_433.dir/rtl_433.c.o -MF src/CMakeFiles/rtl_433.dir/rtl_433.c.o.d -o src/CMakeFiles/rtl_433.dir/rtl_433.c.o -c /var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999/src/rtl_433.c
ninja: build stopped: subcommand failed.
 [31;01m*[0m ERROR: net-wireless/rtl_433-9999::maintainer-zero failed (compile phase):
 [31;01m*[0m   ninja -v -j12 -l12 failed
 [31;01m*[0m 
 [31;01m*[0m Call stack:
 [31;01m*[0m     ebuild.sh, line  127:  Called src_compile
 [31;01m*[0m   environment, line 2584:  Called cmake_src_compile
 [31;01m*[0m   environment, line  976:  Called cmake_build
 [31;01m*[0m   environment, line  945:  Called eninja
 [31;01m*[0m   environment, line 1402:  Called die
 [31;01m*[0m The specific snippet of code:
 [31;01m*[0m       "$@" || die "${nonfatal_args[@]}" "${*} failed"
 [31;01m*[0m 
 [31;01m*[0m If you need support, post the output of `emerge --info '=net-wireless/rtl_433-9999::maintainer-zero'`,
 [31;01m*[0m the complete build log and the output of `emerge -pqv '=net-wireless/rtl_433-9999::maintainer-zero'`.
 [31;01m*[0m The complete build log is located at '/var/log/portage/net-wireless:rtl_433-9999:20220316-202111.log'.
 [31;01m*[0m For convenience, a symlink to the build log is located at '/var/tmp/portage/net-wireless/rtl_433-9999/temp/build.log'.
 [31;01m*[0m The ebuild environment file is located at '/var/tmp/portage/net-wireless/rtl_433-9999/temp/environment'.
 [31;01m*[0m Working directory: '/var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999_build'
 [31;01m*[0m S: '/var/tmp/portage/net-wireless/rtl_433-9999/work/rtl_433-9999'
```

zuckschwerdt, 2022-03-16:
Thanks! It seems like lib data linking is missing from the linker args (the first errored line). Strange, it's listed for linking here https://github.com/merbanan/rtl_433/blob/master/tests/CMakeLists.txt#L6

But visibility is a clue here, maybe the lib is empty. We should either remove the fvisibility=hidden or add proper visibility attributes.

ZeroChaos-, 2022-03-17:
I lack the ability to tell you what or how to do it, but I'm setup to easily test anything you put in (any branch of) git.  Standing by :-)

zuckschwerdt, 2022-03-17:
Thanks, I can't reproduce, must be some combination of compile/link flags.
Please test this branch https://github.com/merbanan/rtl_433/tree/fix-exports -- work in progress though.

zuckschwerdt, 2022-03-17:
Ah, wait. That still doesn't export the other symbols from abuf and term_ctl. I guess we should drop the `fvisibility=hidden`. I need to check if that flag does anything sensible for C libs anyway.

zuckschwerdt, 2022-03-17:
Ok, to reproduce the critical piece is `-DBUILD_SHARED_LIBS:BOOL=ON` which is mostly broken.
We are working towards an API but there are still many dependecies between all parts, we shouldn't export abuf and term_ctl symbols, but the data outputs will need those currently.

ZeroChaos-, 2022-03-17:
Thanks for the quick turn around.  I will not be at my desk for another two hours but I will have this tested asap today.

zuckschwerdt, 2022-03-17:
Thanks. No hurry, I don't think anyone else tried to build shared libs -- as it's obviously broken ;)
Interestingly I don't see where `BUILD_SHARED_LIBS` is set in your build?

ZeroChaos-, 2022-03-17:
> Thanks. No hurry, I don't think anyone else tried to build shared libs -- as it's obviously broken ;) Interestingly I don't see where `BUILD_SHARED_LIBS` is set in your build?

Yeah, it kinda gets hidden in a distro specific config file called gentoo_common_config.cmake.  I totally forgot to mention that, sorry about that.

ZeroChaos-, 2022-03-17:
> The branch should work now, can you check before I release that?

Confirmed, this branch builds.  Thanks!

zuckschwerdt, 2022-03-18:
Fixed now. Though you likely want an explicit `-DBUILD_SHARED_LIBS:BOOL=OFF` as the lib built is awkwardly named just `data` -- which surely isn't great to install globally ;)

commit eda74c17 (2022-03-18): minor: Add visibility to dynamic lib exports (closes #2010)
closed 2022-03-18 by zuckschwerdt