merbanan/rtl_433 #2015: Honeywell 5816 security contact decoding issues

rct, 2022-03-19:
I have a bunch of Honeywell 5800 series security contacts that decode reliably, but I have two 5816 door sensors that don't decode reliably.  If the sensor within 15 ft of the rtl-sdr it doesn't decode at all. If the distance is larger than that I'll get some decodes, but it's not as reliable as the others, only some of the repeats will decode. 

The analyzer (-A) isn't able to detect the pulses. Maybe it's saturating the receiver. Using `-Y autolevel` doesn't seem to help.

Frequency of both of the 5816 seems to be around 345.06 Mhz.  I've got rtl_433 set to 345.0M. So they seem to be within range and not falling on the DC spike.

I'll attach samples. Thanks for taking a look.

rct, 2022-03-19:
[honeywell-5816-1.zip](https://github.com/merbanan/rtl_433/files/8310340/honeywell-5816-1.zip)

```
rtl_433 version 21.05-111-gf30ee220 branch master at 202110220900 inputs file rtl_tcp RTL-SDR with
 TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/rct/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 168 out of 198 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63
 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 ]
Test mode active. Reading samples from file: g070_345M_250k.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Detected OOK package    @0.076016s
Analyzing pulses...
Total count:    4,  width: 368.91 ms            (92227 S)
Pulse width distribution:
 [ 0] count:    4,  width: 21732 us [21728;21736]       (5433 S)
Gap width distribution:
 [ 0] count:    3,  width: 93992 us [93992;93996]       (23498 S)
Pulse period distribution:
 [ 0] count:    3,  width: 115724 us [115724;115728]    (28931 S)
Pulse timing distribution:
 [ 0] count:    4,  width: 21732 us [21728;21736]       (5433 S)
 [ 1] count:    4,  width: 95496 us [93992;100004]      (23874 S)
Level estimates [high, low]:  16049,   3704
RSSI: -0.1 dB SNR: 6.4 dB Noise: -6.5 dB
Frequency offsets [F1, F2]:   16340,      0     (+62.3 kHz, +0.0 kHz)
Guessing modulation: Un-modulated signal. Maybe a preamble...
view at https://triq.org/pdv/#AAB10254E475088181818155

Detected OOK package    @1.244376s
Analyzing pulses...
Total count:    1,  width: 21.73 ms             ( 5433 S)
Pulse width distribution:
 [ 0] count:    1,  width: 21732 us [21732;21732]       (5433 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width: 21732 us [21732;21732]       (5433 S)
 [ 1] count:    1,  width: 100004 us [100004;100004]    (25001 S)
Level estimates [high, low]:  15984,   3384
RSSI: -0.1 dB SNR: 6.7 dB Noise: -6.8 dB
Frequency offsets [F1, F2]:   17247,      0     (+65.8 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
view at https://triq.org/pdv/#AAB10254E486A48155

Detected OOK package    @1.377480s
Analyzing pulses...
Total count:    1,  width: 21.73 ms             ( 5433 S)
Pulse width distribution:
 [ 0] count:    1,  width: 21732 us [21732;21732]       (5433 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width: 21732 us [21732;21732]       (5433 S)
 [ 1] count:    1,  width: 100004 us [100004;100004]    (25001 S)
Level estimates [high, low]:  16062,   4154
RSSI: -0.1 dB SNR: 5.9 dB Noise: -6.0 dB
Frequency offsets [F1, F2]:   17937,      0     (+68.4 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
view at https://triq.org/pdv/#AAB10254E486A48155

Detected OOK package    @1.510596s
Analyzing pulses...
Total count:    1,  width: 21.74 ms             ( 5434 S)
Pulse width distribution:
 [ 0] count:    1,  width: 21736 us [21736;21736]       (5434 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width: 21736 us [21736;21736]       (5434 S)
 [ 1] count:    1,  width: 100004 us [100004;100004]    (25001 S)
Level estimates [high, low]:  15888,   2880
RSSI: -0.1 dB SNR: 7.4 dB Noise: -7.6 dB
Frequency offsets [F1, F2]:   17058,      0     (+65.1 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
view at https://triq.org/pdv/#AAB10254E886A48155

Detected OOK package    @1.643704s
Analyzing pulses...
Total count:    1,  width: 21.75 ms             ( 5438 S)
Pulse width distribution:
 [ 0] count:    1,  width: 21752 us [21752;21752]       (5438 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width: 21752 us [21752;21752]       (5438 S)
 [ 1] count:    1,  width: 100004 us [100004;100004]    (25001 S)
Level estimates [high, low]:  15449,   5065
RSSI: -0.3 dB SNR: 4.8 dB Noise: -5.1 dB
Frequency offsets [F1, F2]:   14438,      0     (+55.1 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
view at https://triq.org/pdv/#AAB10254F886A48155
```

zuckschwerdt, 2022-03-20:
This looks like PSK, not FSK. We would need samples at a higher sample rate (`-s 1024k`) to verify this.

rct, 2022-03-20:
> This looks like PSK, not FSK. We would need samples at a higher sample rate (`-s 1024k`) to verify this.

Ok I will capture and attach later today. However,   I think FSK would be pretty surprising:

- the 5816 is an older member of the Honeywell 5800 series of contacts and can be used with the same alarm system receivers.
- This contact can be decoded if I move it far enough away from the rtlsdr.

zuckschwerdt, 2022-03-20:
Could be due to the low samplerate, but it looks like a solid carrier. And there are two side frequencies which don't look like aliasing, the changes switch the phase by 180 deg, best I could see.
[image]

rct, 2022-03-20:
(Side note: When I first tried 1024k sample rate, a sensor that is reliably decoded at the default sample rate no longer decoded. reliably. While scratching my head, I noticed the noise level was now -3 dB...   So there was some noise I didn't pick up at 250K but did pick up with the wider bandwidth.)

Here's 4 recordings at 1024K.
2 for a 5800mini that decode just fine
2 for a 5816 that doesn't decode

```
rtl_433 version 21.12-78-gf380976d branch master at 202203180841 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/rct/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 184 out of 215 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 ]
Test mode active. Reading samples from file: g003_345M_1024k.cu8
baseband_demod_FM: low pass filter for 1024000 Hz at cutoff 102400 Hz, 9.8 us
Detected OOK package    @0.047855s
Analyzing pulses...
Total count:    6,  width: 604.14 ms            (618643 S)
Pulse width distribution:
 [ 0] count:    6,  width: 21857 us [21850;21874]       (22382 S)
Gap width distribution:
 [ 0] count:    5,  width: 94599 us [94593;94603]       (96869 S)
Pulse period distribution:
 [ 0] count:    5,  width: 116457 us [116450;116467]    (119252 S)
Pulse timing distribution:
 [ 0] count:    6,  width: 21857 us [21850;21874]       (22382 S)
 [ 1] count:    6,  width: 95499 us [94593;100001]      (97791 S)
Level estimates [high, low]:  15821,   1316
RSSI: -0.2 dB SNR: 10.8 dB Noise: -11.0 dB
Frequency offsets [F1, F2]:    4984,      0     (+77.9 kHz, +0.0 kHz)
Guessing modulation: Un-modulated signal. Maybe a preamble...
view at https://triq.org/pdv/#AAB1025561FFFF81818181818155
```

[honeywell-5816-1024k-1.zip](https://github.com/merbanan/rtl_433/files/8311672/honeywell-5816-1024k-1.zip)

rct, 2022-03-20:
It looks like the 1st, 3rd, and 4th 5816 samples in rtl_433_tests from 2019 show similar problems, but the 2nd sample (g002) does seem to get recognized as pulses and sliced up.

- https://github.com/merbanan/rtl_433_tests/tree/master/tests/honeywell_5816

```
Pulse timing distribution:
 [ 0] count:    1,  width: 21856 us [21856;21856]	(5464 S)
 [ 1] count:    1,  width: 100004 us [100004;100004]	(25001 S)
```

[image]

zuckschwerdt, 2022-03-20:
Something is very strange there. The 1024k signal is clearly good OOK and 135µs pulse/gap should always be okay at 250k.
But as you show in the screenshots, the "Good - 5800mini" looks proper OOK, and the "5816 failed to decode" and the one above does not.
Very much Iooks like FSK. Could that be something with the antenna? Maybe play with different center frequencies to inspect how that "harmonic" / "shadow" behaves?

rct, 2022-03-20:
@zuckschwerdt Thanks for looking into this.

> Very much Iooks like FSK. Could that be something with the antenna? Maybe play with different center frequencies to inspect how that "harmonic" / "shadow" behaves?

I will try some more experiments and I certainly won't rule out something particular to my set up, but it is worth keeping in mind that:

- 3 of the 4 samples in the rtl_433_tests repo for the 5816 don't seem to decode correctly using https://triq.org/explorer/. Those aren't my recordings they are from @adamweiss back in 2019
- [@rhinomods reported similar distance related problems for the 5816](https://github.com/merbanan/rtl_433/pull/1384#issuecomment-670278842) back in some of the comments around PR #1384 

> Maybe play with different center frequencies to inspect how that "harmonic" / "shadow" behaves?

So at 250K that harmoninc the spectrogram shows at 344.786 Mhz shouldn't be in the passband. But I'll try different center frequencies. Is there any reason to keep using a sample rate of 1024k?  Ideally I'd like to keep the sample rate to 250k since it works just fine with the other devices and is much less of a load.

zuckschwerdt, 2022-03-20:
Sorry, I didn't mean that true harmonic. I meant that artifact the looks like FSK. When the decoding works the signal looks like plain OOK (something like morse code), but in the not-decoding samples it looks like FSK (a zigzag pattern, a carrier at shifting frequency).
That will be seen by rtl_433 also, so no chance to decode what already looks bad in the spectrogram.

If the sender really is pulsing OOK, then that carrier in the gaps must be coming from somewhere else. Maybe some resonance in the sender or receiver antenna?

Using 250k is fine. But the artifact will be less visible -- it just looked like a continuous carrier not that FSK zigzag

rct, 2022-03-20:
> If the sender really is pulsing OOK, then that carrier in the gaps must be coming from somewhere else. Maybe some resonance in the sender or receiver antenna?

No antenna on the RTLSDR seems to be a little hit and miss, some full decodes of most of the repeats and some complete misses.

The spectogram between good and bad looks somewhat similar (to my eyes) with regard to the artifacts I think you are describing. What appears different between the successful ones seems more like a question of gain?

### decode failed

[image]
223

### decode succeeded

[image]

[honeywell-5816-1024k-no-antenna.zip](https://github.com/merbanan/rtl_433/files/8312380/honeywell-5816-1024k-no-antenna.zip)

thinkjk, 2022-10-11:
@rct any update on this? I'm running into the exact same issue! I have a few sensors right near my antenna and it doesn't pick any of those up, but all of my father ones worked. 

Currently I'm running an RTL-SDR with just an antenna extension cable dangling from it but I'd rather find a proper antenna or figure out why this is happening.

I've a novice with rtl_433 and using it to pick up my Honeywell 5816 sensors for home assistant.

sandstormkeshav, 2022-10-20:
I am experiencing the same issues as well

thinkjk, 2022-11-26:
Final bump from me on this issue. Does anyone have a solution?

zuckschwerdt, 2022-11-27:
From what we have seen here the problem looks like antenna overloading and with that some harmonics. Not ideal but should be recoverable. The `minmax` demod is not good at handling this and seems wrongly to run that signal as FSK.
Workarounds would be: use `-Y classic` or disable FSK (don't select any FSK decoder) or someone needs to work on the minmax demod.

rct, 2022-11-27:
@zuckschwerdt - Thank you for these tips!  I'll experiment with just `-R 70` as well as `-Y classic`.   

How long ago (approx) did minmax become the default?   That might explain why I think I was getting more successful decodes when I first started experimenting with these sensors.

(@thinkjk - sorry for not responding to you back in October I missed the mention.  I had somewhat given up on this for a while and have mostly been using 5800mini / versa mini contacts which don't have this issue.)

Note: It's not just the 5816,  I have some 5820L (slim sensors in awning windows) that haven't been being received reliably.

zuckschwerdt, 2022-11-27:
> How long ago (approx) did minmax become the default?

Around end of 2019, then demod was in a664d0b.

rct, 2022-11-29:
I haven't had a chance to do anything but some very superficial analysis, the results so far:

`-R 70` by itself, without `-Y classic`, did not improve reception of the problematic sensors.  I _think_ I still saw indications of attempted FSK decoding when the problematic sensors were triggered even with only protocol 70 enabled .   (I had stats set for every 5 minutes which is too long to be sure.)

```
{"time" : "2022-11-27 10:50:37.029478", "enabled" : 1, "since" : "2022-11-27T10:45:37", "frames" : {"count" : 63, "fsk" : 7, "events" : 12}, "stats" : [{"device" : 70, "name" : "Honeywell Door/Window Sensor, 2Gig DW10/DW11, RE208 repeater", "events" : 68, "ok" : 12, "messages" : 12, "abort_length" : 56}]}
```

I _think_ `-Y classic` improved reception of some sensors, but might have negatively impacted one or two others.  It could also be environmental differences. I'll have to dig deeper, and probably reenable saving unknown decodes.  

@thinkjk, @sandstormkeshav - did you have a chance to try any of this yet?

thinkjk, 2022-12-02:
> 

@rct I've been using '-R 70' but I tried using '-Y classic' and I still have the exact same issue. All of the sensors that are close to my antenna don't get picked up, but anything slightly further away works perfectly. If I remove the antenna the closer ones get picked up perfectly the the ones further away don't.

sandstormkeshav, 2022-12-03:
@rct I concur with @thinkjk findings. I don't notice any appreciable increase in successful decodes when modifying the settings. I can only get reliable decodes when operating with no antenna and nearby the sensors.

sandstormkeshav, 2023-01-11:
Any more thoughts on this, is there any other settings I can try?

thinkjk, 2023-01-17:
> Any more thoughts on this, is there any other settings I can try?

I haven't been able to find a solution. I've resulted to using a USB extender to try to move the antenna away from some devices and then wrapping half of the antenna in aluminum foil :/

thinkjk, 2023-03-26:
Bump again, seeing if anyone has a solution to this issue.

klohner, 2023-03-30:
I took a look at all the provided samples here and I'd agree with @zuckschwerdt that the root issue is signal overload.  By the time the signal gets to rtl_433's decoders, the signal is so overblown that there's nothing useful for it decode.  I might suggest you play around with rtl_433's gain setting (-g) to see if you can attenuate the signal better instead of using aluminum foil around the antenna.

The signal is definitely ASK, not FSK (even though a blown-out signal might start to look like this).  If the signal is attenuated within reason, like a few of these samples are, rtl_433 can find the signal with a flex spec like this:

```
$ rtl_433 -R0 -r good-5800mini-g001_345M_1024k.cu8 -X "n=Honeywell,m=OOK_MC_ZEROBIT,s=136,l=272,r=600"` 
codes : {64}00017bf70c7f066c
```

And you can feed a code like that back into rtl_433 to have it tell you its meaning with this:

```
$ rtl_433 -y {64}00017bf70c7f066c
model     : Honeywell-Security                     id        : 408f3
channel   : 8            event     : 80            state     : open          contact_open: 1           reed_open : 0
alarm     : 0            tamper    : 0             Battery   : 1             heartbeat : 0             Integrity : CRC
```

Without the OOK_MC_ZEROBIT line code decoding, you can get the raw PCM code of these signals with this flex spec:

```
$ rtl_433 -R0 -r good-5800mini-g001_345M_1024k.cu8 -X "n=Honeywell,m=OOK_PCM,s=136,l=136,r=408"`
codes :  {130}aaaaaaacd53554d4ab4ad554aad2d34a0
```

And here's an ASK demodulated view of that signal:
[image]

With some band pass filtering, this looks much nicer:

[image]

Now let's compare this with a message in the bad-5816-g003_345M_1024k.cu8 signal file:

[image]

The entire message signal is completely blown out.   Even what should be gaps in the ASK signal are blown out.  If I do the same band-pass filtering to try to clean it up, I get this:

[image]

Again, both pulses and gaps are blown out.  Interestingly, before the message starts, there's about 4,451 μs of what should be "gap" silence that is so loud it's already blowing out the signal.  Maybe the sensor is sending what it thinks is a very quiet or negligible signal for the gap, but your receiver's gain is so high that even this is too loud.  Then the ASK pulses it sends that you want to receive can't be discerned above this floor.  The transitions between gaps and pulses just look like very short useless spikes of noise.

To try to fix this, maybe start with something like this flex spec instead of the built-in decoder and provide a specific gain level:

`$ rtl_433 -f 345M -s 1024k -R 0 -X "n=Honeywell,m=OOK_PCM,s=136,l=136,r=408" -S all -g 12.5`

and see how well it picks up the signal.  Try different gain levels other than 12.5 dB (I think the useful range is from 0.9 to 49.6) and see if you can find a setting to get a reliable and consistent signal from your various devices at their various distances.  Once you find a gain setting that works for you, use that option with your normal rtl_433 usage.

rct, 2023-03-30:
@klohner - Thank you for your detailed analysis. I need to re-read this and digest it a bit in the morning. 

> Maybe the sensor is sending what it thinks is a very quiet or negligible signal for the gap, but your receiver's gain is so high that even this is too loud.

FWIW, I'm not setting an explicit gain. 

`-Y autolevel -Y classic -R 70 -f 345.0M  -M level -M noise -M stats:1:300` 

I'm away from those sensors for a while, but when I'm back there I'll try some experimentation with fixed gains.

> Interestingly, before the message starts, there's about 4,451 μs of what should be "gap" silence that is so loud it's already blowing out the signal.

I think that ~4.5ms of "gap signal" is likely a good area to investigate. I don't think the other sensors that are received well do that.  It would be interesting to see how the presence of that long pulse (which if it was true OOK wouldn't exist) is impacting the various gain settings.

rct, 2023-03-30:
@thinkjk @sandstormkeshav have you experimented with fixed gain settings?  You might be in a better position to do this sooner then I will.

You might want to take a look at @hdtodd's rtl_433_stats (formerly rtl_snr) and rtl_watch when you do your experiments.  rtl_433's built in web interface should also be a help. 

Also @klohner:

> see if you can find a setting to get a reliable and consistent signal from your various devices at their various distances. Once you find a gain setting that works for you, use that option with your normal rtl_433 usage.

For what it is worth, I suspect a fixed gain isn't going to work to reliably receive a range of devices that are spread all over the house and deal with changes in background noise.  But I do agree that if we can figure out that to what degree the various gains are effecting things, we can develop insights into a better solution.

klohner, 2023-03-30:
Here's what that 4.5 ms of "gap signal" looks like in the "no-antenna-good-5816-g001_345M_1024k.cu8" signal:

[image]

It's rather faint, so I guess it doesn't interfere with the decoding.

However, compare that to "bad-5816-g003_345M_1024k.cu8":

[image]

You can't see the pulses through the noise of the gaps!

rct, 2023-03-30:
Yes, but how to figure out whether there is just too much power in the "gap" portion of the 5816's signal, or whether something has cause one (or more) of the gain functions to be turned up too high in software or hardware?

rct, 2023-03-30:
This is probably off topic, here's the rtl_433 web UI graph for my 345mhz rtl-sdr:

[image]

There is no one in that place right now, so the receives are infrequent and there isn't anything else on 345Mhz that I know of.  Looks like the web UI only changes the noise level when there are decodes. 

One minor curiosity to me: it looks like for some devices the noise (and presumably gain) change during the course of the 6 repeated messages that the device sends.  (see the green dots in a vertical line).    

```
{"freq" : 345.066, "rssi" : -0.135, "snr" : 15.410, "noise" : -15.545}
{"freq" : 345.066, "rssi" : -0.134, "snr" : 17.341, "noise" : -17.476}
{"freq" : 345.064, "rssi" : -0.111, "snr" : 18.689, "noise" : -18.800}
{"freq" : 345.066, "rssi" : -0.135, "snr" : 21.715, "noise" : -21.850}
{"freq" : 345.068, "rssi" : -0.108, "snr" : 26.015, "noise" : -26.124}
{"freq" : 345.065, "rssi" : -0.112, "snr" : 30.893, "noise" : -31.005}
```

zuckschwerdt, 2023-03-30:
Very interesting stats. Those receives indicate that the signal was at least at +15 dB, that's very much too loud.
(The noise dropping means the (auto)gain is reduced.)

rct, 2023-04-13:
@klohner - Thanks for your work on this. 

Per #2479 - I've added the flex decoder so I can start capturing some data comparing successful decodes between the builtin decoder and the flex spec.

`-X n=Honeywell-Flex,m=OOK_PCM,s=136,l=136,r=408,preamble={24}555556,symbol_zero={2}8,symbol_one={2}4,get=@0:{4}:channel,get=@4:{20}:id,get=@24:{8}:event,get=@32:{16}:crc` 

At first glance, just counting output lines, it seems like there are a few more Flex decodes that the builtin decoder didn't capture.  This appears to be the case for both 5816 and 5800minis, but there is a much bigger difference for the minis. 

However this could be at least partially due to CRC checking.  The flex decoder isn't checking the CRC.

klohner, 2023-04-14:
@rcl Thanks. I'd love to look over .cu8 sample captures of all the devices you have. I only have one 5817 that I can experiment on and see if I can reverse what's going on with the CRC.  

I'm not sure how correct this is, but this is my latest draft at a flex decoder .conf file for these devices.

```
decoder {
    name=345MHz_Security,
    modulation=OOK_PCM,
    short=136,
    long=136,
    reset=408,
    preamble={24}555556,
    symbol_zero={2}8,
    symbol_one={2}4,
    get=Mfg:@0:{4}:[0:0 1:1 2:2 3:2GIG 4:4 5:5 6:6 7:7 8:8-Ademco/Honeywell/Resideo 9:9 10:2GIG 11:b 12:c 13:d 14:e 15:f],
    get=ID:@4:{20},
    get=Alerts:@24:{4}:[ 0:All_OK            1:Loop-3                    2:Loop-2                     3:Loop-2_&_Loop-3           
                         4:Tamper            5:Loop-3_&_Tamper           6:Loop-2_&_Tamper            7:Loop-2_&_Loop-3_&_Tamper 
                         8:Loop-1            9:Loop-1_&_Loop-3          10:Loop-1_&_Loop-2           11:Loop-1_&_Loop-2_&_Loop-3 
                        12:Loop-1_&_Tamper  13:Loop-1_&Loop-3_&_Tamper  14:Loop-1_&_Loop-2_&_Tamper  15:Loop-1_&_Loop-2_&_Loop-3_&_Tamper],
    get=Battery:@28:{1}:[0:ok 1:LOW],
    get=Trigger:@29:{1}:[0:ALERT 1:supervision],
    get=Power:@30:{1}:[0:ok 1:INITIALIZED],
    get=Source:@31:{1}:[0:origin 1:REPEATER],
    get=CRC:@32:{16}:
}

```

rct, 2023-04-14:
> see if I can reverse what's going on with the CRC.

@klohner - Sorry for any confusion. I believe the CRC is correct with the builtin decoder. I just haven't learned enough about flex decoders yet to figure out how to verify it, especially given that the Honeywell vs 2Gig/Versa devices have slightly different CRCs.   (I also have a bunch of the 2Gig/Versa version of the 5800mini's)

https://github.com/merbanan/rtl_433/blob/f8489e9ba611aaa3ed539e75fa4e21ed4d55acdc/src/devices/honeywell.c#L86-L91

I haven't had a chance to do any experimentation with gain settings.  But I figured I could quickly add your flex decoder to my existing set up to start capturing data for comparison.

thinkjk, 2023-04-15:
I haven't been able to test but I'm happy to donate to you guys working on fixing this (or at least trying to figure it out)

rct, 2023-04-15:
@klohner - Here's some cu8s from a nearby 5820L (slim sensor).   This sensor is the opposite of the 5816 in that reliability drops off quickly once the sensor get 10-20ft from the receiver.

The flex decoder seems to do a better job with this than the builtin one.

[Honeywell-5820L-near-cu8.zip](https://github.com/merbanan/rtl_433/files/11240660/Honeywell-5820L-near-cu8.zip)

rct, 2023-04-15:
@klohner - Here are cu8's from a Versa-2gig. This is basically the same hardware as the Noneywell 5800mini. These are currently the ones that are most reliably received and decoded with the builtin decoder at both close range and distances.  Though there is likely some room for improvement.

[Versa-2gig-near-cu8.zip](https://github.com/merbanan/rtl_433/files/11240690/Versa-2gig-near-cu8.zip)

Also, I used the google form from your GitHub profile to reach out to you directly.

thinkjk, 2023-04-21:
I've tried changing the only the gain. I tried 12.5, 40.x, 20.x, 32.8, 15.7, and none of them changed my results. I have much better results when I keep the gain at auto. @rct @klohner

thinkjk, 2023-04-22:
I'm currently running rtl_433 in home assistant, I'll run it on my desktop and provide a cu8 file this weekend.

thinkjk, 2023-04-24:
@klohner  

Sorry for the triple post. Attached is 2 zip files with cu8 files in them.

small_ant was run with a small antenna attached
large_ant was with a large antenna attached

In both tests I opened and closed the same 5 doors/windows.

This is the command I was running
```
rtl_433 -f 345M -R 70 -S all -T 120
```

Hope this helps!
[hw_large_ant.zip](https://github.com/merbanan/rtl_433/files/11306022/hw_large_ant.zip)
[hw_small_ant.zip](https://github.com/merbanan/rtl_433/files/11306023/hw_small_ant.zip)

scottames, 2023-05-07:
I am experiencing the same

gdt, 2023-09-29:
What's the status and path forward?   It looks like a fair bit has been understood but it's not easy to follow.

rct, 2023-09-30:
This is still unresolved. 

- @klohner discovered the demod used by the Honeywell decoder misses some, I think initial, bits which causes the signal not to be decoded. He posted a flex decoder which is able to capture more messages than the built in decoder.
- Some of the honeywell devices seem to overload the receiver, but it also seems like something, possibly the long initial "gap" signal may be causing one of the gain controls to overshoot.
- I think we also saw that since messages are sent infrequently, the gain controls/detection levels trend towards trying to capture very faint signals during the quiet periods.

gdt, 2023-10-15:
Thanks.   So it seems like the way forward is:
  - someone prepares a PR for the honeywell decoder that is more forgiving about initial sync.
  - we create a new issue about the general problem of AGC and overload, where we discuss what if anything rtl_433 can or should do about  it, or decide it's intractable.   Perhaps this involves scripts to listen at various gains and scoreboard results and help people decide how to operate.

jsnyder, 2024-03-26:
Thanks for this topic and the work everyone has done on this thread.  Using @klohner's flex decoder I've gotten better performance catching events with mine though some of them are still a bit flaky even though I've tried some of the different configurations here.  In general I've found for some 5816's it's more likely to catch them opening vs closing the reed switch and changing around some of the parameters might flip which one it's better at detecting.  If anyone has been experimenting with solutions I'm happy to test.

klohner, 2024-03-26:
I'd guess the flex decoder may give more results since it's not checking the CRC.  I seem to recall that the CRC differs on some of the devices that use this general protocol.  If you can record a sample for one of these devices that the flex decoder doesn't decode, please post the .cu8 so we can analyzer further.

afallows, 2024-03-30:
Hi Guys, I have a similar issue with a 2gig garage door sensor, but the flex decoder @klohner has posted above is working awesome and detecting it consistently.

The issue I'm having is getting the flex decoder to publish correctly to my MQTT server the same way the sensors using Device type 70 do. Right now the flex decoder is publishing to the same MQTT topic for all sensors instead of splitting it up by "id" per my config file. 

The ID and data are showing up under 
security/rows/0/id
security/rows/0/data
ect

Anyway to modify the flex decoder to map this data the same at device type 70 for honeywell does?

My MQTT setup
output kv
output mqtt://192.168.1.199,retain=1,devices=sensors/rtl_433/security[/id]

KV shown on the flex type sensor.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-03-30 07:15:37.375180
model     : 345MHz_Security                        count     : 1             num_rows  : 1             rows      :
len       : 48           data      : cd776200efb0  Mfg       : c             id        : 882530
Alerts    : All_OK       Battery   : ok            Trigger   : ALERT         Power     : ok            Source    : origin        CRC       : 61360
codes     : {48}cd776200efb0
Modulation: ASK          Freq      : 345.5 MHz
RSSI      : -0.2 dB      SNR       : 14.9 dB       Noise     : -15.1 dB

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-03-30 07:15:30.034598
model     : 345MHz_Security                        count     : 1             num_rows  : 1             rows      :
len       : 48           data      : cd7762805f80  Mfg       : c             id        : 882530
Alerts    : Loop-1       Battery   : ok            Trigger   : ALERT         Power     : ok            Source    : origin        CRC       : 24448
codes     : {48}cd7762805f80
Modulation: ASK          Freq      : 345.5 MHz
RSSI      : -0.2 dB      SNR       : 15.4 dB       Noise     : -15.6 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

jsnyder, 2024-03-30:
@afallows I'm away from my setup right now so I can't look directly but what I think I did was change Id in the flex decoder to be lower case instead of capital I (the MQTT output I think expects "id" rather than "Id") and I think I added the "unique" option which you can see referenced in the flex decoder docs. I can't remember if I had to change the rows or repeat options or anything else. I can check later when I have access to my machine again if that doesn't help.

@klohner I'll post up some captures when I'm back as well. I have a few different conditions I can share things under (where some work or not depending on state). I do also get some ids that are incorrect in decodes so the CRC being off allowing more in makes sense. I should probably at least check that the CRC matches a good value but I believe for the most part when the id is correct I've gotten a limited number of CRCs corresponding to number of states the sensor might transmit under.

jsnyder, 2024-03-31:
Here's what I'm currently using with HA that lets me split things up by id. My only changes were adding the unique option and changing Id to id:

```
decoder {
    name=345MHz_Security,
    modulation=OOK_PCM,
    unique,
    short=136,
    long=136,
    reset=408,
    preamble={24}555556,
    symbol_zero={2}8,
    symbol_one={2}4,
    get=Mfg:@0:{4}:[0:0 1:1 2:2 3:2GIG 4:4 5:5 6:6 7:7 8:8-Ademco/Honeywell/Resideo 9:9 10:2GIG 11:b 12:c 13:d 14:e 15:f],
    get=id:@4:{20},
    get=Alerts:@24:{4}:[ 0:All_OK            1:Loop-3                    2:Loop-2                     3:Loop-2_&_Loop-3           
                         4:Tamper            5:Loop-3_&_Tamper           6:Loop-2_&_Tamper            7:Loop-2_&_Loop-3_&_Tamper 
                         8:Loop-1            9:Loop-1_&_Loop-3          10:Loop-1_&_Loop-2           11:Loop-1_&_Loop-2_&_Loop-3 
                        12:Loop-1_&_Tamper  13:Loop-1_&Loop-3_&_Tamper  14:Loop-1_&_Loop-2_&_Tamper  15:Loop-1_&_Loop-2_&_Loop-3_&_Tamper],
    get=Battery:@28:{1}:[0:ok 1:LOW],
    get=Trigger:@29:{1}:[0:ALERT 1:supervision],
    get=Power:@30:{1}:[0:ok 1:INITIALIZED],
    get=Source:@31:{1}:[0:origin 1:REPEATER],
    get=CRC:@32:{16}:
}
```

afallows, 2024-03-31:
@jsnyder Thanks a lot for this. the *Unique* Tag and making sure the "id" was lower case worked for me.

Additionally, I modified as below so that it writes to similar points as the Honeywell protocol 70. I'm going to test for a couple days, but this is reading my garage door sensor and most of my other sensors! So thanks everyone on this thread.
`
#Generic Honeywell Flex decoder
decoder {
    name=345MHz_Security,
    modulation=OOK_PCM,
    unique,
    short=136,
    long=136,
    reset=408,
    preamble={24}555556,
    symbol_zero={2}8,
    symbol_one={2}4,
	get=reed_open2:@24:{4}:[ 0:0 1:0 2:0 3:0 4:0 5:0 6:0 7:0 8:1 9:1 10:1 11:1 12:1 13:1 14:1 15:1],
	get=contact_open2:@24:{4}:[ 0:0 1:0 2:1 3:1 4:0 5:0 6:1 7:1 8:0 9:0 10:1 11:1 12:0 13:0 14:1 15:1], 
	get=tamper2:@24:{4}:[ 0:0 1:0 2:0 3:0 4:1 5:1 6:1 7:1 8:0 9:0 10:0 11:0 12:1 13:1 14:1 15:1],
	get=event2:@24:{8},
    get=id:@4:{20},
    get=battery_ok2:@28:{1}:[0:1 1:0],
    get=CRC:@32:{16}:
}
`

zuckschwerdt, 2024-03-31:
If you just want to pick out a single bit you can shorten that e.g.
`get=reed_open2:@24:{4}:[ 0:0 1:0 2:0 3:0 4:0 5:0 6:0 7:0 8:1 9:1 10:1 11:1 12:1 13:1 14:1 15:1],`
is the same as
`get=reed_open2:@24:{1},`
i.e. just the top bit, at position 24. They other two would be the bit at @26, @25.

afallows, 2024-03-31:
Thanks a lot. That looks a lot cleaner, was so busy converting I didn't think outside of the box! haha.

It's interesting that the built-in decoder isn't capturing my sensor. I have a clean CRC being calculated with the same polynomial as the honeywell sensors in the Honeywell.c file. When I run the data manually through [http://www.sunshine2k.de/coding/javascript/crc/crc_js.html](url) and using polynomial 0x8050 and intit value 0 the CRCs are being calculated correctly, even for the 2gig tilt sensor that the built-in honeywell decoder isn't capturing.

rct, 2024-03-31:
@klohner wrote:

> I'd guess the flex decoder may give more results since it's not checking the CRC. I seem to recall that the CRC differs on some of the devices that use this general protocol. If you can record a [...]

I'm under the impression that the flex decoder is more successful because of the issue you discovered in #2479.  I think but could be wrong, that the built in decoder didn't get a fix for that issue.  https://github.com/merbanan/rtl_433/issues/2479#issuecomment-1504555749

zuckschwerdt, 2024-03-31:
To use the custom decoding like `symbol_zero`… a decoder needs to read raw PCM and then apply `extract_bits_symbols()`