merbanan/rtl_433 #2016: Feature request: Add "Z" at end when "-M time:iso:utc"

baxerus, 2022-03-20:
If the option `-M time` is used with `iso` **and** `utc` (so e.g. `-M time:iso:utc`) it would be superb to have a `Z` at the end of the date string, so it is parsed as UTC (e.g. by Javascript `new Date()` or similar).

#### Example:
The result for `-M time:iso:utc` should be `2022-03-20T19:41:55Z` instead of `2022-03-20T19:41:55`.

#### Additional information:
See [ISO 8601-1:2019(en) Date and time — Representations for information interchange — Part 1: Basic rules](https://www.iso.org/obp/ui/#iso:std:iso:8601:-1:ed-1:v1:en) in section 3.2.5 "Designator symbols".

zuckschwerdt, 2022-03-20:
Use "time:tz" to output time with timezone offset.
E.g. `-M time:iso:utc:tz` ;)

zuckschwerdt, 2022-03-20:
We really need a comprehensive page explaining all the options with examples. I think many users won't easily find fitting options or miss them altogether and then need  to build hacks.

closed 2022-03-20 by zuckschwerdt

baxerus, 2022-03-20:
I’m sorry for creating extra work. I partly looked at the help output, but also here https://triq.org/rtl_433/OPERATION.html#outputs (under Meta Information), simply because I needed the terminal to enter new commands 😉.  
On this page, what I thought is the next best thing to
> a comprehensive page explaining all the options with examples
especially the **tz** option is not explained😐.

Sorry again and a great project +1 (have to figure out, why Somfy messages are not detected, but that will be solvable)

zuckschwerdt, 2022-03-20:
No worries, good detailed request and an easy solution ;)

Thanks! Fixed in the docs now.

knarrff, 2022-11-02:
> We really need a comprehensive page explaining all the options with examples. I think many users won't easily find fitting options or miss them altogether and then need to build hacks.

We still do. I only found this closed report *after* I did something like this:
```
 ./bin/rtl_433 -F json | jq -c '.time |= (. +"Z" | gsub(" ";"T") | fromdateiso8601)'
```
when the following would have been a lot easier:
```
./bin/rtl_433 -F json -M time:unix
```
with two differences still though: the last is better because it actually does output UTC (takes the offset correctly into account), but as a downside, it does report the unix timestamp as string (but this has also already been reported in #1650).

So, as of now, for using the unix timestamp, one of your options is (keeping this here for reference):
```
./bin/rtl_433 -F json -M time:unix | jq -c '.time |= tonumber'
```