merbanan/rtl_433 #2018: Acurite 609TX differenciation

bkenobi, 2022-03-21:
I just picked up an Acurite 609TX to expand my temperature/humidity measurement capabilities.  The other devices I use are Acurite 592TXR.  The 609 seems to be part of the unit 11 code and the 592 are decoded by 40.  The 592 have a channel switch inside the battery bay to set either A, B, or C.  The 609 does not have a switch that I have seen and from testing seems to pick a new ID each time the batteries are replaced.  My intent was to source several more temperature/humidity sensors to place in various locations.  The problem is, it doesn't appear that the ID can be tracked as is.  Is there anything in rtl_433 that would let me identify individual 609 sensors so I could use multiple without having to figure out the ID each time batteries are replaced?

zuckschwerdt, 2022-03-21:
Yes, the 609 has only ID, temp, hum: https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L165-L177
The ID though is supported to identify devices. What output are you using?

rct, 2022-03-21:
The 609 and other devices like it that have volatile IDs (determined at power up) aren't really well suited to long term data tracking.  If the battery voltage briefly drops low enough the unit may power cycle and change IDs without you knowing it.  This happens on my old Lacrosse TX7 sensors, especially if they are in lower temperatures.

While the Acurite tower family of sensors (592TXR and related indoor/outdoor units) only have 3 choices of "channel" (sic), they do have **a fixed ID that will never change**.  So you can use more than 3 592TXR tower sensors.   You only need to use the the model + ID to uniquely identify the unit.  

The thing I don't like about the Acurite devices is they transmit way more often than is needed for temperature/humidity monitoring, so the more of those you have the more likely you'll have 'collisions' where devices "talk over" each other.

My current preferred temperature/humidity sensor is the Ambient Weather FT012TH and F007TH. (These may be sold under other names in other countries)  They have a DIP switch to let you set 8 channel IDs.  They also have a volatile ID that changes at each power up. For a unique name use model + channel.  They transmit once a minute, which is good if you have a bunch of them.  Currently the FT012TH is only $9 USD on Amazon.

The one problem I have with the Ambient units, is occasional bad data getting through the message integrity checks. Adding a reasonableness check to the code like was added to the Acurite devices a month or two ago should eliminate many of those. 

Hope this helps

bkenobi, 2022-03-21:
I'll look at those Ambient sensors.  I like the 592 because they are fairly accurate, transmit probably too often but can track quickly changing things.  I used the 3 to test my furnace by placing one at the thermostat, one at the outlet register, and one at the inlet register.  Things were cycling every 10 minutes or so and I could watch at 30s intervals as the register temperatures changed.  For things like outside temps, they report too frequently though.

The down side to the 592 is that there are only 3 channels and so adding more would mean I'd have no way to know which was reporting other than somehow tracking where it was and hoping they never cross paths.  Better to just get something like the Ambient that has 8 channels to start with.

rct, 2022-03-21:
> The down side to the 592 is that there are only 3 channels and so adding more would mean I'd have no way to know which was reporting 

If you are using rtl_433, you have access to the ID field. **On the 592 the ID field never changes** so you only need that (and model) to identify the unit.  Note that the Acurite Access / and "My Acurite" only use the ID and do not pay attention to channel at all. The access supposrts up to 7 devices

So yes you can have more than 3.  I'm currently using 4 of the Acurite "tower" sensors, the 3 original 592TXRs and a 6044m which is the indoor/display version of the 592TXR. 

If you are looking for outdoor units, two drawbacks of the F007TH I should mention:  1) the mounting hole is really small, certainly not as convenient for mounting on a hook as the 592TXR  2) Acurite's "pagoda" for the 592TXR is much more reasonably priced than the Ambient one.

bkenobi, 2022-03-22:
Oh interesting. I didn't see that before.  That may be the way to go then since I don't really need an additional 7-8, just a total of 7-8.  Very useful info!

bkenobi, 2022-03-22:
There are newer models than the 592 that I've seen in some of the stores.  Have these been tested to see if they work and, if so, if they hold their ID?  The 592 also has a non-humidity version but this one is listed as 592TXR / 06002 usually.  Acurite lists 3 options for replacement sensors: 592, 609, and 606.  

https://www.acurite.com/shop-all/weather-instruments/weather-sensors-and-parts/replacement-parts/indoor-outdoor-temperature-humidity-sensor.html?ref=name

But it looks like the 606 is just temperature and the 609 is the one in this issue that varies the ID by reset.  So I guess that leaves the 592 as the only option.

rct, 2022-03-22:
I think the 606 uses a different protocol, and has a decoder in rtl_433.   The 275 series has also an rtl_433 decoder.  You might want to skim some of the comments in `src/devices/acurite.c`

TIP: If you look for sensors that are compatible with "My Acurite" you find devices that have fixed IDs"
- https://www.acurite.com/media/documents/MyAcuRite-Compatible-Sensors.pdf
-

bkenobi, 2022-03-28:
I picked up a few more 592TXR sensors and they do appear to work together without issue.  I don't know what the upper limit of these would be, but I imagine it would come down to flooding the 433MHz bandwidth with sensors that transmit at the same period.  At some point, the odds that more than one will be reset at the same time become high enough that there are regular collisions.  I have 7x 592 and one 609 as well as a 5in1 weather station and a pool temperature sensor.

I'm currently planning on using 3 for my attic fan controller so the other 4x 592 and 1x 609 for transient things like checking the HVAC or confirming the camper isn't too cold/humid.

gdt, 2022-07-05:
This issue is really a discussion not a bug report and I think it has run its course.  @bkenobi  are you ok with closing?

bkenobi, 2022-07-05:
I have a working solution so I will close the issue.

closed 2022-07-05 by bkenobi

bkenobi, 2025-01-06:
I have a new finding with these sensors but it is not an issue with rtl-433.  I am posting here for others to see as a form of documentation in case it helps someone down the road.

I have a number of Acurite 592TXR thermohygrometers in use that have been working flawlessly for a few years.  If you use 3 or less of a specific model of sensor, you can use the channel to differentiate between the different units perfectly.  If you have more than 3, then the best option is to use the sensor's unique ID which is transmitted with every packet.  I do not know how this ID is specified but to date, I have found that they never change...that is until recently.  I have one sensor that disapeared and has not come back so I assumed it was dead.  After some debugging I found that it changed it's ID to something new.  I can simply change my setup to the new number but if others do the same again, I will have to recommend NOT using these sensors in cases with more than 3 deployed at the same site.