merbanan/rtl_433 #2019: old  tiny AVIDSEN Weather station

jcmichot, 2022-03-21:
I have a very old AVIDSEN Weather station, the temperature and battery decoded by generic_temperature_sensor.c
is fully false.

If that may be usefull to someone, after few log analyze, i'm able to decode the right values with:

    temp_raw = (int16_t)(((b[1] & 0x3f) * 10) | (b[2] & 0xf));
    battery = (b[2] & 0x80) ? 1:0;
    if ( b[1] & 0x80 ) // temperature negative
        temp_f = (float) temp_raw*-0.1f;
    else
        temp_f = (float) temp_raw*0.1f;

Regards,
JC

zuckschwerdt, 2022-03-23:
The `* 10` seems to indicate that it's BCD? Then you should use `+`. But it's only BCD on the decimal, right? With 0x3f on the integer part that allows up to +/-63 C, which seems reasonable. But is it really not 0x7f with +/- 127 C?

If you could provide more details on what model that is and a sample .cu8 we can add that decoder.

gdt, 2023-09-30:
@jcmichot Are you up for figuring this out more fully and submitting a PR?

jcmichot, 2023-10-01:
I have lost this old AVIDSEN Weather stations during a move, so I can no longer provide any information. Maybe it's at the bottom of a box that I'll find next time I move ;-)

gdt, 2024-06-05:
feel free to ask for reopening if you pick this up again

closed 2024-06-05 by gdt