merbanan/rtl_433 #2020: Error when trying to compile Visual Studio 15 Code (in Visual Studio 2017)

OrangePledgeFanatic, 2022-03-22:
I get the following errors even after checking to make sure all of the files are in the right places as described in the build instructions:

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol __imp_rtlsdr_cancel_async referenced in function sdr_set_auto_gain	rtl_433 
\Downloads\rtl_433-master\rtl_433\vs15\sdr.obj	1

Error	LNK2019	unresolved external symbol __imp_rtlsdr_close referenced in function sdr_close	rtl_433	 \Downloads\rtl_433-master\rtl_433\vs15\sdr.obj	1	

Error	LNK2019	unresolved external symbol __imp_rtlsdr_get_center_freq referenced in function sdr_get_center_freq	rtl_433	\Downloads\rtl_433-master\rtl_433\vs15\sdr.obj	1	

Error	LNK2019	unresolved external symbol __imp_rtlsdr_get_device_count referenced in function sdr_open_rtl	rtl_433	\Downloads\rtl_433-master\rtl_433\vs15\sdr.obj	1	

Error	LNK2019	unresolved external symbol __imp_rtlsdr_get_device_name referenced in function sdr_open_rtl	rtl_433	\Downloads\rtl_433-master\rtl_433\vs15\sdr.obj	1

Error	LNK2019	unresolved external symbol __imp_rtlsdr_get_device_usb_strings referenced in function sdr_open_rtl	rtl_433	\Downloads\rtl_433-master\rtl_433\vs15\sdr.obj	1

Error	LNK2019	unresolved external symbol __imp_rtlsdr_get_index_by_serial referenced in function sdr_open_rtl	rtl_433	\Downloads\rtl_433-master\rtl_433\vs15\sdr.obj	1	

Error	LNK2019	unresolved external symbol __imp_rtlsdr_get_sample_rate referenced in function sdr_get_sample_rate	rtl_433	\Downloads\rtl_433-master\rtl_433\vs15\sdr.obj	1	

Error	LNK2019	unresolved external symbol __imp_rtlsdr_get_tuner_gains referenced in function rtlsdr_find_tuner_gain	rtl_433	\Downloads\rtl_433-master\rtl_433\vs15\sdr.obj	1	

The rest of the errors are all of the same type. There are 22 of them. I am using a Windows 10 Computer and I was trying to make the 64-bit build (x64). I shortened the path length here to get rid of personal identifying information about me and my computer's setup.

OrangePledgeFanatic, 2022-03-22:
Should it not be trying to import from a DLL? There are a lot of prerequisites and the compiler was having a hard time finding things, It took a lot of work adjusting the included folder names to get to where I am. I must have solved like 50 errors before I got these twenty-two errors. :( Unfortunately, I have never compiled a large C program like this before, I usually program in Visual Basic. I know that according to the documentation I need resources from libusb, librtlsdr, and pthread. Am I missing anything? A step by step tutorial would be great if somebody had the time or could point me in the right direction.

zuckschwerdt, 2022-03-23:
I guess Windows users never figured out a proper way to compile things with deps. It looks too confusing.

I simply compile with (for Visual Studio 16 2019 or Visual Studio 15 2017):

(get OpenSSL, if TLS wanted)
```
choco install openssl
```
(get Pothos and unpack)
```
curl -L -O https://downloads.myriadrf.org/builds/PothosSDR/PothosSDR-2021.05.01-vc16-x64.exe
mkdir -p pothos
7z x -opothos -y PothosSDR-2021.05.01-vc16-x64.exe
```
(build )
```
cmake -G"Visual Studio 16 2019" -Ax64 -DCMAKE_PREFIX_PATH=YOUR/PATH/TO/pothos -DENABLE_RTLSDR=ON -DENABLE_SOAPYSDR=ON -DENABLE_OPENSSL=ON
cmake --build . -j 10 --config Release
```

Or if you really want the VS15 solution, then you need to shuffle the libs around so they match up:
(get OpenSSL, if wanted)
```
choco install openssl
```
(get Pothos and split out the libs)
```
curl -L -O https://downloads.myriadrf.org/builds/PothosSDR/PothosSDR-2021.05.01-vc16-x64.exe
mkdir -p pothos
7z x -opothos -y PothosSDR-2021.05.01-vc16-x64.exe
mkdir -p libusb/include
cp -a pothos/include libusb
mkdir -p libusb/MS64/dll
cp pothos/lib/libusb-1.0.lib libusb/MS64/dll
cp pothos/bin/libusb-1.0.dll libusb/MS64/dll
mkdir -p rtl-sdr/x64
cp -a pothos/include/* rtl-sdr
cp pothos/lib/rtlsdr.lib rtl-sdr/x64
cp pothos/bin/rtlsdr.dll rtl-sdr/x64
```
(compile)
```
msbuild rtl_433.sln /property:Configuration=Release /property:Platform=x64
```

zuckschwerdt, 2022-03-23:
I never use Windows. Perhaps you can answer some questions and we can then update the build hints.

Do people still use choco? Or Is it winget now? (there is https://winget.run/pkg/ShiningLight/OpenSSL)

Can you use cmake? Do you use cmake?

The pre-build vs15 solution was great to get started, but you would need to setup lots of paths.
Why would people want that? Would there be some magic in MSBuild to auto-configure paths?

OrangePledgeFanatic, 2022-03-25:
I think in order to use cmake on windows I would have to install MinGW which I would rather not do, the fewer programs installed on my new computer the better. I can use a command line if I have to but would prefer to be able to configure everything with a GUI. I even tried copying the files around to where they were expected as the last resort, but that still did not work. Looking at the multiple different paths that can be set in Visual Studio is confusing in itself, there are path settings under General Configuration, VC++, C/C++, and Linker Properties.
[image]

zuckschwerdt, 2022-03-25:
It looks like CMake is integrated in Visual Studio these days? https://docs.microsoft.com/en-us/cpp/build/cmake-projects-in-visual-studio
And I don't think a full MinGW install is needed. At least there is https://community.chocolatey.org/packages/cmake and https://winget.run/pkg/Kitware/CMake

zuckschwerdt, 2022-04-24:
There are new build instructions for Visual Studio 2017: https://triq.org/rtl_433/BUILDING.html#visual-studio-2017
Should be really easy now, just a few clicks.

closed 2022-04-24 by zuckschwerdt