merbanan/rtl_433 #2024: It's not possible to enable all protocols at once with -G 4. what is the alternative?

henkiejan1, 2022-03-27:
Hi,

Today i discovered that -G 4 to enable all protocols is deleted from rtl_433. The suggestion is to use a config file or -R. But i can't figure out how to enable all protocols at once like -G 4 option does. How can i simple enable all protocols at once without making a config file?

Thank you very much!

zuckschwerdt, 2022-03-27:
Sorry, -G is not a sensible option, it should be avoided and in fact has now been removed.
Every protocol that can safely be enabled is active by default. Disabled protocols need special care, things will generally not work and produce warnings and false output.
If you really need extra protocols look through the list at https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf#L233 and decide for each protocol if you have that device.

closed 2022-03-27 by zuckschwerdt

sheilbronn, 2022-03-31:
Here's some code that might help or inspire - however, it depends on the formatting of the help message :(
You could also filter on starred (=disabled) entries....

```
# rtl_433 -R 99999 2>&1 | awk '$1 ~ /\[[0-9]+\]/ { gsub("[\\]\\[\\*]","",$1) ; print "protocol " $1  }'
protocol 01
protocol 02
protocol 03
protocol 04
protocol 06
protocol 07
protocol 08
protocol 10
...

```

zuckschwerdt, 2022-04-01:
Well, at least it's complicated enough to deter most people, hopefully.
Enabling all protocols blindly note that e.g. "sample rate too low for protocol" is *not an error* -- we don't want to hear about it.