merbanan/rtl_433 #2025: Jansite TPMS model TY-468-eu2.

Gadgeteering, 2022-03-27:
The Jansite TPMS currently sold on Amazon does work with #123 and #180, it seems to be encrypted using the first 8bits

[bitBench](https://triq.net/bitbench?c=407456f4fab4cf&c=337456f4edb4cf&c=277456f4e1b4cf&c=197456f4d3b4cf&c=107456f4cab4cf&c=087456f4c2b4cf&c=7e7456f4b8b4cf&c=2e7456f4c9b3ef&c=377456f4d2b3ef&c=3d7456f4d8b3ef&c=457456f4e0b3ef&c=4b7456f4e6b3ef&c=507456f4ebb3ef&c=557456f4f0b3ef&c=587456f4f3b3ef&c=5e7456f4f9b3ef&c=607456f4fbb3ef&c=617456f4fcb3ef&c=637456f4feb3ef&c=647456f4ffb3ef&f=CHK%3A8h%20ID%3Ahhhhhh%20PRESSURE%3A8d%20TEMP%3A8d%20FLAG%3A%204h%20PAD%3A4h&i=true&cf=ADD&co=8&cl=48&cw=8#c=00005f5f64f00a189f5f%20%5BRR%209C.%20A%5D&c=0000636361f63e739b63%20%5BFR%208C.%20A%5D&c=00005d5d6766fb50a15d%20%5BRL%208C%20A%5D&c=00000f0fb5285141ef0f%20%20%5BRR%20%3F%208C%20A%5D&c=0000333391c436eccb33%20%5BRR%208C%20A%5D&c=00001b1baaae862be31b%20%5B%3F%5D&c=00002525a02e3318d925%20%5BFL%208%20A%5D&c=00000f0fb5285141ef0f%20%5BFL%2024%20b%5D&c=0000333391c436eccb33%20%5BRL%2024%20b%5D&c=00002b2b8953c3fcd22b%20%5BFR%206%20a%5D&c=000059596cea1012a559%20%5BRR%207%20A%5D&c=000079793b82df6c4379%20%5BFL%2024C%201.6BAR%5D&c=0000f2f27705bed886f2&c=000033338a3c25268933&c=000033338a3c25268933%20%5BFL%2015C%201.6%5D&c=00000505b90e53f8bc05&c=00001101ad1a4704ed11&c=000067375670f15a5667&c=00007f7f38b86ea97f7f&c=00005555685e03486855&c=00007f7f38b86ea97f7f&c=000031218c3a2724cd31&c=00001101ad1a4704ed11&c=00005121675a07445e51&c=00002121962a3714dd21&c=00003101873a2724a731&c=00000bfbac144dfef30b%20%5BFlat%20tyre%5D&f=%20ID%3Ahh%20PRESS%7C%20TEMP%208d%20%2C%20%7C8d%7C%208d%208d%208h%208h%20&a=Preamble&m=0000&cf=XOR&co=16&cl=24&cw=8)

zuckschwerdt, 2022-03-27:
Does it need (two-way) pairing with the head unit? That exchange then should have some key. Otherwise those few bits would just be a bad obfuscation.

Gadgeteering, 2022-03-27:
There is no two way pairing, I have managed to pair sensors to a different receiver but this required the sensor to be pressurised with the receiver in pairing mode.

It is some sort of encryption/obfuscation similar to  #1059 @Tomazzzzz

[Jansite TPMS](https://www.amazon.co.uk/gp/product/B08HD231RZ/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1)

[bitBench](https://triq.net/bitbench#c=0x00%200x49%200xad%200x52%200xad%200xf8%200x0%20&c=0x00%200x88%200x6d%200x92%200x6d%200xf8%200x0%20&c=0x00%200x22%200xad%200x52%200xad%200xf8%200x0%20&c=0x00%200x4e%200x00%200xfb%200xf4%200xf8%200x0%20&c=0x50%200x2d%200x00%200xfb%200x14%200x2f%200x0%20%5BPump%20up%5D&c=0x10%200x81%200x00%200xfb%200x34%200xfc%200x0%20%5BPump%20down%5D&c=0x00%200xb1%200x00%200xfb%200x34%200xfc%200x0%20&c=0x00%200x01%200x00%200xfb%200x34%200xf8%200x0%20&c=0x70%200x3b%200x00%200xfb%200x34%200xdc%200x0%20%20%20%5B17C%20Pump%20up%5D&c=0x50%200x33%200x00%200xf3%200x1c%200xe4%200x0%20%20%20%5BPump%20down%5D&f=Flag%20h%20ID%3Abbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb&cw=4)

merbanan, 2022-03-27:
Can you elaborate exactly how to decrypt the payload? And can you upload some samples also?

You can get one of these to generate the needed pressure:

https://www.biltema.se/fritid/tradgard/tradgardsredskap/trycksprutor/tryckspruta-15-l-2000030676

merbanan, 2022-03-27:
What method did you use to figure out the encryption/obfuscation?

zuckschwerdt, 2022-03-28:
Looks somewhat plausible, yes.
You can get the xor like in [this BitBench](https://triq.net/bitbench?c=B4B95F99FC&c=C6BD619BFA&c=97BB4E9CEB&c=B7BB5E9CFB&c=A8B1339DD8&c=A8B1539DF8&c=C8B1639D08&c=A9BF509EE9&c=C9BF609EF9&c=CAB5659F06&c=CBB3729007&c=ADB74492D5&c=ADB76492F5&f=8h%20%0Ah4x.%20%0A.4xh%20%0A.#c=7f4f367f7c93477f%5B20PSI%2023C%20ATTACHED%5D&c=39398439c24d8d39%5B20PSI%2023C%20STABLE%5D&c=6f5f506f8c838f6f%5BDETACHED%20SENSOR%5D&c=6f6f506f8c838f6f%5BDETACHED%200PSI%5D&c=27f79727d43b9f27%5BATTACHED%2020PSI%2014C%5D&c=714148718a855771%5B19PSI%2019C%5D&c=69694f69927d6369%5B18PSI%2020C%5D&c=31318931ca459f31%5B16PSI%2018C%5D&c=29299229d23dab29%5B15PSI%2017C%5D&c=6363596398779b63%5B0PSI%2016C%5D&c=69694f69927d4669%5B28PSI%2020C%5D&c=11e1a811ea25c311%5B15PSI%2019C%5D&f=8h%20&x=00).

Gadgeteering, 2022-03-28:
[bitBench](https://triq.net/bitbench?c=B4B95F99FC&c=C6BD619BFA&c=97BB4E9CEB&c=B7BB5E9CFB&c=A8B1339DD8&c=A8B1539DF8&c=C8B1639D08&c=A9BF509EE9&c=C9BF609EF9&c=CAB5659F06&c=CBB3729007&c=ADB74492D5&c=ADB76492F5&f=8h%20%0Ah4x.%20%0A.4xh%20%0A.#c=002600fb142f00%20%5B28PSI%2020C%5D&c=30490003ec3800%20%5B20PSI%2023C%20ATTACHED%5D&c=00bd00fb74b400%20%5B20PSI%2023CSTABLE%5D&c=002600fb140a00%20%5B18PSI%2020C%5D&c=f0b900fb34d200%20%5B15PSI19C%5D&c=00b800fb74ae00%20%5B16PSI18C%5D&c=00bb00fb148200%20%5B15PSI%2017C%5D&c=003a00fb14f800%20%5B0PSI%20%2016C%5D&c=d0b000f31cb800%20%5BATTACHED%2020%20PSI%2014C%5D&c=&c=%5BDifferent%20sensor%5D&c=009bf54170fc00%20%5B-15C%5D&c=00ba275e4ee000%20%5B8C%5D&c=&c=%5BDifferent%20sensor%5D&c=&c=&c=00391b5e3220%20%5B21C%2020%20PSI%5D&c=003d7d5df040%20%5B16C%200%20PSI%5D&c=00a600fb14fc%20%5B16C%20Alarm%20Flat%5D&c=00a1b581308f&f=8h%20%0Ah4x.%20%0A.4xh%20%0A.&cw=4)

merbanan, 2022-03-28:
First the question regarding the message integrity. The first and the last bytes are the same. So this can be used for checking. But what is not obvious is how the first byte is calculated. There must be a rule for that. By looking at the following messages we can see that at least the last value of the message does not affect the selection of the initial xor value.

```
'69 69 4f 69 92 7d 63 69', #18PSI 20C
'69 69 4f 69 92 7d 46 69', #28PSI 20C
```

These messages also indicate where the pressure is stored.

```
0x00 0x26 0x00 0xfb 0x14 [0x0a] 0x0 #18PSI 20C
0x00 0x26 0x00 0xfb 0x14 [0x2f] 0x0 #28PSI 20C
```

But looking at other messages,

```
'71 41 48 71 8a 85 57 71', #19 PSI 19C
'11 e1 a8 11 ea 25 c3 11', #15 PSI 19C
0x30 0x39 0x00 0xfb 0xf4 0x26 0x0 #19 PSI 19C
0xf0 0xb9 0x00 0xfb 0x34 0xd2 0x0 #15 PSI 19C
```
does not show the same pattern.

My opinion is that there is something more at play then just xor of the first/last byte. TPMS devices usually has a 3 byte id, flags, temp and pressure. This payload contains 6 bytes of values which would contain the typical TPMS contents.

I suggest that the id is contained in the first 3 bytes.

```
'69 [69 4f 69] 92 7d 63 69', #18PSI 20C
'31 [31 89 31] ca 45 9f 31', #16PSI 18C
'29 [29 92 29] d2 3d ab 29', #15PSI 17C
'63 [63 59 63] 98 77 9b 63', #0PSI 16C
```
The relation that the first byte is the same as the third indicate a deterministic algorithm. Other messages does not follow this pattern but more messages and corresponding readings are missing to clearly indicate a constant pattern.

sp372, 2022-06-30:
Hello, I have been reviewing the comments of the different sensors for a few days, I have these TPMS sensors that I bought a few years ago, I have tried to replicate the readings with the configurations for the jansite sensors, which are the closest since they use the infineon sp372 chip however I have not been able to find any matches.
I leave the necessary sensor data, what would be the parameters to use the rtl_433 and start uploading readings from my sensors to be able to decode them, thanks
https://articulo.mercadolibre.com.mx/MLM-885521300-kkmoon-tpms-sistema-de-monitoreo-de-presion-de-los-neumati-_JM#position=7&search_layout=grid&type=item&tracking_id=cff5def2-7586-41ec-b754-39b1fe42e58e
[image]

zuckschwerdt, 2022-06-30:
Try to capture a sample https://triq.org/rtl_433/ANALYZE.html
We can't otherwise know the timings. Usually for TPMS it would be `-X 'n=name,m=FSK_PCM,s=50,l=50,r=150'` or  `-X 'n=name,m=FSK_PCM,s=50,l=50,r=1200'` or  `-X 'n=name,m=FSK_PCM,s=100,l=100,r=300'`

sp372, 2022-07-03:
lecturas rtl_433>rtl_433 -A g009_433.92M_250k[29psi,24c].cu8
rtl_433 version 21.12-108-g5c17e4e3 branch master at 202206100919 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Users\ldvs\Desktop\lecturas rtl_433\rtl_433.conf"...
Trying conf file at "C:\Users\ldvs\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 186 out of 218 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-218 ]
Test mode active. Reading samples from file: g009_433.92M_250k[29psi,24c].cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Detected OOK package    @0.195784s
Analyzing pulses...
Total count:    1,  width: 14.81 ms             ( 3702 S)
Pulse width distribution:
 [ 0] count:    1,  width: 14808 us [14808;14808]       (3702 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width: 14808 us [14808;14808]       (3702 S)
 [ 1] count:    1,  width: 100004 us [100004;100004]    (25001 S)
Level estimates [high, low]:  12957,      8
RSSI: -1.0 dB SNR: 32.1 dB Noise: -33.1 dB
Frequency offsets [F1, F2]:   -3716,      0     (-14.2 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
view at https://triq.org/pdv/#AAB10239D8FFFF8155

Detected OOK package    @0.688896s
Analyzing pulses...
Total count:   32,  width: 16.56 ms             ( 4139 S)
Pulse width distribution:
 [ 0] count:    2,  width: 1536 us [1528;1544]  ( 384 S)
 [ 1] count:   14,  width:  308 us [308;320]    (  77 S)
 [ 2] count:    8,  width:  432 us [408;508]    ( 108 S)
 [ 3] count:    4,  width:  116 us [116;120]    (  29 S)
 [ 4] count:    2,  width:  700 us [700;700]    ( 175 S)
 [ 5] count:    2,  width:  212 us [212;212]    (  53 S)
Gap width distribution:
 [ 0] count:   30,  width:   76 us [72;84]      (  19 S)
 [ 1] count:    1,  width: 1008 us [1008;1008]  ( 252 S)
Pulse period distribution:
 [ 0] count:    3,  width: 1512 us [1320;1616]  ( 378 S)
 [ 1] count:   12,  width:  388 us [388;396]    (  97 S)
 [ 2] count:    8,  width:  512 us [488;588]    ( 128 S)
 [ 3] count:    4,  width:  196 us [196;196]    (  49 S)
 [ 4] count:    2,  width:  780 us [780;780]    ( 195 S)
 [ 5] count:    2,  width:  292 us [292;296]    (  73 S)
Pulse timing distribution:
 [ 0] count:    2,  width: 1536 us [1528;1544]  ( 384 S)
 [ 1] count:   14,  width:  308 us [308;320]    (  77 S)
 [ 2] count:    8,  width:  432 us [408;508]    ( 108 S)
 [ 3] count:    4,  width:  116 us [116;120]    (  29 S)
 [ 4] count:    2,  width:  700 us [700;700]    ( 175 S)
 [ 5] count:    2,  width:  212 us [212;212]    (  53 S)
 [ 6] count:   30,  width:   76 us [72;84]      (  19 S)
 [ 7] count:    1,  width: 1008 us [1008;1008]  ( 252 S)
 [ 8] count:    1,  width: 15444 us [15444;15444]       (3861 S)
Level estimates [high, low]:  14748,      9
RSSI: -0.5 dB SNR: 32.1 dB Noise: -32.6 dB
Frequency offsets [F1, F2]:   -4526,      0     (-17.3 kHz, +0.0 kHz)
Guessing modulation: No clue...

sp372, 2022-07-03:
[g009_433.92M_250k[29psi,24c].zip](https://github.com/merbanan/rtl_433/files/9034022/g009_433.92M_250k.29psi.24c.zip)

sp372, 2022-07-03:
Hello, the previous one is a preliminary sample before making samples in a massive way, any previous analysis for those who have more experience in the data?

sp372, 2022-07-03:
-X n=name,m=FSK_PCM,s=50,l=50,r=150 : without result (without '      ')
 -X n=name,m=FSK_PCM,s=50,l=50,r=1200 : without result

C:\Users\ldvs\Desktop\lecturas rtl_433>rtl_433 -X n=name,m=FSK_PCM,s=100,l=100,r=300
rtl_433 version 21.12-108-g5c17e4e3 branch master at 202206100919 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Users\Desktop\lecturas rtl_433\rtl_433.conf"...
Trying conf file at "C:\Users\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 187 out of 218 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-218 ]
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 (non-zero-copy) user-space buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 0
codes     : {4}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 5            data      : c0
codes     : {5}c0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 6            data      : 20
codes     : {6}20
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 5            data      : c0
codes     : {5}c0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 10           data      : b20
codes     : {10}b20
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 5            data      : c0
codes     : {5}c0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 5            data      : 00
codes     : {5}00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 8            data      : 88
codes     : {8}88
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 8            data      : 28
codes     : {8}28
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 22:45:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8

zuckschwerdt, 2022-07-03:
If you get just that few bits (the {...} part) then it's not working.

The sample you posted is OOK. There is a 15 ms warmup toggeling at 50 µs (i.e. 300 bits).
Then two packets with 32 bit preamble each, the data is 120 or 128 bit.

The sample rate of 250k is not enough to capture this signal, you need `-s 1000k` to get a new sample.

Then `-X 'n=name,m=OOK_PCM,s=50,l=50,r=150'`

sp372, 2022-07-03:
command: rtl_433  -X  n=name,m=OOK_PCM,s=50,l=50,r=150
[17psi,22c]
shows:
time      : 2022-07-03 10:52:51
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 300          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff8
codes     : {300}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:51
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:51
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:51
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:51
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:51
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:51
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 5            data      : c0
codes     : {5}c0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:51
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 157          data      : fffffffedef679e79e679e67ffe7e79bff3cf3c0
codes     : {157}fffffffedef679e79e679e67ffe7e79bff3cf3c0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:51
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 154          data      : fffffffedef33de7bcde7b3fff3f799ff9e79e0
codes     : {154}fffffffedef33de7bcde7b3fff3f799ff9e79e0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:52
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:52
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:52
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 0
codes     : {1}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:52
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-03 10:52:52
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 21           data      : 8fffc0
codes     : {21}8fffc0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time

zuckschwerdt, 2022-07-03:
> you need `-s 1000k -X 'n=name,m=OOK_PCM,s=50,l=50,r=150'`

sp372, 2022-07-04:
Te command should be :
rtl_433 -X n=name, m=OOK_PCM, s= 1000,l=50,r=150 ?

zuckschwerdt, 2022-07-04:
> > `rtl_433 -s 1000k -X 'n=name,m=OOK_PCM,s=50,l=50,r=150'`

sp372, 2022-07-04:
Ok i will try

sp372, 2022-07-04:
I'll try later

sp372, 2022-07-05:
C:\Users\Desktop\lecturas rtl_433>rtl_433 -s 1000k -X n=name,m=OOK_PCM,s=50,l=50,r=150
rtl_433 version 21.12-108-g5c17e4e3 branch master at 202206100919 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Users\Desktop\lecturas rtl_433\rtl_433.conf"...
Trying conf file at "C:\Users\ldvs\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 187 out of 218 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-218 ]
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 (non-zero-copy) user-space buffers
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 100000 Hz, 10.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 16           data      : 45ff
codes     : {16}45ff
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
[... 315 lines cut ...]
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:41:40
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
at 27psi, 25c

sp372, 2022-07-05:
C:\Users\Desktop\lecturas rtl_433>rtl_433 -s 1000k -X n=name,m=OOK_PCM,s=50,l=50,r=150
rtl_433 version 21.12-108-g5c17e4e3 branch master at 202206100919 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Users\Desktop\lecturas rtl_433\rtl_433.conf"...
Trying conf file at "C:\Users\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 187 out of 218 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-218 ]
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 (non-zero-copy) user-space buffers
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 100000 Hz, 10.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 8
codes     : {3}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 7            data      : 90
codes     : {7}90
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 4            data      : 8
codes     : {4}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
[... 1677 lines cut ...]
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-04 21:44:31
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 0
codes     : {3}0

LIBUSB_ERROR_NOT_FOUND: Entity not found!
Check your RTL-SDR dongle, USB cables, and power supply.

WARNING: async read failed (-5).
at 25psi,25c
C:\Users\Desktop\lecturas rtl_433>

zuckschwerdt, 2022-07-05:
That did not work.

Use:

`rtl_433 -s 1000k -X 'n=name,m=OOK_PCM,s=50,l=50,r=150,bits>=20'`

sp372, 2022-07-13:
rtl_433 -s 1000k -X 'n=name,m=OOK_PCM,s=50,l=50,r=150
25psi,26c
time      : 2022-07-12 18:42:28
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 162          data      : aaaaaaaa569656969695aa6669aaa96996a956960
codes     : {162}aaaaaaaa569656969695aa6669aaa96996a956960
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-12 18:42:28
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 162          data      : aaaaaaaa569656969695aa6669aaa96996a956960
codes     : {162}aaaaaaaa569656969695aa6669aaa96996a956960

sp372, 2022-07-13:
time      : 2022-07-12 19:11:45
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 179          data      : ffffd55555552b2d2b2d4b5554d5353354aaccab2b2d0
codes     : {179}ffffd55555552b2d2b2d4b5554d5353354aaccab2b2d0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-12 19:11:45
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 41           data      : aaaaaaaa560
codes     : {41}aaaaaaaa560

sp372, 2022-07-13:
time      : 2022-07-12 19:11:38
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 162          data      : aaaaaaaa565a565a96aaa9aa6a66a9559956565a0
codes     : {162}aaaaaaaa565a565a96aaa9aa6a66a9559956565a0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-12 19:11:38
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 162          data      : aaaaaaaa565a565a96aaa9aa6a66a9559956565a0
codes     : {162}aaaaaaaa565a565a96aaa9aa6a66a9559956565a0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

zuckschwerdt, 2022-07-13:
Looks good.

Collect more data in [a BitBench like this](https://triq.net/bitbench#c=%7B162%7Daaaaaaaa569656969695aa6669aaa96996a956960%5Btemp%20%3F%20pres%20%3F%5D&c=%7B162%7Daaaaaaaa569656969695aa6669aaa96996a956960%5Btemp%20%3F%20pres%20%3F%5D&c=%7B179%7Dffffd55555552b2d2b2d4b5554d5353354aaccab2b2d0%5Btemp%20%3F%20pres%20%3F%5D&c=%7B162%7Daaaaaaaa565a565a96aaa9aa6a66a9559956565a0%5Btemp%20%3F%20pres%20%3F%5D&c=%7B162%7Daaaaaaaa565a565a96aaa9aa6a66a9559956565a0%5Btemp%20%3F%20pres%20%3F%5D&f=7h%208h8h%208h8h%208h8h%208h8h&a=Sync&m=aaa9&d=MC).

sp372, 2022-07-14:
time      : 2022-07-14 12:52:15
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 306          data      : aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa0
codes     : {306}aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-14 12:52:16
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 162          data      : aaaaaaaa566a566a96aaaa5a6a56a9659aa5566a0
codes     : {162}aaaaaaaa566a566a96aaaa5a6a56a9659aa5566a0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-14 12:52:16
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 162          data      : aaaaaaaa566a566a96aaaa5a6a56a9659aa5566a0
codes     : {162}aaaaaaaa566a566a96aaaa5a6a56a9659aa5566a0
[
22C,15psi]

zuckschwerdt, 2022-07-14:
Add that to the BitBench above like:
```
{162}aaaaaaaa566a566a96aaaa5a6a56a9659aa5566a0[22C,15psi]
{162}aaaaaaaa599a599a959a556a66a6aa959955599a0[22C,18PSI]
```

But: are those 4 different sensors? All data bits change -- there is no common pattern.

sp372, 2022-07-14:
ok, the same sensor

sp372, 2022-07-14:
https://triq.net/bitbench#c=%7B162%7Daaaaaaaa56a656a69665aa96699aa999999956a60%5B27C%2C19PSI%5D&c=%7B162%7Daaaaaaaa5956aa569655aaa6696aa9a996a659560%5B27C%2C22PSI%5D&c=%7B162%7Daaaaaaaa556a556a9996a95a9556a6659a95556a0%5B28C%2C22PSI%5D&c=%7B162%7Daaaaaaaa56665666969aaa566a5aa959995a56660%5B28C%2C25PSI%5D&c=%7B162%7Daaaaaaaa6956695665aa5aa6596a59a9665969560%5B28C%2C27PSI%5D&c=%7B162%7Daaaaaaaa56665666969aaa566a5aa959995a56660%5B28C%2C25PSI%5D&c=%7B162%7Daaaaaaaa5656565696aaa9a66a6aa6a996a556560%5B28C%2C29PSI%5D&c=%7B162%7Daaaaaaaa659a659a6966596a5aa656956996659a0%5B28C%2C26PSI%5D&c=%7B162%7Daaaaaaaa6a5a6a5a5aa665aa5666655565966a5a0%5B28C%2C23PSI%5D&c=%7B162%7Daaaaaaaa555a555a99a6a6aa9566a6559aaa555a0%5B28C%2C21PSI%5D&c=%7B162%7Daaaaaaaa6a5a6a5a5aa665aa5666655565966a5a0%5B28C%2C23PSI%5D&c=%7B162%7Daaaaaaaa569656969695aa6669aaa96996a956960%5B27C%2C25PSI%5D&c=%7B162%7Daaaaaaaa69a669a665656596569a5a99656669a60%5B27C%2C27PSI%5D&c=%7B162%7Daaaaaaaa65a665a6696659965a9a56996a5565a60%5B26C%2C23PSI%5D&c=%7B162%7Daaaaaaaa5566556699a6a956955aa6599aa655660%5B26C%2C21PSI%5D&c=%7B162%7Daaaaaaaa659a659a696a596a5aa656956aa6659a0%5B26C%2C19PSI%5D&f=7h%208h8h%208h8h%208h8h%208h8h&a=Sync&m=aaa9&d=MC&cw=4

sp372, 2022-07-14:
the above readings are from the RL sensor

zuckschwerdt, 2022-07-15:
It's the same encryption as the SP372 #1820

sp372, 2022-07-15:
I await directions. I don't know how to dump the firmware, I have some experience I made the rgh to an xbox 360 corona plate

zuckschwerdt, 2022-07-15:
Best to ask the people at #1820 -- I can't help.

sp372, 2022-07-15:
Ok thank you very much

mhardy6351, 2022-07-15:
I have the same sensors. I hope for a proper decode as well.

sp372, 2022-07-16:
The information found speaks of a change in ID length for the sensor from ID=6 to ID=10. The data sheet has a lot of information I am trying to understand it

sp372, 2022-07-16:
[Sp372 data.pdf]datasheet (https://github.com/merbanan/rtl_433/files/9125130/Sp372.data.pdf)

Gadgeteering, 2022-08-07:
The receiver uses a sh79F7416 microcontroller this needs a JET51a programmer to reader the firmware so that it can be reverse engineered using the [dis51 command](http://plit.de/asem-51/dis51.html).

I think this is the only way to decipher the message transmitted by these sensors.

merbanan, 2022-08-07:
It seems the F7416 is arm based. Anyway it does not matter. If we can dump the firmware we can figure out the encoding.

White

merbanan, 2022-08-07:
"Encryption bits and code options can be programmed", this is text from an ad for the JET51a device. So there is a protection bit that most likely is set.

Hacking a mcu is a serious undertaking.

Gadgeteering, 2022-08-07:
I have gone back to the drawing board and analysed the signals there seems to be two

[image]
[image]

rtl_433 -s 1000k -X 'n=name,m=OOK_PCM,s=50,l=50,r=150,bits>=20'

aaaaaaaaaaaa5a565a56955a5a56a5996566a5a65a568
55555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555402d2b4aad2d2b52ccb2b352d32d2b40

zuckschwerdt, 2022-08-07:
Manchester decoded that's
`3131 8331 ca45 cd31 1`
`98c1 98e5 22e6 983`
or (inverted)
`cece 7cce 35ba 32ce 0`
`673e 671a dd19 670`
which XORed is
`0000 b200 fb74 fc00 1`
`0059 007d ba7e 001`
or (Differential)
`5352 8552 5ece 5752 1`
`a942 a92f 672b a91`
XORed
`0001 d601 0d9d 0401 1`
`00eb 0086 ce82 003`

Both the same pattern where the first byte suspiciously appears repeated.

sp372, 2022-10-28:
Hello then it is already decoded?

gdt, 2023-10-15:
@Gadgeteering Where are we on this?  I skimmed and it's not clear if there is enough to write even a partial decoder, but I read too fast..  Are you still working on this?