merbanan/rtl_433 #2028: Support for the ThermoPro TP20B

PaulKierstead, 2022-03-28:
So, I wanted a TP20 for my smoker, as it is supported, but amazon sent me a TP20B. Ah, well, lets have a look.

Its a 915 MHz device, using a CMT2119A radio, clearly in OOK mode. I've poked at it with URH, but ... it has too many ones :D I'm not new to protocol reveng, but am new to *radio* protocol reveng and don't really know where to go from here. There are never two "highs" in a row, and they seem fairly consistent in pulse length. The lows vary a little. I can run a lot of captures and the pr
[g002_915M_1000k-18c-18c.zip](https://github.com/merbanan/rtl_433/files/8366915/g002_915M_1000k-18c-18c.zip)
esumable preamble (12 or 16 bits) is very consistent, but things get less consistent after that. I've attached a sample, this one had both temperature sensors at 18 C. Where do I go from here? Any ideas on what encoding might be used here?

-A output:
```
Detected OOK package	@0.018656s
Pulse data: 117 pulses
[  0] Pulse:  766, Gap:  225, Period:  991
[  1] Pulse:  735, Gap:  257, Period:  992
[  2] Pulse:  789, Gap:  202, Period:  991
[  3] Pulse:  817, Gap:  174, Period:  991
[  4] Pulse:  705, Gap:  287, Period:  992
[  5] Pulse: 1413, Gap:  559, Period: 1972
[  6] Pulse:  761, Gap:  243, Period: 1004
[  7] Pulse:  832, Gap:  160, Period:  992
[  8] Pulse:  726, Gap:  265, Period:  991
[  9] Pulse:  821, Gap: 1147, Period: 1968
[ 10] Pulse:  719, Gap: 1255, Period: 1974
[ 11] Pulse:  772, Gap: 1207, Period: 1979
[ 12] Pulse:  790, Gap: 1178, Period: 1968
[ 13] Pulse:  755, Gap: 1213, Period: 1968
[ 14] Pulse:  747, Gap: 1221, Period: 1968
[ 15] Pulse:  750, Gap:  247, Period:  997
[ 16] Pulse:  783, Gap:  220, Period: 1003
[ 17] Pulse:  810, Gap:  182, Period:  992
[ 18] Pulse:  719, Gap:  273, Period:  992
[ 19] Pulse: 2724, Gap:  235, Period: 2959
[ 20] Pulse:  703, Gap:  289, Period:  992
[ 21] Pulse: 1698, Gap: 1279, Period: 2977
[ 22] Pulse:  819, Gap:  172, Period:  991
[ 23] Pulse:  736, Gap: 1232, Period: 1968
[ 24] Pulse:  739, Gap: 1229, Period: 1968
[ 25] Pulse:  777, Gap: 1197, Period: 1974
[ 26] Pulse:  759, Gap: 1220, Period: 1979
[ 27] Pulse:  745, Gap: 1223, Period: 1968
[ 28] Pulse:  772, Gap: 1196, Period: 1968
[ 29] Pulse:  738, Gap:  254, Period:  992
[ 30] Pulse:  800, Gap: 1168, Period: 1968
[ 31] Pulse:  778, Gap:  213, Period:  991
[ 32] Pulse:  706, Gap: 1268, Period: 1974
[ 33] Pulse:  838, Gap: 1142, Period: 1980
[ 34] Pulse:  744, Gap: 1224, Period: 1968
[ 35] Pulse:  787, Gap:  204, Period:  991
[ 36] Pulse:  724, Gap: 1244, Period: 1968
[ 37] Pulse:  708, Gap:  290, Period:  998
[ 38] Pulse: 1715, Gap: 3224, Period: 4939
[ 39] Pulse:  743, Gap:  248, Period:  991
[ 40] Pulse:  736, Gap:  256, Period:  992
[ 41] Pulse:  820, Gap:  170, Period:  990
[ 42] Pulse:  868, Gap:  125, Period:  993
[ 43] Pulse:  736, Gap:  261, Period:  997
[ 44] Pulse:  809, Gap: 1170, Period: 1979
[ 45] Pulse:  757, Gap:  235, Period:  992
[ 46] Pulse:  747, Gap:  245, Period:  992
[ 47] Pulse:  756, Gap:  240, Period:  996
[ 48] Pulse:  805, Gap: 1175, Period: 1980
[ 49] Pulse:  792, Gap: 1176, Period: 1968
[ 50] Pulse:  747, Gap: 1221, Period: 1968
[ 51] Pulse:  803, Gap: 1165, Period: 1968
[ 52] Pulse:  749, Gap: 1219, Period: 1968
[ 53] Pulse:  783, Gap: 1186, Period: 1969
[ 54] Pulse:  792, Gap:  199, Period:  991
[ 55] Pulse:  771, Gap:  221, Period:  992
[ 56] Pulse:  871, Gap:  126, Period:  997
[ 57] Pulse:  784, Gap:  219, Period: 1003
[ 58] Pulse:  800, Gap: 1168, Period: 1968
[ 59] Pulse:  718, Gap:  274, Period:  992
[ 60] Pulse:  708, Gap:  283, Period:  991
[ 61] Pulse: 1713, Gap: 1264, Period: 2977
[ 62] Pulse:  711, Gap:  280, Period:  991
[ 63] Pulse:  757, Gap: 1211, Period: 1968
[ 64] Pulse:  812, Gap: 1157, Period: 1969
[ 65] Pulse:  769, Gap: 1199, Period: 1968
[ 66] Pulse:  736, Gap: 1232, Period: 1968
[ 67] Pulse:  752, Gap: 1222, Period: 1974
[ 68] Pulse:  763, Gap: 1216, Period: 1979
[ 69] Pulse:  721, Gap:  271, Period:  992
[ 70] Pulse: 2716, Gap:  243, Period: 2959
[ 71] Pulse:  877, Gap: 1091, Period: 1968
[ 72] Pulse:  765, Gap: 1203, Period: 1968
[ 73] Pulse:  716, Gap: 1258, Period: 1974
[ 74] Pulse:  795, Gap:  208, Period: 1003
[ 75] Pulse:  707, Gap: 1261, Period: 1968
[ 76] Pulse:  814, Gap:  178, Period:  992
[ 77] Pulse:  717, Gap:  282, Period:  999
[ 78] Pulse:  821, Gap: 3107, Period: 3928
[ 79] Pulse:  742, Gap:  255, Period:  997
[ 80] Pulse: 1837, Gap:  157, Period: 1994
[ 81] Pulse:  710, Gap:  283, Period:  993
[ 82] Pulse:  902, Gap:   88, Period:  990
[ 83] Pulse:  790, Gap: 1179, Period: 1969
[ 84] Pulse:  705, Gap:  292, Period:  997
[ 85] Pulse: 1900, Gap:   94, Period: 1994
[ 86] Pulse:  750, Gap: 1219, Period: 1969
[ 87] Pulse:  761, Gap: 1207, Period: 1968
[ 88] Pulse:  746, Gap: 1222, Period: 1968
[ 89] Pulse:  756, Gap: 1218, Period: 1974
[ 90] Pulse:  810, Gap: 1169, Period: 1979
[ 91] Pulse:  754, Gap: 1214, Period: 1968
[ 92] Pulse:  776, Gap:  215, Period:  991
[ 93] Pulse:  802, Gap:  190, Period:  992
[ 94] Pulse:  778, Gap:  219, Period:  997
[ 95] Pulse:  745, Gap:  257, Period: 1002
[ 96] Pulse: 2784, Gap:  176, Period: 2960
[ 97] Pulse:  780, Gap:  212, Period:  992
[ 98] Pulse:  731, Gap:  266, Period:  997
[ 99] Pulse: 1904, Gap:   75, Period: 1979
[100] Pulse:  752, Gap:  240, Period:  992
[101] Pulse:  704, Gap: 1264, Period: 1968
[102] Pulse:  723, Gap: 1245, Period: 1968
[103] Pulse:  759, Gap: 1209, Period: 1968
[104] Pulse:  742, Gap: 1232, Period: 1974
[105] Pulse:  807, Gap: 1172, Period: 1979
[106] Pulse:  746, Gap: 1222, Period: 1968
[107] Pulse:  733, Gap:  259, Period:  992
[108] Pulse:  873, Gap: 1100, Period: 1973
[109] Pulse:  777, Gap:  226, Period: 1003
[110] Pulse:  758, Gap: 1210, Period: 1968
[111] Pulse:  758, Gap: 1210, Period: 1968
[112] Pulse:  797, Gap: 1171, Period: 1968
[113] Pulse:  743, Gap:  249, Period:  992
[114] Pulse:  710, Gap: 1258, Period: 1968
[115] Pulse:  798, Gap:  199, Period:  997
[116] Pulse:  749, Gap: 25001, Period: 25750
Analyzing pulses...
Total count:  117,  width: 754.10 ms		(188525 S)
Pulse width distribution:
 [ 0] count:  107,  width: 3064 us [2812;3608]	( 766 S)
 [ 1] count:    7,  width: 6960 us [5652;7616]	(1740 S)
 [ 2] count:    3,  width: 10964 us [10864;11136]	(2741 S)
Gap width distribution:
 [ 0] count:   44,  width:  984 us [796;1168]	( 246 S)
 [ 1] count:    9,  width:  692 us [628;760]	( 173 S)
 [ 2] count:    1,  width: 2236 us [2236;2236]	( 559 S)
 [ 3] count:   55,  width: 4824 us [4364;5116]	(1206 S)
 [ 4] count:    2,  width: 12660 us [12428;12896]	(3165 S)
 [ 5] count:    2,  width:  500 us [500;504]	( 125 S)
 [ 6] count:    3,  width:  340 us [300;376]	(  85 S)
Pulse period distribution:
 [ 0] count:   52,  width: 3976 us [3960;4016]	( 994 S)
 [ 1] count:   57,  width: 7884 us [7872;7976]	(1971 S)
 [ 2] count:    5,  width: 11864 us [11836;11908]	(2966 S)
 [ 3] count:    1,  width: 19756 us [19756;19756]	(4939 S)
 [ 4] count:    1,  width: 15712 us [15712;15712]	(3928 S)
Pulse timing distribution:
 [ 0] count:  107,  width: 3064 us [2812;3608]	( 766 S)
 [ 1] count:    7,  width: 6960 us [5652;7616]	(1740 S)
 [ 2] count:    5,  width: 11644 us [10864;12896]	(2911 S)
 [ 3] count:   44,  width:  984 us [796;1168]	( 246 S)
 [ 4] count:    9,  width:  692 us [628;760]	( 173 S)
 [ 5] count:    1,  width: 2236 us [2236;2236]	( 559 S)
 [ 6] count:   55,  width: 4824 us [4364;5116]	(1206 S)
 [ 7] count:    2,  width:  500 us [500;504]	( 125 S)
 [ 8] count:    3,  width:  340 us [300;376]	(  85 S)
 [ 9] count:    1,  width: 100004 us [100004;100004]	(25001 S)
Level estimates [high, low]:   6330,    473
RSSI: -8.3 dB SNR: 22.5 dB Noise: -30.8 dB
Frequency offsets [F1, F2]:    6106,      0	(+23.3 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 3064, long_width: 6960, reset_limit: 12900, sync_width: 10964
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=3064,l=6960,r=12900,g=0,t=0,y=10964'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 4 
[00] {19} fb ff e0             : 11111011 11111111 111
[01] {50} bf ff df ff ff bf c0 : 10111111 11111111 11011111 11111111 11111111 10111111 11
[02] {25} ff bd ff 80          : 11111111 10111101 11111111 1
[03] {20} df ff f0             : 11011111 11111111 1111
```

zuckschwerdt, 2022-03-29:
Demod to OOK: `rtl_433 -w g002_915M_1000k-18c-18c.ook g002_915M_1000k-18c-18c.cs16` drop on https://triq.org/pdv/

Compare visually by also dropping the cs16 there. Not perfect, you are in the DC and also got a spike there (from the cs16 conversion likely).

Looks like PPM, short 200, long 1200. But he signal is not clean enough to pickup automatically.

PaulKierstead, 2022-03-29:
> Demod to OOK: `rtl_433 -w g002_915M_1000k-18c-18c.ook g002_915M_1000k-18c-18c.cs16` drop on https://triq.org/pdv/
>
>
>
> Compare visually by also dropping the cs16 there. Not perfect, you are in the DC [...]

Ah, that put me on the right track. I tried a couple of other capture devices (that one was with soapySDR/hackrf, I think the gain needed adjusted...) and it actually occasionally gets picked up by decoder 97 (with frequency forced to 915 MHz), but not very reliably, but a custom OOK_PPM will pick it up very reliably and the encoding is clearly identical to the TP-12/20 (#97). I have a way forward now, thank you very much!

zuckschwerdt, 2022-03-29:
The "Thermopro TP08/TP12/TP20 thermometer" has PPM and short_width = 500, long_width  = 1500, so that should work if you can get better reception (offset center frequency and less gain).

PaulKierstead, 2022-03-30:
500/1500 works, but is riiiiight on the line, a little lower would be more reliable. Not sure how that would affect the other ones (they could be on the line in the other direction for all I know...). 

However, there is a bug in thermopro_tp12.c (or bitbuffer.c, not sure which to blame as not sure the intent), which combined with the behavior of the TP20B ends up missing a lot samples. From the TP20B, I get a lot like this:
```
time      : 2022-03-30 13:09:35
model     : name         count     : 3             num_rows  : 3
rows      : 
len       : 41           data      : 0477116d660, 
len       : 41           data      : 0477116d668, 
len       : 40           data      : 0477116d66
codes     : {41}0477116d660, {41}0477116d668, {40}0477116d66
```
That last bit is always missing from the last row (like the other models), but also it is sometimes 0, and sometimes 1 in the previous 2 rows. Now if you look at line 60 in [thermopro_tp12.c](https://github.com/merbanan/rtl_433/blob/0ab9adf92917c238d67452e848bedb9dee174627/src/devices/thermopro_tp12.c), you'd think (as was clearly the intent) that row 0 would match row 2, or even row 1. However, looking at  [bitbuffer.c](https://github.com/merbanan/rtl_433/blob/0ab9adf92917c238d67452e848bedb9dee174627/src/bitbuffer.c) in line 492 we will call `bitbuffer_count_repeats` with max_bits of `0`. Because of that, looking at line 471-473, row 0/1 cannot match row 2 despite the intention, as they do not have the same number of bits, and row 0 will not match row 1 in our case because of the flipped last bit (and we passed max_bits == 0)

I'd put this bug on thermopro_tp12.c as the find repeated row says *min_bits*, not "compare this exact number" so I think it `bitbuffer_find_repeated_row` is working as intended, it is just not what was desired in this case. I can try to figure out how to fix thermopro_tp12.c to match rows while ignoring the last bit altogether without breaking anything else and do a PR if you like.

PaulKierstead, 2022-03-30:
Put in a PR, will not affect anything else, tested it out and much more reliable now (truly only requires 2 rows to match, ignoring the last bit)

zuckschwerdt, 2022-03-30:
Good analysis! Thanks. There might be overlap with `bitbuffer_find_repeated_prefix()` recently added. I guess this is a "prefix" of length 40 we want to match, right?

PaulKierstead, 2022-03-30:
> 

Lol, I should have scrolled down a bit in the file, indeed that will do the trick nicely. I'll update to use that.

closed 2022-04-04 by PaulKierstead