merbanan/rtl_433 #2031: Extend -M to allow for frequency metadata

sheilbronn, 2022-03-29:
May I suggest adding an option `-M frequency` to have the frequency added to each outputted line?!

There are two rationales for this feature request:
1. When running the DVB stick in hopping mode the message announcing the frequency hop sometimes seems to be issued to the pipe after the line issuing the received output, therefor making the output to seem to have come from the wrong frequency.
2. A logic trying to cache the value from the last frequency hop message can be avoided.

If there is interest I could try to find out the necessary code changes myself?! Thanks a lot anyway!

zuckschwerdt, 2022-03-29:
Yes, `-M level` has the detected carrier frequency ;)

sheilbronn, 2022-03-29:
@zuckschwerdt : However, it seems to me that there might be some sort of race condition with "-M level" where a received 433Mhz message is outputted to late and therefor carries an 866 Mhz tag (see line 4 below). Is this worth an issue? (The first column is a time HHMMSS):
```
152754 rtl/433/Bresser-3CH/1 {"id":20,"channel": 1,"battery_ok": 1,"temperature":18,"humidity":50,"freq":433}
152754 rtl/bridge/log {"note":"rtl2mqtt: Tuned to 868.300MHz."}
152754 rtl/bridge/log {"center_frequency":868300000, "frequencies":[433900000, 868300000], "hop_times":[61]}
152755 rtl/868/Bresser-3CH/1 {"id":20,"channel": 1,"battery_ok": 1,"temperature":18,"humidity":50,"freq":868}
152856 rtl/bridge/log {"note":"rtl2mqtt: Tuned to 433.900MHz."}
152856 rtl/bridge/log {"center_frequency":433900000, "frequencies":[433900000, 868300000], "hop_times":[61]}
152947 rtl/433/Bresser-3CH/1 {"id":20,"channel": 1,"battery_ok": 1,"temperature":19,"humidity":50,"freq":433}
152958 rtl/bridge/log {"note":"rtl2mqtt: Tuned to 868.300MHz."}
152959 rtl/bridge/log {"center_frequency":868300000, "frequencies":[433900000, 868300000], "hop_times":[61]}
153000 rtl/bridge/log {"note":"rtl2mqtt: Current noise level -15.2 dB, estimated noise -8.4 dB"}

```

zuckschwerdt, 2022-03-29:
You might want to try #1978 which eliminates "dead-time" when hopping.
Otherwise, yes, there might be a logic bug when changing frequencies while a frame is still being processed.
I'll work on that once the acquire is changed to a thread, basically adding the state (freq, rate) to each frame, not just globally. So a change, which might be the fix, is already planned.

sheilbronn, 2022-03-29:
OK, I did the following and will report back after verifying for some time that the spurious reports of 868 MhZ packets have gone.
```
cd /opt/rtl_433
git fetch origin feat-sdrthread:feat-sdrthread
git fetch origin feat-sdrthread
git pull
./do_build.sh && cd build && make install
```

zuckschwerdt, 2022-03-29:
See if `git status` shows "On Branch feat-sdrthread", otherwise `git checkout feat-sdrthread`

sheilbronn, 2022-03-29:
> See if `git status` shows "On Branch feat-sdrthread", otherwise `git checkout feat-sdrthread`

So I now  did (sorry if there is a mistake):
```
# git status
On branch master
Your branch is ahead of 'origin/master' by 3 commits.
  (use "git push" to publish your local commits)

nothing to commit, working tree clean
# git checkout feat-sdrthread
Switched to branch 'feat-sdrthread'
# git pull origin feat-sdrthread
From https://github.com/merbanan/rtl_433
 * branch              feat-sdrthread -> FETCH_HEAD
Already up to date.
# ./do_build.sh && cd build && make install
```
... will report back.

sheilbronn, 2022-03-29:
BTW: I just noticed that there seems to be an "one off" error in the interpretion of the value given with -H (hop secs), e.g. a value of `-H 7` consistently leads to a hop period of **8 seconds**.... Shall I open a separate issue?

zuckschwerdt, 2022-03-29:
Well, we hop on `difftime(rawtime, cfg->hop_start_time) > cfg->hop_time[hop_index])`-- perhaps that should be `>=`.
I'll change that, but currently hopping takes a second or more itself, so there's no real accuracy anyway ;)

zuckschwerdt, 2022-03-29:
Fixed with 8e343ed now.

sheilbronn, 2022-03-29:
> Well, we hop on difftime(rawtime, cfg->hop_start_time) > cfg->hop_time[hop_index])-- perhaps that should be >=.
> I'll change that, but currently hopping takes a second or more itself, so there's no real accuracy anyway ;)

The observation on my Raspi 4 came from:
`# /usr/local/bin/rtl_433 -M protocol -M noise:300 -M level -C si -f 433.9M -f 868.3M -s 256k -Y minmax -H 22 -M stats:1:22 -F json`
which resulted in the following - note the consistent difference of **23** seconds between two hop logging messages ...:
```
223151 rtl/bridge/log {"note":"rtl2mqtt: rtlsdr_set_center_freq 868300000 = 0"}
223154 rtl/bridge/stats {"enabled":184,"since":"2022-03-29T22:31:32","frames":{"count":426,"fsk":59,"events":8},"stats":[{"device":30,"name":"Generic Remote SC226x EV1527","events":525,"ok":11,"messages":11,"abort_length":514},{"device":52,"name":"Bresser Thermo-/Hygro-Sensor 3CH","events":664,"ok": 3,"messages": 3,"abort_length":661},{"device":86,"name":"Wireless Smoke and Heat Detector GS 558","events":426,"ok": 1,"messages": 1,"abort_early":425}]}
223214 rtl/bridge/log {"note":"rtl2mqtt: rtlsdr_set_center_freq 433900000 = 0"}
223216 rtl/bridge/stats {"enabled":184,"since":"2022-03-29T22:31:54","frames":{"count":40,"fsk":29,"events":0},"stats":[]}
223237 rtl/bridge/log {"note":"rtl2mqtt: rtlsdr_set_center_freq 868300000 = 0"}
223238 rtl/bridge/stats {"enabled":184,"since":"2022-03-29T22:32:16","frames":{"count":490,"fsk":11,"events":2},"stats":[{"device":52,"name":"Bresser Thermo-/Hygro-Sensor 3CH","events":506,"ok": 2,"messages": 2,"abort_length":504}]}
223300 rtl/bridge/stats {"enabled":184,"since":"2022-03-29T22:32:38","frames":{"count": 0,"fsk": 0,"events":0},"stats":[]}
223300 rtl/bridge/log {"note":"rtl2mqtt: rtlsdr_set_center_freq 433900000 = 0"}
223322 rtl/bridge/stats {"enabled":184,"since":"2022-03-29T22:33:00","frames":{"count":498,"fsk":80,"events":0},"stats":[]}
223323 rtl/bridge/log {"note":"rtl2mqtt: rtlsdr_set_center_freq 868300000 = 0"}
223344 rtl/bridge/stats {"enabled":184,"since":"2022-03-29T22:33:22","frames":{"count":25,"fsk": 0,"events":0},"stats":[]}
223346 rtl/bridge/log {"note":"rtl2mqtt: rtlsdr_set_center_freq 433900000 = 0"}

```
Will verify the fix https://github.com/merbanan/rtl_433/commit/8e343edcd44dd843a3d8b7b0d0820ce5e25f8cc2 .

sheilbronn, 2022-03-30:
BTW @zuckschwerdt : I noticed the debugging output for the (verbose) log message for the frequency hop has changed to:
`rtlsdr_set_center_freq 868300000 = 0`

It would be a lot easier to parse and more concise if it were JSON again.v. Thanks!
`{"center_frequency":868300000}`

zuckschwerdt, 2022-03-31:
That's really just a temporary debugging output ;)

sheilbronn, 2022-03-31:
> That's really just a temporary debugging output ;)

As I mentioned in https://github.com/merbanan/rtl_433/issues/2036#issuecomment-1085145365 I was trying to derive some sort of indication on which frequency band the signal was received.... either by using the "center_frequency message" or by using the additional frequency data per signal (from `-M level`).