merbanan/rtl_433 #2033: oregon_scientific.c: Oregon-BTHGN129 Channel :4  is displayed as Channel : 3

reba0, 2022-03-30:
Hi  I  made some tests  wit  Oregon-BTHGN129
It show to that channel 4 is displayed as channel 3 but with different House Code: 
model: Oregon-BTHGN129     House Code: 174 is set to channel 4 by dip Switch and Is also correct displayed on to  Oregon Base Station 

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2022-03-30 14:22:01                    brand     : OS
model     : Oregon-BTHGN129                        House Code: 174
Channel   : 3            Battery   : 1             Celsius   : 14.90 C       Humidity  : 49 %          Pressure  : 930.00 hPa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2022-03-30 14:22:06                    brand     : OS
model     : Oregon-THGR122N                        House Code: 124
Channel   : 3           Battery   : 1             Temperature: 22.10 C      Humidity  : 36 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

zuckschwerdt, 2022-03-30:
I don't understand. BTHGN129 and THGR122N are different sensors, right?

reba0, 2022-03-31:
Correct ,different types
THGR122N is on. Channel 3
BTHGN129 is on channel 4 but displayed as 3 in rtl-433

zuckschwerdt, 2022-03-31:
Ok, true. There is https://github.com/merbanan/rtl_433/blob/master/src/devices/oregon_scientific.c#L94-L99
```
    if ((channel == 4)
            && (sensor_id & 0x0fff) != ID_RTGN318
            && sensor_id != ID_THGR810
            && (sensor_id & 0x0fff) != ID_RTHN129
            && sensor_id != ID_THGR328N)
        channel = 3; // sensor 3 channel number is 0x04
```
Sorry, I have no idea why. It does not seem right.

It looses information (4->3) and we really should delete that.

zuckschwerdt, 2022-03-31:
Looks like it was always there: https://github.com/merbanan/rtl_433/blob/5283961137a1d07710c9a88f1c6907eb9ba0bb3e/src/rtl_433.c#L218-L219

We can't know what sensors it was intended for. Delete and wait for bug reports?
But really we should "fix" what some UI displays. If the channels are 1,2,4 on some sensor then that's ok.

reba0, 2022-03-31:
Hi  Christian
Thanks for replay 
I am not able to follow you
If I can help with debug something  please  let me know and how I can do it 

Best regards 
Reto 

Reto Battaglia
Heitiwil 21 
3186 Düdingen 

Tel : +41 79 705 58 68 
EMail: ***@***.***

gdt, 2023-09-25:
Can you remove the code in question, rebuild, and test?  Set the sensor to each available channel and see what rtl_433 prints.  Do this for  every sensor you have that works with this driver, please.

zuckschwerdt, 2023-09-26:
I'm in favor of just removing that quirk, it is not documented, was never clear what devices it was intended for and generally we don't aim to emulate quirks of some display.
This will have breaking impact on some users with a not nown type of device on channel "3" -- it will rename to channel "4".

gdt, 2023-09-26:
OK - I have retitled this to be a bug that the mapping happens.

commit 4c34673c (2023-09-26): Fix Oregon Scientific channel renumbering, BREAKING change to THN132N channel 3 (closes #2033)
closed 2023-09-26 by zuckschwerdt