merbanan/rtl_433 #2036: freq1 and freq2 not calculated correctly when doing frequency hopping (at least for the LaCrosse-TX29IT)

sheilbronn, 2022-03-31:
I noticed differences regading the values of freq1 and freq2 depending whether frequency hopping is requested or not.
(I'm on branch `feat-sdrthread`)

With hopping **freq1** and **freq2** incorrectly remain at 433 Mhz (instead of 868Mhz) for the LaCrosse-TX29IT :
```
# /usr/local/bin/rtl_433 -M protocol -M noise:300 -M level -C si -f 433.9M -f 868.3M -s 1024k -Y minmax -H 61 -M stats:1:60 -F json -v
rtl_433 version 21.12-84-g6ed19f34 branch feat-sdrthread at 202203261440 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registering protocol [1] "Silvercrest Remote Control"
Registering protocol [2] "Rubicson Temperature Sensor"
Registering protocol [3] "Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor"
Registering protocol [4] "Waveman Switch Transmitter"
.....
Registering protocol [215] "Altronics X7064 temperature and humidity sensor"
Registered 184 out of 215 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, DVB-T Dongle, SN: 00000991
Detached kernel driver
Found Fitipower FC0013 tuner
Using device 0: Dexatek DK DVB-T Dongle (Logilink VG0002A)
Sample rate set to 1024000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
rtlsdr_set_center_freq 433900000 = 0
Tuned to 433.900MHz.
acquire_thread enter...
Allocating 15 zero-copy buffers
Current signal level -10.3 dB, estimated noise -15.0 dB
baseband_demod_FM: low pass filter for 1024000 Hz at cutoff 204800 Hz, 4.9 us
{"time" : "2022-03-31 21:35:00", "protocol" : 52, "model" : "Bresser-3CH", "id" : 20, "channel" : 1, "battery_ok" : 1, "temperature_C" : 7.778, "humidity" : 84, "mic" : "CHECKSUM", "mod" : "ASK", "freq" : 433.908, "rssi" : -6.470, "snr" : 12.171, "noise" : -18.642}
{"time" : "2022-03-31 21:35:57", "protocol" : 52, "model" : "Bresser-3CH", "id" : 20, "channel" : 1, "battery_ok" : 1, "temperature_C" : 7.778, "humidity" : 84, "mic" : "CHECKSUM", "mod" : "ASK", "freq" : 433.952, "rssi" : -1.325, "snr" : 16.426, "noise" : -17.751}
{"time" : "2022-03-31 21:35:58", "enabled" : 184, "since" : "2022-03-31T21:34:59", "frames" : {"count" : 1552, "fsk" : 443, "events" : 2}, "stats" : [{"device" : 52, "name" : "Bresser Thermo-/Hygro-Sensor 3CH", "events" : 1987, "ok" : 2, "messages" : 2, "abort_length" : 1985}]}
rtlsdr_set_center_freq 868300000 = 0
{"time" : "2022-03-31 21:36:03", "protocol" : 76, "model" : "LaCrosse-TX29IT", "id" : 13, "battery_ok" : 1, "newbattery" : 0, "temperature_C" : 23.700, "humidity" : 45, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.800, "freq2" : 433.905, "rssi" : -1.424, "snr" : 24.923, "noise" : -26.346}
{"time" : "2022-03-31 21:36:11", "protocol" : 76, "model" : "LaCrosse-TX29IT", "id" : 13, "battery_ok" : 1, "newbattery" : 0, "temperature_C" : 23.700, "humidity" : 45, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.800, "freq2" : 433.910, "rssi" : -0.691, "snr" : 27.473, "noise" : -28.165}
{"time" : "2022-03-31 21:36:19", "protocol" : 76, "model" : "LaCrosse-TX29IT", "id" : 13, "battery_ok" : 1, "newbattery" : 0, "temperature_C" : 23.700, "humidity" : 45, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.801, "freq2" : 433.906, "rssi" : -1.476, "snr" : 26.197, "noise" : -27.673}
{"time" : "2022-03-31 21:36:28", "protocol" : 76, "model" : "LaCrosse-TX29IT", "id" : 13, "battery_ok" : 1, "newbattery" : 0, "temperature_C" : 23.700, "humidity" : 45, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.800, "freq2" : 433.904, "rssi" : -1.126, "snr" : 27.401, "noise" : -28.527}
{"time" : "2022-03-31 21:36:36", "protocol" : 76, "model" : "LaCrosse-TX29IT", "id" : 13, "battery_ok" : 1, "newbattery" : 0, "temperature_C" : 23.700, "humidity" : 45, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.799, "freq2" : 433.905, "rssi" : -1.533, "snr" : 26.298, "noise" : -27.831}
{"time" : "2022-03-31 21:36:44", "protocol" : 76, "model" : "LaCrosse-TX29IT", "id" : 13, "battery_ok" : 1, "newbattery" : 0, "temperature_C" : 23.700, "humidity" : 45, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.802, "freq2" : 433.908, "rssi" : -1.222, "snr" : 25.482, "noise" : -26.704}
{"time" : "2022-03-31 21:36:52", "protocol" : 76, "model" : "LaCrosse-TX29IT", "id" : 13, "battery_ok" : 1, "newbattery" : 0, "temperature_C" : 23.700, "humidity" : 45, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.800, "freq2" : 433.908, "rssi" : -1.058, "snr" : 26.772, "noise" : -27.831}
{"time" : "2022-03-31 21:36:59", "enabled" : 184, "since" : "2022-03-31T21:35:59", "frames" : {"count" : 92, "fsk" : 29, "events" : 7}, "stats" : [{"device" : 76, "name" : "LaCrosse TX29IT, TFA Dostmann 30.3159.IT Temperature sensor", "events" : 7, "ok" : 7, "messages" : 7}]}
{"time" : "2022-03-31 21:37:00", "protocol" : 76, "model" : "LaCrosse-TX29IT", "id" : 13, "battery_ok" : 1, "newbattery" : 0, "temperature_C" : 23.700, "humidity" : 45, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.800, "freq2" : 433.905, "rssi" : -1.425, "snr" : 27.709, "noise" : -29.134}
rtlsdr_set_center_freq 433900000 = 0
^CSignal caught, exiting!
sdr_stop: EXITING...
......
```

Without hopping the frequency **868 MhZ (and not 433 Mhz)** is outputted for the same sensor LaCrosse-TX29IT:
```
# /usr/local/bin/rtl_433 -M protocol -M noise:300 -M level -C si  -f 868.3M -s 1024k -Y minmax  -M stats:1:60 -F json -v
rtl_433 version 21.12-84-g6ed19f34 branch feat-sdrthread at 202203261440 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registering protocol [1] "Silvercrest Remote Control"
Registering protocol [2] "Rubicson Temperature Sensor"
Registering protocol [3] "Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor"
Registering protocol [4] "Waveman Switch Transmitter"
....
Registering protocol [215] "Altronics X7064 temperature and humidity sensor"
Registered 184 out of 215 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, DVB-T Dongle, SN: 00000991
Detached kernel driver
Found Fitipower FC0013 tuner
Using device 0: Dexatek DK DVB-T Dongle (Logilink VG0002A)
Sample rate set to 1024000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
rtlsdr_set_center_freq 868300000 = 0
Tuned to 868.300MHz.
acquire_thread enter...
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 1024000 Hz at cutoff 204800 Hz, 4.9 us
{"time" : "2022-03-31 21:43:42", "protocol" : 76, "model" : "LaCrosse-TX29IT", "id" : 13, "battery_ok" : 1, "newbattery" : 0, "temperature_C" : 23.700, "humidity" : 45, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.199, "freq2" : 868.304, "rssi" : -1.776, "snr" : 24.806, "noise" : -26.581}
^CSignal caught, exiting!
sdr_stop: EXITING...
```
....

sheilbronn, 2022-03-31:
Thanks for explaining!
The reason I filed this issue is that I was looking for some indication **on which frequency band** (e.g. 433 or 868 MhZ) the signal was received! By using `-M level` I got a frequency indication **freq** on several other sensors, but not for the _LaCrosse-TX29IT_ where I only got the (wrong) values **freq1** and **freq2** as described above...

Please note that they are consistently wrong, i.e. even when several seconds have passed sind the hop.... - so a race condition doesn't seem to be the reason!

l-maes, 2022-05-07:
Hi sheilbronn,

This may be a bit awkward, but I am trying to get my rtl-sdr dongle (also FC0013) to work at 868MHz.

For a while I assumed my issues were due to FC0013 not working well around that frequency.

So imagine the surprise when I read you got it working at that frequency.

Naturally I wonder if its:

1. my antenna (stock monopole, placed on an aluminum ground plate...).
2. my software setup.

Would you mind specifying which rtl-sdr you use? osmocom, keenerd, merbanan, ...?
When I look at the histogram of output values, I see essentially 2 values per channel (80,FF).

When I use rtl_sdr with -O 'agc=0:dagc=1' I see a wider histogram (but I suspect its simply the diversity in the digital sine and cosine waves in the RTL's DDC).

Looking at the source file for tuner_fc0013.c the gain values don't make any sense, theres even the same input value mapping to different hex values. They are written into different bits of the same register (0x14) where merbanan recently found an improvement (setting bit 7 of the register but clearing bit 6, the code before did the reverse).

I temporarily modified fc0013_set_params() so that when this register is written, I first compute a decimal modulo 1000 of the frequency (extracting the last 3 decimal digits), then if this is in the range 0-255 I use this value to write into the register. This way I could cycle through all 255 values say in gqrx. I noticed better reception when the byte had the following binary/decimal values as follows:
0. very weak or absent
1. 01010xxx: 80   -   87 decimal: STRONG
2. very weak or absent
3. 01110xxx: 112 - 119 d: MEDIUM/WEAK but better than outside these ranges
4. very weak or absent
5. 11010xxx: 218 - 215 d: STRONG again
6. very weak or absent
7.  11110xxx: 240-247 decimal: MEDIUM/WEAK again
8. very weak or absent

So it seems x1y10xxx gives significantly better values
where x seems arbitrary while y=0 results in the "STRONG" and y=1 results in "MEDIUM" signals

I only inspected visually a bit and cycled linearly through the values so I only actually compared consecutive values.

so I believe the 3 lower bits *may* represent tuner gain, but certainly not the 5 lowest bits, and certainly not in the order as specified in tuner_fc0013.c.

I didnt try building a new antenna yet, perhaps that simply resolves my issue.

Of note: while I don't see much at 868 MHz (except for some interference phenomena if held close to my laptopscreen, or the occasional short blip) I do see the expected activity at 929.2 MHz (so the tuner's mixer seems to be working).

Did you ever pry open your dongle? does it simply say FC0013 or does it say more like FC0013/G or so ?
Mine simply states FC0013.

l-maes, 2022-05-07:
Also bizarre that we have the same usb VID, PID and for both of us rtl-sdr detects FC0013 but the osmocom listing states this VID/PID to have FC2580... in my case I confirm that my chip says FC0013 on the package, so the driver's detection is right.

sheilbronn, 2022-05-17:
> Would you mind specifying which rtl-sdr you use? osmocom, keenerd, merbanan, ...?

Not sure what I should be looking for but these are my rtl_433 startup parameters and the resulting output:
```
# /usr/local/bin/rtl_433 -M protocol -M noise:300 -M level -C si -f 433.91M -f 868.3M -s 1024k -Y minmax -f 433.91M -f 915M -s 1024k -Y minmax -f 433.91M -f 868.3M -s 1024k -Y minmax -f 433.91M -f 27.161M -s 1024k -Y minmax -R -17 -R -197 -R 29 -H 61 -M stats:1:300 -F json
rtl_433 version 21.12-102-gde4f0812 branch master at 202205102141 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 185 out of 218 device decoding protocols [ 1-4 8 11-12 15-16 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-196 199 201-215 217-218 29 ]
Detached kernel driver
Found Fitipower FC0013 tuner
Sample rate set to 1024000 S/s.
Tuner gain set to Auto.
Tuned to 433.910MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 1024000 Hz at cutoff 204800 Hz, 4.9 us
{"time" : "2022-05-17 21:41:16", "protocol" : 52, "model" : "Bresser-3CH", "id" : 20, "channel" : 1, "battery_ok" : 1, "temperature_C" : 15.500, "humidity" : 85, "mic" : "CHECKSUM", "mod" : "ASK", "freq" : 434.015, "rssi" : -10.658, "snr" : 18.062, "noise" : -28.720}
....
```

gdt, 2023-09-30:
Dup of #2468 most likely.   Please feel free to retest with current git master and file a new issue if it is still not right.

closed 2023-09-30 by gdt