merbanan/rtl_433 #2037: New LaCrosse Wind Sensor: LTV-W2 (Different than LTV-W1)

W3AXL, 2022-03-31:
There is a newer Lacrosse wind sensor, the LTV-W2, which does not get detected by the built-in LTV-W1 decoder.

This is interesting, because when you open up the unit its internal label claims it's still an LTV-W1 and the FCC ID is the same. Maybe updated firmware? The unit still transmits at 915 MHz but the protocol appears to be different and isn't detected by any of the built-in decoders.

Attached are the captures for several different wind speeds. From my very uneducated poking around using [PDV](https://triq.org/pdv/) it seems that it's still NRZ encoding but with a period of 430 uS instead of 104 uS as described [in the source](https://github.com/merbanan/rtl_433/blob/5f0ff6db624270a4598958ab9dd79bb385ced3ef/src/devices/lacrosse_r1.c) for the LTV-W1 decoder.

[lacross_ltv_w2_wind_captures.zip](https://github.com/merbanan/rtl_433/files/8393638/lacross_ltv_w2_wind_captures.zip)

W3AXL, 2022-03-31:
Some data decodes (shifted by 1 byte from the raw PDV decode and the trailing `0x00` bytes trimmed):

`FFFFFFFFFFFFFF77FE767D9B6DB015FFFFFE001FFEFC` (0 mph)
`FFFFFFFFFFFFFF77FE767D9B6DB01FFFFFFE0F8FFE3C` (5 mph)
`FFFFFFFFFFFFFF77FE767D9B6DB015FFFFFE14EFFF58` (12 mph)

W3AXL, 2022-04-01:
Updated zip file with more captures. Hoping someone can point me in the right direction towards figuring out how to implement a decoder for this new sensor.

[lacrosse_ltv-w2_rtl_433_captures.zip](https://github.com/merbanan/rtl_433/files/8393743/lacrosse_ltv-w2_rtl_433_captures.zip)

zuckschwerdt, 2022-04-01:
Yes, NRZ. But you tinkered with the filenames (metadata) too much ;) The ook are not at the right sample rate.
Use `rtl_433 -R 0 -X 'n=LTV-W2,m=FSK_PCM,s=100,l=100,r=6000' *.cu8`to get:
```
{233}555555555555555a5545ba912492001555554001554f800000000000000 [g_915.02M_1000k_ltv-w2_0mph_2]
{237}aaaaaaaaaaaaaaad2aa2dd48924900eaaaaaa000aaab7000000000000000 [g_915.02M_1000k_ltv-w2_0mph_3]
{237}aaaaaaaaaaaaaaad2aa2dd489249004aaaaaa000aaa8b000000000000000 [g_915.02M_1000k_ltv-w2_0mph_4]
{237}aaaaaaaaaaaaaaad2aa2dd489249000aaaaaa0a8aaa81000000000000000 [g_915.02M_1000k_ltv-w2_10mph]
{234}aaaaaaaaaaaaaaad2aa2dd48924900caaaaaa0c5aaac800000000000000 [g_915.02M_1000k_ltv-w2_12mph]
{236}aaaaaaaaaaaaaaad2aa2dd48924900eaaaaaa042aaa4200000000000000 [g_915.02M_1000k_ltv-w2_4mph]
{235}aaaaaaaaaaaaaaad2aa2dd48924900aaaaaaa057aaa1400000000000000 [g_915.02M_1000k_ltv-w2_5mph]
{235}aaaaaaaaaaaaaaad2aa2dd489249002aaaaaa07faaa6400000000000000 [g_915.02M_1000k_ltv-w2_8mph]
{237}aaaaaaaaaaaaaaad2aa2dd48924900aaaaaaa098aaa77000000000000000 [g_915.02M_1000k_ltv-w2_9mph]
```
[As BitBench](https://triq.net/bitbench#c=%7B233%7D555555555555555a5545ba912492001555554001554f800000000000000%20%5Bg_915.02M_1000k_ltv-w2_0mph_2%5D&c=%7B237%7Daaaaaaaaaaaaaaad2aa2dd48924900eaaaaaa000aaab7000000000000000%20%5Bg_915.02M_1000k_ltv-w2_0mph_3%5D&c=%7B237%7Daaaaaaaaaaaaaaad2aa2dd489249004aaaaaa000aaa8b000000000000000%20%5Bg_915.02M_1000k_ltv-w2_0mph_4%5D&c=%7B237%7Daaaaaaaaaaaaaaad2aa2dd489249000aaaaaa0a8aaa81000000000000000%20%5Bg_915.02M_1000k_ltv-w2_10mph%5D&c=%7B234%7Daaaaaaaaaaaaaaad2aa2dd48924900caaaaaa0c5aaac800000000000000%20%5Bg_915.02M_1000k_ltv-w2_12mph%5D&c=%7B236%7Daaaaaaaaaaaaaaad2aa2dd48924900eaaaaaa042aaa4200000000000000%20%5Bg_915.02M_1000k_ltv-w2_4mph%5D&c=%7B235%7Daaaaaaaaaaaaaaad2aa2dd48924900aaaaaaa057aaa1400000000000000%20%5Bg_915.02M_1000k_ltv-w2_5mph%5D&c=%7B235%7Daaaaaaaaaaaaaaad2aa2dd489249002aaaaaa07faaa6400000000000000%20%5Bg_915.02M_1000k_ltv-w2_8mph%5D&c=%7B237%7Daaaaaaaaaaaaaaad2aa2dd48924900aaaaaaa098aaa77000000000000000%20%5Bg_915.02M_1000k_ltv-w2_9mph%5D&f=ID%3A24h%20BATTLOW%3A1b%20STARTUP%3A1b%20%3F%3A2b%20SEQ%3A3h%20%3F%3A1b%208h8h8h%20WIND%3A12d%2012h%20CRC%3A8h%20TRAILER%208h8h8h8h8h8h8h8h&a=Preamble&m=aa2dd4&cw=4).

commit ec75dacf (2022-04-01): minor: Add LaCrosse LTV-W2 note (closes #2037)
closed 2022-04-01 by zuckschwerdt