merbanan/rtl_433 #2038: Add Support for Digitech XC-0430 Rain Gauge

PetarMilevski, 2022-04-01:
Hi all, I would like some help to add support for the Digitech XC-0430 rain gauge. 

I want to build a rainfall pluviograph from this device. So exporting the temperature reading is not that important at the moment (but others may want this feature).

I want to collect the number of the bucket tips that occur each time the unit transmits data and then to save that data to a text file. 

The unit transmits data at 917MHz frequency every 12 seconds.

I have created a pull request here:

https://github.com/merbanan/rtl_433_tests/pull/423

If you need any other data please let me know.

Thanks in advance, great project!!

zuckschwerdt, 2022-04-01:
Drop the files on https://triq.org/pdv/ -- you see it is FSK.
Run `rtl_433 -A` on some files. Too much noise, not detected as FSK. Someone needs to look for the problem.
Maybe try a clean recording to analyze the protocol, then solve the noise issue later.

> To get a clean signal remove the receiver antenna and place the device at 10cm to the receiver, that mostly isolates the transmissions.

PetarMilevski, 2022-04-01:
Thanks for your response. Leave it with me, I'll do as you suggest and try to get a cleaner recording.

PetarMilevski, 2022-04-02:
Hi i just push up some new data collected as per your recommendations. Please have a look and let me know if it's about good. If it's not, i really don't know what more i can do. Regards

zuckschwerdt, 2022-04-02:
If you have Pulseview (from SigRok) installed, then you can take a look at the demod:

`rtl_433 -W 16tips_temp16_6DegC_917M_1000k.sr 16tips_temp16_6DegC_917M_1000k.cu8`
[image]

The FSK is there and plain to see, but for some strange reason the demod ignores it. Something wrong with the pulse_detect. This will take some time to debug.

For comparison, here is what a signal from e.g. Bresser-6in1 looks like in the demod:

[image]

PetarMilevski, 2022-04-02:
Ok so the data i sent is "ok". I'm in no hurry, I'll leave it in your capable hands to debug. I have a La Crosse LTV-R1 coming in the mail next week which is supported by your software.I will try getting it to do what i need.

gdt, 2023-10-02:
@PetarMilevski Are you still working on this?  Plans?

gdt, 2024-06-05:
abandoned (>2y) and not enough useful info to justify remaining open

closed 2024-06-05 by gdt