merbanan/rtl_433 #2039: Is the wetter station WS2320 from MISOL supported?

psychos4, 2022-04-04:
Is the wetter station WS2320 (433 Mhz) from MISOL supported? 

for example: 
https://de.aliexpress.com/item/1005001848787647.html?gatewayAdapt=glo2deu&spm=a2g0o.cart.0.0.2f563c00Pocc7S&mp=1

zuckschwerdt, 2022-04-04:
That's maybe a rebranded LaCrosse WS2320. The WS2310 is supported. https://github.com/merbanan/rtl_433/blob/master/src/devices/lacrossews.c

LeDocteurDesBits, 2022-11-21:
Hello, I know that this question is a bit old but I had the same question and it took me a while to find the answer so I thought it would be beneficial to post the answer here. I bought the same weather station (433MHz too) from AliExpress and the station wasn't detected as a WS2320. It seems that these stations are rebranded WH65B (FSK PCM instead of OOK modulation). After enabling the FineOffset WH65B decoder the data are decoded successfully.

zuckschwerdt, 2022-11-21:
Thank you for the report! I'll add that as note to the decoder.

commit fca5408b (2022-11-21): minor: Add Misol WS2320 note to Fineoffset WH65B (closes #2039)
closed 2022-11-21 by zuckschwerdt