merbanan/rtl_433 #2040: rtl4333 package config file not processed during build

knight-of-ni, 2022-04-07:
The package config file rtl433.pc.in in the root folder does not get processed during cmake build.

Not sure if this is a leftover from the previous autotools build and should be deleted, or perhaps an appropriate configure_file() statement should be added to CMakeLists.txt. Please advise.

zuckschwerdt, 2022-04-07:
Thanks. Yes, that's currently an unused file. We hope to provide a proper library to integrate rtl_433 functions or use the decoders standalone someday soon. Timeframe is maybe end-of-year-ish.
If that file is making trouble for packagers then we could remove it, otherwise `configure_file` from CMake can pick this up once a self-contained lib is available.

knight-of-ni, 2022-04-07:
Thank you. I am preparing a specfile, to get rtl_433 added to fedora repos, and I expect to be asked about that pc config file during the peer review process.

zuckschwerdt, 2022-04-07:
Sounds good. Please don't use the `./do_build.sh` that's really just our CI. Unfortunate naming and people picked it up as "official way to build" -- it's not. And for the time being better not build or install the `data` lib (so no explicit `-DBUILD_SHARED_LIBS:BOOL=ON`. S.a. #2010

knight-of-ni, 2022-04-17:
Good news. The package review was approved:
https://bugzilla.redhat.com/show_bug.cgi?id=2073196

In the coming weeks, rtl-433 packages should appear in fedora repos, followed shortly thereafter in epel repos.

closed 2022-04-17 by knight-of-ni