merbanan/rtl_433 #2041: LaCrosse TX29U not detected

hamiltlr, 2022-04-09:
I have an outdoor temperature sensor, a LaCrosse TX29U. I used to be able to use rtl_433 to pick up the signal but now it no longer works.

I used to be able to just issue rtl_433 with the `-f 915M` flag and it detected it. I see that there are some similar issues, #1204 and #1870, but I tried adding the `-Y classic` flag but it was still not detected. I also tried with `-Y classic -s 250k` as suggested in the command output, as well as a sample rate of 2048k, but still nothing.

I can see other devices so I know my SDR is working. The indoor receiver is also picking up the signal so I know the thermometer is transmitting, so I'm not sure what else to try here. I assume this is a bug? I'm on version 21.12

zuckschwerdt, 2022-04-09:
As a first diagnostic, try to grab a sample (at 1M and also at `-s 250k`) https://triq.org/rtl_433/ANALYZE.html and upload here.
IIRC we could not determine what exactly causes the demod to misread the transmissions.

hamiltlr, 2022-04-09:
OK, I collected some samples with both 1M and 250k, assuming I did it correctly. I've attached them here
[rtl_433_dump.zip](https://github.com/merbanan/rtl_433/files/8457720/rtl_433_dump.zip)
.

hamiltlr, 2022-04-09:
Well this whole thing might have been user error on my part. I noticed sometime after collecting those samples that the indoor thermometer was no longer showing the outdoor temperature. So I re-inserted the batteries in both, and now it is detected by rtl_433:

```
time      : 2022-04-09 20:52:17
model     : LaCrosse-TX29IT
id        : 7
Battery   : 1
NewBattery: 1
Temperature: 21.5 C
Integrity : CRC
```

It even works without `-Y classic` or `-s 250k`, so I'm not sure what was happening before. It was definitely in a state where the indoor unit could read it but rtl_433 could not. But it's working now so ¯\\_(ツ)_/¯

I guess I'll update this issue if I have any more problems, unless anyone wants me to try anything else to reproduce it.

zuckschwerdt, 2022-04-10:
Good to hear it worked out!
Without `-Y autolevel` the reception might be cut off for faint devices, yes.

Closing for now.

closed 2022-04-10 by zuckschwerdt