merbanan/rtl_433 #2045: Unable to set Center Frequnecy during Hop

loca5790, 2022-04-14:
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9
WARNING: Failed to set center freq.

Trying to hop between frequencies of 315M and 345M and this message appears.  If i run one center frequency I have no issues.

gvanem, 2022-04-15:
On Windows? If yes, try to build with this [RTLSDR](https://github.com/old-dab/rtlsdr) library instead. It does not depend on the libusb junk.

loca5790, 2022-04-15:
> On Windows? If yes, try to build with this [RTLSDR](https://github.com/old-dab/rtlsdr) library instead. It does not depend on the libusb junk.

I am not on windows Ubuntu 22.04

csindle, 2023-02-13:
I get the same error message (on Ubuntu 22.04, with the distro's rtl-433 package version: 21.12-1), using a Flight Aware RTL SDR with built-in LNA.

```
rtl_433  -H5 -f433M -f868M

...
center_frequency: 868000000                        frequencies: 433000000, 868000000                     hop_times : 5
Allocating 15 zero-copy buffers
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9
WARNING: Failed to set center freq.

center_frequency: 433000000                        frequencies: 433000000, 868000000                     hop_times : 5
...
```

zuckschwerdt, 2023-02-13:
That version is too old to diagnose. But there will be major changes to that part with #1978 soon and things might just work then.

zuckschwerdt, 2023-03-07:
Any SDR parameter changes are now effected from a different thread, this should be faster and without any further problems.

closed 2023-03-07 by zuckschwerdt