merbanan/rtl_433 #2047: Govee water leak detector recognized as Smoke-GS558

AndreasBVB, 2022-04-15:
Hi all,

My system also decodes some RF buttons into Smoke-GS558. It receives the AcuRite 06002RM sensor as AcuRite-Tower. For the Govee leak detector, when the button is pressed, or when leak is detected, it receives the same message with code "5f7d58".

Not sure where to start debugging this, but the Smoke-GS558 message received is:
Message 0 received on rtl_433/Smoke-GS558/31722 at 6:43 PM:
{
    "time": "2022-04-15 22:43:04",
    "model": "Smoke-GS558",
    "id": 31722,
    "unit": 24,
    "learn": 0,
    "code": "5f7d58"
}

Thanks!

zuckschwerdt, 2022-04-15:
Yes, the Smoke-GS558 is similar to those very common "doorbell"-style transmissions. False positives are expected.
Disable the Smoke-GS558 protocol or just ignore it ;)

AndreasBVB, 2022-04-15:
but it is not receiving anything else from Govee except for the Smoke-GS558 now. If disable it should I expect something else?

AndreasBVB, 2022-04-15:
Also looking into rtl_433/src/devices/govee.c I am not receiving the rich data structure like this:
static char *output_fields[] = {
        "model",
        "id",
        "battery_ok",
        "battery_mV",
        "event",
        "code",
        "mic",
        NULL,
};

AndreasBVB, 2022-04-16:
OK found the problem. Somehow the service script was calling the Ubuntu 20.04 repository version of rtl_433 (outdated) and I haven't been noticed until now. Uninstalled the repository version and reinstalled the github version. Working now!

closed 2022-04-16 by AndreasBVB