merbanan/rtl_433 #2053: Insteon Protocol outputs no data

chaos511, 2022-04-22:
When I use the Insteon device option it doesn't show anything but if I use rtl_sdr tuned to the same 915 freq it shows that the SDR is receiving data and rtl_433 is just not decoding it properly

zuckschwerdt, 2022-04-22:
Grab a sample and upload here as zip so we can take a look.
https://triq.org/rtl_433/ANALYZE.html

chaos511, 2022-04-22:
[g003_914.95M_1000k.zip](https://github.com/merbanan/rtl_433/files/8543220/g003_914.95M_1000k.zip)
This should contain a direct message to the id of 2E3AE9 requesting the status and i think the ack from that device to the hub with id of 301A86

zuckschwerdt, 2022-04-22:
That signal basically looks ok, but is somewhat "loud" and clipped at the start. Also there is some other carrier in there.
View at https://triq.org/pdv/ to see that.
Compare to some signals at https://github.com/merbanan/rtl_433_tests/tree/master/tests/insteon

But the raw data is there, using `-X 'n=Insteon_F16,m=FSK_PCM,s=110,l=110,t=15,g=20000,r=20000,invert,match={16}0x6666'` we get
`{1583}7534b54ead4cb572d552d6cd34b574d55556b52b354ccccccccccccccccccccccccccccccccccccccccccceaaacd56b334caf334cad6d34acd7534b54ead196ae5aaa5ad9a696ae9aaaaad6cb4cb6ccccd9999999999999999999999999999999999999d5556aad666995e66995ada69735d4d2d53ab532d5cb6a96b669a5aba6aaaab5a955a666cce66666c99999933333333333333333333333333aaab555accd32bccd32b5b4d366ba9a5aa6ad4cb572d552d6cd34b574d55556b52cb34ccca656d534b4e`
We need to look into that code to figure out more.

zuckschwerdt, 2022-04-23:
If you look at the start of the signal it looks good. But then the noise takes over and blurs the bits too much:
[image]

You can get more output with e.g. `rtl_433 -R 159:vv g003_914.95M_1000k.cu8` (or running live). It sees 3 different packets but the noise truncates them all.

I guess it should work if you manage to reduce the noise.

gdt, 2023-09-30:
Seems like a bad capture, and thus not an actionable bug.  @chaos511 Feel free to restart an issue after you can resolve the signal quality issue.

closed 2023-09-30 by gdt